libusb/libusb #48: libusb_get_device_list always return 0 on Mac OSX Mountain Lion (10.8)

tranhuanltv, 2015-01-02:
I had tested libusb 1.0.19 on Mac OSX 10.9, 10.9.5, 10.10.1 everything work fine. But when I try it on Mac OSX 10.8, libusb_get_device_list function always return number device is zero then I try to run examples/listdevs nothing showing.

tranhuanltv, 2015-01-02:
Here ./examples/listdevs debug logging on OSX 10.8

[ 0.000005] [00000f07] libusb: debug [libusb_init] created default context
[ 0.000079] [00000f07] libusb: debug [libusb_init] libusb v1.0.19.10905
[ 0.000213] [00000f07] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x44016482
[ 0.000256] [00000f07] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x44016482
[ 0.002803] [00000f07] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x17441711e
[ 0.002846] [00000f07] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x17441711e
[ 0.004751] [00000f07] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1b5cdb4c7
[ 0.004843] [00000f07] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1b5cdb4c7
[ 0.006312] [00000f07] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1bd500342
[ 0.006377] [00000f07] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1bd500342
[ 0.015444] [00000f07] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x25ecce40940
[ 0.015599] [00000f07] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x25ecce40940
[ 0.017886] [00000f07] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x25edf2e5213
[ 0.017952] [00000f07] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x25edf2e5213
[ 0.019905] [0000120f] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.020466] [0000120f] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.020791] [00000f07] libusb: debug [usbi_add_pollfd] add fd 4 events 1
[ 0.020820] [00000f07] libusb: debug [usbi_add_pollfd] add fd 6 events 1
[ 0.020831] [00000f07] libusb: debug [libusb_get_device_list] 
[ 0.020887] [00000f07] libusb: debug [libusb_exit] 
[ 0.020891] [00000f07] libusb: debug [libusb_exit] destroying default context
[ 0.020897] [00000f07] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.020901] [00000f07] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.020913] [00000f07] libusb: debug [handle_events] poll() 2 fds with timeout in 0ms
[ 0.020934] [00000f07] libusb: debug [handle_events] poll() returned 0
[ 0.020942] [00000f07] libusb: debug [usbi_remove_pollfd] remove fd 4
[ 0.020960] [00000f07] libusb: debug [usbi_remove_pollfd] remove fd 6
[ 0.021343] [0000120f] libusb: debug [darwin_event_thread_main] darwin event thread exiting

hjelmn, 2015-01-08:
> After investigation I had detect the problem happen with OSX 10.8. When we use IOUSBDeviceInterface500 or IOUSBDeviceInterface550 it will cause cannot detect usb devices, I try to use IOUSBDeviceInterface320 everything working fine. So currently anyone who want compability with OSX 10.8 can force use approriate IOUSBDeviceInterface by remove IOUSBDeviceInterface500/550 define in darwin_usb.h

Interesting. If 500 and 550 are available then they should have worked. I will add a workaround to use the older version.

-Nathan

closed 2015-01-08 by hjelmn

seanm, 2017-12-29:
@tranhuanltv 

1) you mentioned "IOUSBDeviceInterface550", but there is no such thing.  You probably meant "IOUSBInterfaceInterface550", right?

2) IOUSBLib.h says that IOUSBInterfaceInterface550 was added in 10.8.2, so that fits with you saying that it worked in 10.8.5, but not in 10.8.0.

@hjelmn 

Your fix to also check deployment target was correct and necessary, but (presumably) because of Tran's typo you guarded kIOUSBDeviceInterfaceID500 too strictly.

I have a PR that revisits your fix it this issue:

https://github.com/libusb/libusb/pull/375

commit 8c23764f (2015-01-08): Do not use IOUSBDeviceInterface550 or kIOUSBInterfaceInterfaceID550 on OSX versions
older than 10.9.

Closes #48