libusb/libusb #49: Arguments for fxload.exe

prateek1108, 2015-01-13:
Hello Sir,

I want to upgrade my ios phone firmware by code while connecting iPhone by usb with my windows computer. Is by this libusb code Fxload.c can i upgrade my iphone firmware for .exe or code what parameter i need to pass.please help me out regarding this issue your help is really precious for me. please guide me to do this job.

mcuee, 2015-01-13:
No, you can not use fxload to upgrade your iPhone firmware. You should use Apple's software to do that.

closed 2015-01-13 by mcuee

prateek1108, 2015-01-13:
Hello,
Thanks a lot sir for your reply, Sir i want to update firmware of iPhone by code please help if you have any idea about the project or article to do my job. Please guide me

mcuee, 2015-01-14:
Sorry but this has nothing to do with libusb. You have to seek help
elsewhere.

-- 
Xiaofan