libusb/libusb #50: error while opening libusb solution

nehachauhan, 2015-01-23:
When opening a solution file, getting error "The project consists entirely of configurations that require support for platforms which are not installed on this machine. The project cannot be loaded." on windows vista 32-bit machine, using visual studio 2008

pbatard, 2015-02-16:
I'm afraid nobody is maintaining the Visual Studio 2005 solution any more (which I assume you are using for 2008, since we don't have a dedicated 2008 solution).

I vaguely remember trying to reduce the number of solutions we had to maintain, but being asked by old VS users to keep the 2005 solution files in. 

Until someone who uses 2005/2008 cares enough to send us fixes so that the solution is kept up to date, issues pertaining to VS2005/VS2008 user will simply not be addressed.

I will therefore close this issue.

closed 2015-02-16 by pbatard

nehachauhan, 2015-02-17:
Hi,
After installing STANDARDSDK_500 i was able to open the solution and it's working fine now .
Thank You.