libusb/libusb #54: libusb bulk transfer size limit 2GB

Xargas, 2015-02-25:
Hallo!

I'm trying to transfer a big amount of data over USB3.0, but the libusb API only allows ~2GB per libusb transfer (libusb_fill_bulk_transfer call: http://libusb.sourceforge.net/api-1.0/group__asyncio.html#gad4ddb1a5c6c7fefc979a44d7300b95d7 )

So I tried to allocate multiple transfers, submited them in order and expected that they seamlessly take over immediatly after each other. But this seems not to be the case. (I can verify this with sequence numbers/timestamps in the transferred data).

So my question is:
Is there a possibility to overcome the 2GB limit of one transfer with libusb?

Regards,
Markus

hjelmn, 2015-02-25:
The 2 GiB limit comes from the type used for the transfer length: int. With is 32bits on Linux i386 and X86_64. So there is no way to overcome the limit. We would have to break ABI and change it to long (or int64_t) to support large transfers. 

Should we make this change in 2.0?

-Nathan

Xargas, 2015-02-26:
Yes, I would say so.
If you look at the current development of CPUs: They are also going (or are already there) to 64 bit architecture. As I see it, the ABI would break, but the API could stay the same. So it only needs a recompile of the application to use an int64_t. This effort should be ok for the Linux/Unix community.
And if there is a major version jump it should also be ok for the Windows world.

LudovicRousseau, 2015-02-26:
We talked about using ``size_t`` instead of ``int`` for buffer sizes more than 2 years ago. See https://sourceforge.net/p/libusbx/mailman/message/29890205/

I am surprised I could not find any bug reports or a wiki page about the future v2.0 API. Maybe this got lost in the migration from libusbx to libusb?

mcuee, 2015-08-17:
This is related to V2.0 change.
Ref: http://libusb.org/ticket/85

ya1gaurav, 2015-08-18:
Currently, in libusb APIs data types are used like "int" is used at many places. So for such a change (as above) in future is a tedious task. If libusb moving to new version (aka 2.0), we can change data types to typedefs which can be modified easily in future. like UInt, UInt64, Int, Int64 etc. We can start these changes in chunks.

BrentRector, 2015-08-18:
Why not simply use the stdint.h typedefs: int16_t, uint16_t, and so on? Why create new typedefs?

hjelmn, 2015-08-18:
long should be fine for v2.0. I don't think anyone will want to transfer more than 2^64-1 bytes in a single usb transaction anytime soon.

LudovicRousseau, 2015-08-18:
size_t is THE correct type for a byte array size.
See https://en.wikipedia.org/wiki/C_data_types#stddef.h