libusb/libusb #56: libusb_get_device_list() list same device twice in Win 8.1

podonoghue, 2015-03-06:
Hi,

I have been using libusb with a custom USB device for a few years without problem but when updating to a WIndows 8.1 machine there is an issue with the device appearing twice in the list returned from
 libusb_get_device_list().

The really confusing thing is that the concurrence seems to depend on the serial number of the device.

The above occurs with my custom software but also can be seen with listdevs (offending device is 16d0:0567 and appears twice in the list with the same path) :
...\libusb-1.0.19-rc1-win\examples\bin32>listdevs.exe
045e:07dc (bus 1, device 27) path: 3
2109:0811 (bus 1, device 26) path: 11.3
2109:0811 (bus 1, device 23) path: 11.3.4
045e:07be (bus 1, device 6) path: 7
17e9:4305 (bus 1, device 33) path: 11.3.1
046d:c52f (bus 1, device 18) path: 2.6
045e:07bf (bus 1, device 5) path: 8
1286:204b (bus 1, device 9) path: 6
2109:3431 (bus 1, device 7) path: 2.3
16d0:0567 (bus 1, device 4) path: 2.3.3
045e:07ca (bus 1, device 20) path: 2.5
045e:07cc (bus 1, device 29) path: 2
17e9:4305 (bus 1, device 33) path: 11.3.1
2109:3431 (bus 1, device 3) path: 2.3.4
045e:07c6 (bus 1, device 21) path: 11.1
16d0:0567 (bus 1, device 4) path: 2.3.3
045e:07bf (bus 1, device 5) path: 8
8086:9c31 (bus 1, device 1)
2109:3431 (bus 1, device 3) path: 2.3.4
045e:07cb (bus 1, device 28) path: 11

A similar device (same firmware apart from serial number) is only reported once:
C:\Users\podonoghue\Downloads\libusb-1.0.19-rc1-win\examples\bin32>listdevs.exe
045e:07dc (bus 1, device 27) path: 3
16d0:0567 (bus 1, device 13) path: 2.3.3
2109:0811 (bus 1, device 26) path: 11.3
2109:0811 (bus 1, device 23) path: 11.3.4
045e:07be (bus 1, device 6) path: 7
17e9:4305 (bus 1, device 33) path: 11.3.1
046d:c52f (bus 1, device 18) path: 2.6
045e:07bf (bus 1, device 5) path: 8
1286:204b (bus 1, device 9) path: 6
2109:3431 (bus 1, device 7) path: 2.3
045e:07ca (bus 1, device 20) path: 2.5
045e:07cc (bus 1, device 29) path: 2
17e9:4305 (bus 1, device 33) path: 11.3.1
045e:07c6 (bus 1, device 21) path: 11.1
045e:07bf (bus 1, device 5) path: 8
8086:9c31 (bus 1, device 1)
2109:3431 (bus 1, device 3) path: 2.3.4
045e:07cb (bus 1, device 28) path: 11

Any ideas where the problem lies?

bye

mcuee, 2015-03-06:
On Fri, Mar 6, 2015 at 9:03 AM, podonoghue <notifications@github.com> wrote:

> The above occurs with my custom software but also can be seen with
> listdevs (offending device is 16d0:0567 and appears twice in the list with
> the same path) :
>
> ...\libusb-1.0.19-rc1-win\examples\bin32>listdevs.exe
>
Please use libusb-1.0.19 Windows binary and not rc1. Thanks.
And please post the output of "xusb -d 16d0:0567".

Make sure you did not install libusb-win32 filter over your
device.

-- 
Xiaofan

podonoghue, 2015-03-06:
Hi Xiafan,

Appear to be the same results for libusb-1.0.19.

I have attached:

listdevs_fail.txt  - Failing result for listdevs (as before)
xusb_fail.txt – xusb report for failing device

listdev_ok.txt – listdevs OK report for other device
xusb_ok.txt – xusb report for OK device

The two device are meant to only differ in serial number.

bye

16d0:0567 (bus 1, device 16) path: 2.3.3
16d0:0567 (bus 1, device 16) path: 2.3.3

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [0000116c] libusb: debug [libusb_init] created default context
[ 0.000000] [0000116c] libusb: debug [libusb_init] libusb v1.0.19.10905
[ 0.000000] [0000116c] libusb: debug [windows_init] Windows 8.1 64-bit
[ 0.000000] [0000116c] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.000000] [0000116c] libusb: info [winusbx_init] libusbK DLL is not available, will use native WinUSB
[ 0.000000] [0000116c] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.000000] [0000116c] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.000000] [0000116c] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.000000] [00001450] libusb: debug [windows_clock_gettime_threaded] hires timer available (Frequency: 2240906 Hz)
[ 0.000000] [0000116c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.000000] [0000116c] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.000000] [0000116c] libusb: debug [usbi_add_pollfd] add fd 1 events 1
[ 0.000000] [0000116c] libusb: debug [libusb_get_device_list] 
[ 0.000000] [0000116c] libusb: debug [windows_get_device_list] allocating new device for session [2B0]
[ 0.000000] [0000116c] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.000000] [0000116c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [0000116c] libusb: debug [windows_get_device_list] allocating new device for session [1D2]
[ 0.000000] [0000116c] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.000000] [0000116c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [0000116c] libusb: debug [windows_get_device_list] allocating new device for session [164]
[ 0.015627] [0000116c] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.015627] [0000116c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015627] [0000116c] libusb: debug [windows_get_device_list] allocating new device for session [2AC]
[ 0.015627] [0000116c] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.015627] [0000116c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015627] [0000116c] libusb: debug [windows_get_device_list] allocating new device for session [C4]
[ 0.015627] [0000116c] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.015627] [0000116c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015627] [0000116c] libusb: debug [windows_get_device_list] allocating new device for session [17A]
[ 0.015627] [0000116c] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.015627] [0000116c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015627] [0000116c] libusb: debug [windows_get_device_list] allocating new device for session [2CF]
[ 0.015627] [0000116c] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.015627] [0000116c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015627] [0000116c] libusb: debug [windows_get_device_list] allocating new device for session [356]
[ 0.015627] [0000116c] libusb: debug [windows_get_device_list] allocating new device for session [221]
[ 0.015627] [0000116c] libusb: debug [init_device] got bus number from ancestor #2
[ 0.015627] [0000116c] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.015627] [0000116c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 41 bytes)
[ 0.015627] [0000116c] libusb: debug [init_device] (bus: 1, addr: 27, depth: 1, port: 3): '\\.\USB#VID_045E&PID_07DC#026080543854'
[ 0.015627] [0000116c] libusb: debug [windows_get_device_list] found existing device for session [2CF] (0.0)
[ 0.015627] [0000116c] libusb: debug [init_device] got bus number from ancestor #2
[ 0.015627] [0000116c] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.015627] [0000116c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 31 bytes)
[ 0.015627] [0000116c] libusb: debug [init_device] (bus: 1, addr: 26, depth: 1, port: 3): '\\.\USB#VID_2109&PID_0811#6&26C2149F&0&3'
[ 0.015627] [0000116c] libusb: debug [windows_get_device_list] found existing device for session [17A] (0.0)
[ 0.015627] [0000116c] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.015627] [0000116c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 31 bytes)
[ 0.015627] [0000116c] libusb: debug [init_device] (bus: 1, addr: 23, depth: 2, port: 4): '\\.\USB#VID_2109&PID_0811#7&19ECA094&0&4'
[... 536 lines cut ...]
[ 0.046908] [000004d8] libusb: debug [windows_handle_events] checking fd 0 with revents = 0000
[ 0.046908] [000004d8] libusb: debug [windows_handle_events] checking fd 1 with revents = 0000
[ 0.046908] [000004d8] libusb: debug [windows_handle_events] checking fd 2 with revents = 0001
[ 0.046908] [000004d8] libusb: debug [usbi_remove_pollfd] remove fd 2
[... cut ...]
[ 0.046908] [000004d8] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 31, size 0
[ 0.046908] [000004d8] libusb: debug [windows_transfer_callback] detected endpoint stall
[ 0.046908] [000004d8] libusb: debug [usbi_handle_transfer_completion] transfer 004FACA4 has callback 003A9B16
[ 0.046908] [000004d8] libusb: debug [sync_transfer_cb] actual_length=0
[ 0.046908] [000004d8] libusb: debug [libusb_release_interface] interface 0
[ 0.046908] [000004d8] libusb: debug [libusb_release_interface] interface 1
[ 0.046908] [000004d8] libusb: debug [l

bvernoux, 2015-06-18:
Does someone is working on this issue ?

mcuee, 2015-06-21:
Seems no.

When you say that "the two device are meant to only differ in serial
number", what do you mean by that?

bvernoux, 2015-06-21:
It means such configuration for 1st Board:
        ------------------ Device Descriptor ------------------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x01 (Device Descriptor)
bcdUSB                   : 0x200 (USB Version 2.00)
bDeviceClass             : 0x00 (defined by the interface descriptors)
bDeviceSubClass          : 0x00
bDeviceProtocol          : 0x00
bMaxPacketSize0          : 0x40 (64 bytes)
idVendor                 : 0x1D50
idProduct                : 0x60A1
bcdDevice                : 0x100
iManufacturer            : 0x01
 Language 0x0409         : "www.airspy.com"
iProduct                 : 0x02
 Language 0x0409         : "AIRSPY"
iSerialNumber            : 0x03
 Language 0x0409         : "AIRSPY SN:16C463C8212185C7"
bNumConfigurations       : 0x01

Then such configuration for 2nd Board:
        ------------------ Device Descriptor ------------------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x01 (Device Descriptor)
bcdUSB                   : 0x200 (USB Version 2.00)
bDeviceClass             : 0x00 (defined by the interface descriptors)
bDeviceSubClass          : 0x00
bDeviceProtocol          : 0x00
bMaxPacketSize0          : 0x40 (64 bytes)
idVendor                 : 0x1D50
idProduct                : 0x60A1
bcdDevice                : 0x100
iManufacturer            : 0x01
 Language 0x0409         : "www.airspy.com"
iProduct                 : 0x02
 Language 0x0409         : "AIRSPY"
iSerialNumber            : 0x03
 Language 0x0409         : "AIRSPY SN:16C463C8325D5FC7"
bNumConfigurations       : 0x01

And such type of configuration is not managed correctly by libusb.

mcuee, 2015-06-21:
Most likely the device driver installation is the problem. Maybe you want
to uninstall the driver related to this device and the use Zadig to install
WinUSB driver again.

bvernoux, 2015-07-01:
There is no need to install Zadig driver with Win7 & Win8.x with AirSpy as it is managed by the AirSpy firmware which use feature of Windows Auto WinUsb Install (see https://msdn.microsoft.com/en-us/windows/hardware/gg463179 or more details)

mcuee, 2015-07-02:
I see. Just try to use Zadig and see if that makes any differences.

mcuee, 2015-08-26:
Not hearing from the report and I do not see this as a libusb problem. Probably a driver problem or device problem. Will reopen with more information.

closed 2015-08-26 by mcuee

podonoghue, 2015-09-19:
I am the original poster.

Please note that the issue posted by bvernoux is probably unrelated to the original issue which was a single device appearing more than once in the libusb list.

I am still waiting for a response from my second posting which supplied information requested.
I have no problem with you closing the issue but please don't indicate the reason as lack of response since I have provided all information requested and was awaiting a response.

I am no longer much interested in the issue as I added a check to remove repeated devices.

bye

bvernoux, 2015-09-27:
podonoghue could you check if your issue always happen with libusb 1.20 ?