libusb/libusb #57: LIBUSB_ERROR_NOT_FOUND thrown on libusb_bulk_transfer()

jndok, 2015-03-07:
Hi, I am encountering some issues with libusb. I was trying to read input from my keyboard, and came across `libusb_bulk_transfer()`. I wrote a simple prgram that loops through every device attached to my computer and when it finds the correct one (Product ID: 0x203) opens an handle and tries to call `libusb_bulk_transfer()`.

Here is a copy/paste of my code: http://pastebin.com/qjtugkuR

Error in output:
[image]

The problem is that `libusb_bulk_transfer()` absolutely refuses to work, by throwing a `-5` error code, aka `LIBUSB_ERROR_NOT_FOUND`. I have tried to use it in different ways, but the error is always there. Even tried on a different computer! Also tried running it in privileged mode with `sudo` and with different keyboards.

I am using OS X, on both my machines. Is this a `libusb` problem or am I doing something wrong?

mcuee, 2015-03-07:
Forget about using libusb with a keyboard under Mac OS X. It will not work
since the OS attaches the keyboard driver to it. If you want to use libusb
under Mac OS X, choose another device.

-- 
Xiaofan

mcuee, 2015-03-07:
BTW, usually keyboard is an HID device and it will not have bulk
endpoints. So libusb_bulk_transfer() is not correct. You should use
libusb_interrupt_transfer(). And you want to read the Input data, so
LIBUSB_ENDPOINT_OUT should be LIBUSB_ENDPOINT_IN. And that only sets
up the direction, you need a correct Endpoint to use (eg: EP 1 IN,
that will be Endpoint 0x81 including the direction bit).

Is this a learning exercise? If yes, choose another generic USB device.

-- 
Xiaofan

jndok, 2015-03-08:
Thank you for the reply. This is not a learning exercise however, I was trying to set LEDs on my keyboard, so I tried with `libusb` to send data to the device . Guess I'll switch to IOHIDManager stuff, that should do it.

mcuee, 2015-03-11:
I believe that is using the control transfer. You may want to try again to
see if libusb can work or not. It does not use the Interrupt Endpoint, so
it might work.

dickens, 2015-08-05:
Closing this since this doesn't appear to be a libusb problem. If the OP has more info to suggest otherwise this can be re-opened.

closed 2015-08-05 by dickens