libusb/libusb #58: libusb Android 5.0 Selinux policy

bhack, 2015-03-11:
In android 5.0(lollipop) google has changed security policy and now SELinux is enforced on all devices.
Can this be solved cherry picking some commit from [this libusb fork](https://github.com/Gritzman/libusb)?

bhack, 2015-03-11:
/cc @Gritzman

kuldeepdhaka, 2015-04-26:
Hello, i have uploaded a my branch that can be used for Android 5+ (tested on 5.2).  https://github.com/kuldeepdhaka/libusb/tree/android-open2
it contain all the commits that is required for opening device from fd with proper handling of SELinux

CAMOBAP795, 2016-03-17:
+1 for https://github.com/Gritzman/libusb solution because it dosn't break interface, very clean solution

bhack, 2016-03-17:
Is there any PR under review?

CAMOBAP795, 2016-03-17:
Looks like no because https://github.com/Gritzman/libusb isn't github fork

bhack, 2016-03-17:
OK.. Waiting for a @libusb reply

CAMOBAP795, 2016-03-21:
@Gritzman If you can prepare pull request it will be nice.

If you haven't time to do this, and if you do not mind, I can do it, just let me know

mcuee, 2016-03-21:
A pull-request will be nice.

On the other hand, it will be better that people who are interested in
Android libusb improvement to subscribe to the mailing list and discuss the
Android situation in the mailing list. Thanks, It is easier to discuss
things like this in libusb mailing list.

https://lists.sourceforge.net/lists/listinfo/libusb-devel

CAMOBAP795, 2016-03-21:
thanks @marcan sure I will follow up the way which will more appropriate for libusb community

As I can see there is a lot of topic related to Android https://sourceforge.net/p/libusb/mailman/search/?q=android, so I will review them and find more accurate equivalent of this #58 ticket

mcuee, 2016-12-02:
I will close this one and please continue at #188.

closed 2016-12-02 by mcuee