libusb/libusb #63: Cannot build using Visual Studio 2015 CTP

deisenhut, 2015-04-14:
I am attempting to build libusb-1.0.19 on Windows using Visual Studio 2015 CTP.  It builds fine when using Visual Studio 2013 with just a couple of deprecation warnings, but using Visual Studio 2015 CTP I get the following errors.  Note, I used the 2013 solution file and let VS upgrade the projects.  The only difference I see in the solution and project files are changes in the version number.

1>------ Build started: Project: libusb-1.0 (static), Configuration: Debug Win32 ------
2>------ Build started: Project: getopt, Configuration: Debug Win32 ------
3>------ Build started: Project: libusb-1.0 (dll), Configuration: Debug Win32 ------
1>  core.c
3>  core.c
1>D:\Perforce\my-project\3rdParty\Libusb\OriginalSource\libusb-1.0.19\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition
1>  C:\Program Files (x86)\Windows Kits\10\Include\ucrt\time.h(39): note: see declaration of 'timespec'
1>  descriptor.c
2>  getopt1.c
2>  getopt.c
2>  Generating Code...
3>D:\Perforce\my-project\3rdParty\Libusb\OriginalSource\libusb-1.0.19\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition
3>  C:\Program Files (x86)\Windows Kits\10\Include\ucrt\time.h(39): note: see declaration of 'timespec'
3>  descriptor.c
1>D:\Perforce\my-project\3rdParty\Libusb\OriginalSource\libusb-1.0.19\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition
1>  C:\Program Files (x86)\Windows Kits\10\Include\ucrt\time.h(39): note: see declaration of 'timespec'
1>  hotplug.c
3>D:\Perforce\my-project\3rdParty\Libusb\OriginalSource\libusb-1.0.19\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition
3>  C:\Program Files (x86)\Windows Kits\10\Include\ucrt\time.h(39): note: see declaration of 'timespec'
3>  hotplug.c
2>  getopt_2015.vcxproj -> D:\Perforce\my-project\3rdParty\Libusb\OriginalSource\libusb-1.0.19\msvc\..\Win32\Debug\lib\getopt.lib
1>D:\Perforce\my-project\3rdParty\Libusb\OriginalSource\libusb-1.0.19\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition
1>  C:\Program Files (x86)\Windows Kits\10\Include\ucrt\time.h(39): note: see declaration of 'timespec'
1>  io.c
3>D:\Perforce\my-project\3rdParty\Libusb\OriginalSource\libusb-1.0.19\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition
3>  C:\Program Files (x86)\Windows Kits\10\Include\ucrt\time.h(39): note: see declaration of 'timespec'
3>  io.c
1>D:\Perforce\my-project\3rdParty\Libusb\OriginalSource\libusb-1.0.19\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition
1>  C:\Program Files (x86)\Windows Kits\10\Include\ucrt\time.h(39): note: see declaration of 'timespec'
1>..\libusb\io.c(1224): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data
1>..\libusb\io.c(1902): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data
1>..\libusb\io.c(2423): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data
1>  poll_windows.c
3>D:\Perforce\my-project\3rdParty\Libusb\OriginalSource\libusb-1.0.19\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition
3>  C:\Program Files (x86)\Windows Kits\10\Include\ucrt\time.h(39): note: see declaration of 'timespec'
3>..\libusb\io.c(1224): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data
3>..\libusb\io.c(1902): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data
3>..\libusb\io.c(2423): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data
3>  poll_windows.c
1>D:\Perforce\my-project\3rdParty\Libusb\OriginalSource\libusb-1.0.19\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition
1>  C:\Program Files (x86)\Windows Kits\10\Include\ucrt\time.h(39): note: see declaration of 'timespec'
…..   These errors continue …...

We are utilizing VS2015 because of support in it for C++11.  We’ve had problems in the past with linking components built with different VS versions together, so would like to use VS2015 for everything (including libraries such as this).  But for now we’ll have to use a VS2013 built version.

deisenhut, 2015-04-14:
I have created 2015 VS project files (and a corresponding 2015 solution file) that include the preprocessor definition of HAVE_STRUCT_TIMESPEC.  With this change, the build succeeds.  There are a couple more compile warnings with VS2015 (C4244: '=': conversion from 'time_t' to 'long', possible loss of data), but it successfully builds.

Zangetsu38, 2015-06-11:
@deisenhut How you did to solve the problem exactly?

daneisenhut, 2015-06-13:
In each of the project files, I included HAVE_STRUCT_TIMESPEC within the PreprocessorDefinitions tag.  Each project file had four instances of this tag that I modified.

```
<PreprocessorDefinitions>HAVE_STRUCT_TIMESPEC;WIN32;__GNU_LIBRARY__;_DEBUG;_CONSOLE;_CRT_SECURE_NO_WARNINGS;%(PreprocessorDefinitions)</PreprocessorDefinitions>
```

Zangetsu38, 2015-06-27:
Ok thx :)

dickens, 2016-01-06:
Issue resolved via a34ba28

closed 2016-01-06 by dickens