libusb/libusb #70: Problem of using libusb with android device on windows

RiceeeChang, 2015-05-15:


mcuee, 2015-05-15:
If Zadig does not work for your device, then it is a Zadig problem. Make
sure you use the latest version of Zadig,

You can also try libusbK's driver installer to see if that helps.
   http://sourceforge.net/projects/libusbk/files/libusbK-release/3.0.7.0/

Anyway, this is not a libusb problem. You will have to use a supported
driver.

diabolo38, 2015-05-15:
what is your device exactly ? does it appear in teh con,trol pannel wih a question mark on it (no drievr for it)? 

if ther's already a driver for it you can't setup a new one without De-install the first 
CDC serial and Mass storage or media may fall in that case.
An amternative is to use filter driver and de-attach the kernel driver first i(i never used that so far).

if your device is composite (multiple interface) be sure to select the one you want to use with libusb 
you may ask zadig in advanced mode to extract without installing,  add or change the interfaces
in inf file

[LUsbK_DeviceGroup]
%DeviceName%=LUsbK_Device, USB\%DeviceID%, USB\%DeviceID%&MI_00, USB\%DeviceID%&MI_01

in that example i added interface 0 and 1 when only base device was set in Zadig

RiceeeChang, 2015-05-15:
I downloaded the latest version on the official website, Updated 2015.04.30: Zadig for Windows Vista or later (5.0 MB).

My Device is Android Phone(Xiaomi2, Samsung S4, HTC Desire820, Nexus7...)
And this is my revised inf file. https://dl.dropboxusercontent.com/u/10481816/libusb_device_mi2_custom.inf

And I use "libusbK-3.0.7.0-setup.exe" to create inf file to install libusbk driver. It still can used by libusb

diabolo38, 2015-05-15:
without more detail of what you do and with what exact device it tough to help
what libusb example listdevs show ?
what libusb call is returning -12 ? 
Most smartphone have full feature usb driver (modem disk access etc...) and do not require any libusb support .

mcuee, 2015-08-17:
Closing this issue since libusbK installer works. If this is a Zadig issue, it can be sorted out in Zadig. This is not a libusb problem.

closed 2015-08-17 by mcuee