libusb/libusb #72: undefined reference to min on Windows with -DNOMINMAX

Timmmm, 2015-06-03:
When compiling the library under mingw I get the following error:

    ....libusb-master\libusb\os\windows_usb.c:219: error: undefined reference to `min'

Which is this line of code:

	safe_strcpy(&ret_path[add_root], size-add_root, path);

Which uses this macro in `windows_common.h`

      #define safe_min(a, b) min((size_t)(a), (size_t)(b))

I expect this is expected to be defined from `windows.h`, but it isn't if you #define NOMINMAX which I have to on my project because of a different library. I suggest you just implement the macro yourself, something like this:

     #define safe_min(a, b) (((size_t)(a) < (size_t)(b)) ? (a) : (b))

This will safe people faff and is far safer - `#define min ..` is an awful awful thing. You also have to change line 1932 of windows_usb.c to:

	size = safe_min(config_header->wTotalLength, len);

Timmmm, 2015-06-03:
I created [this pull request for it](https://github.com/libusb/libusb/pull/73).

Decided to use already-written MIN macro rather than writing it out.

dickens, 2015-07-29:
Thanks for the pull request. I just brought this in.

commit d473d0b2 (2015-07-29): Use locally defined MIN macro instead of min which is (possibly) defined in windows.h

Some users have to define NOMINMAX which means that windows.h doesn't define min() or max(). Instead of relying on it it is better to use a locally defined function / macro. Fortunately libusbi.h already defines MIN and MAX so we can just use those.

* Closes #72
* Closes #73
closed 2015-07-29 by dickens