libusb/libusb #78: libusb provides duplicate entries

ginkoob, 2015-07-05:
I have two usb devices connected to my computer. lsusb shows a duplicate ID instead of individual entries 

Bus 001 Device 039: ID 19d2:1405 ZTE WCDMA Technologies MSM 
Bus 001 Device 033: ID 19d2:1405 ZTE WCDMA Technologies MSM 

---- 
Tested wotj lsusb (usbutils) 007

PhilOsIvan, 2015-07-13:
Why ID sould be different if this is the same devices?
If difference is only in device path then you will need use libusb_get_port_numbers.

dickens, 2015-07-28:
The list shows two devices on the same bus with different addresses, and by all appearances the two devices are the same. This is to be expected if you indeed have two of the same device connected to the host. Are you saying that two *different* devices are being reported with the same VID/PID and descriptor strings?

ginkoob, 2015-08-01:
@Chris yes, both have same VID/PID. It's a common practice on ZTE, huawei
devices.

dickens, 2015-08-01:
On Saturday, August 1, 2015, ginkoob <notifications@github.com> wrote:

> @Chris yes, both have same VID/PID. It's a common practice on ZTE, huawei
> devices.

Then what exactly is the issue? This is the expected result.

dickens, 2015-08-02:
I think there is a misunderstanding of the lsusb output. The "ID" listed here is not something unique generated by libusb, it is <VID>:<PID> as reported by the device in the device descriptor. Therefore I am closing this issue.

closed 2015-08-02 by dickens