libusb/libusb #79: Protect against spurious wakeups

elfring, 2015-07-19:
I have seen that [no loop is used around a call](https://github.com/libusb/libusb/blob/9b2c8abf134b96b6c1798615e0ed17991b8c0692/libusb/io.c#L1887 "Check libusb_wait_for_event()") of the function "[usbi_cond_wait](http://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_cond_wait.html "wait on a condition")".
Would you like to reuse anything from my message on the topic "[spurious wakeup](http://groups.google.de/group/comp.programming.threads/msg/bb8299804652fdd7 "Discussion about handling of spurious wakeups")"?

mcuee, 2016-12-02:
Close this aging request. Please re-open if you think it is wrongly closed.

closed 2016-12-02 by mcuee

elfring, 2016-12-02:
Unfortunately, I can not reopen this issue myself here directly.

How do you think about to clarify [the affected implementation details](https://github.com/libusb/libusb/blob/a90331a3be0a7919dfd256d700c38cfad5718575/libusb/io.c#L1965 "Update candidate: libusb_wait_for_event()") further?