libusb/libusb #84: Must wait for all control transfers to finish before calling libusb_close (OS X)

pereckerdal, 2015-08-06:
If there are outstanding control transfers when calling `libusb_close`, `do_close` NULLs out the `dev_handle` field of those transfers. Also, when control transfers finish, `darwin_async_io_callback` will dereference that pointer.

The result of this is that if `libusb_close` is called when there is an outstanding control transfer, the program crashes. In practice this means it's impossible to do fire and forget control transfers with `LIBUSB_TRANSFER_FREE_BUFFER | LIBUSB_TRANSFER_FREE_TRANSFER` and no callback, because then you can not know that it is safe to call `libusb_close`.

For now, I'm able to work around this by manually counting outstanding transfers but it is cumbersome and seems error prone.

pereckerdal, 2015-08-06:
By the way this is tested with v1.0.19

diabolo38, 2015-08-06:
Similar issue was reported for Linux on the libusb mail list recently. Have a look

similar solution could apply

By yhe way Is it the lib responsible for keeping track and canceling all user assync i/o?

that would mean close could be blocking even fail or take considerable  time.

pereckerdal, 2015-08-06:
Thanks for the response.

What happens today is that the lib NULLs out an internal pointer and then dereferences it. It doesn't have to cancel all i/o, it just needs to keep track of when it has invalidated itself and not crash when callbacks that only library internals have control over are invoked.

pereckerdal, 2015-08-06:
Which libusb mailing list thread are you referring to? What's the subject? I can't find anything about it.

dickens, 2015-08-06:
Closing a device implies that the application is done using it. The "fire
and forget" concept is not supported by libusb--it is the application's
responsibility to manage transfers.

libusb_close() is documented as a non-blocking call, so waiting for
transfers to complete is not an option. If you have the log level set to at
least LIBUSB_LOG_LEVEL_WARNING, libusb will happily inform when an
application misbehaves and leaves outstanding transfers when closing the
device.

It's more complicated than just leaving pointers there or checking these
pointers when transfers complete. This will just lead to race conditions,
memory leaks, etc.

diabolo38, 2015-08-06:
oops ... it was an issue with canceled transfer and time out  reporting so not so similar 

http://libusb.6.n5.nabble.com/libusb-PATCH-0-2-Fix-reporting-of-cancellations-due-to-timeouts-tc5714939.html

pereckerdal, 2015-08-06:
@diabolo38 ok. Thanks for the pointer anyway :smile: 

@dickens I see. Thank you for your response.

I think it might be nice to add this information to the documentation of `libusb_close`. I can't see anything there that says that all outstanding requests must be finished before it is called. That it's non-blocking might imply that transfers aren't cancelled from the perspective of the library user, but I don't think that's obvious at all.

How would you recommend handling this? As I understand how libusb works today, the only safe way of using libusb is to (implicitly or explicitly) maintain a counter of outstanding transfers by incrementing it after every successful call to `libusb_submit_transfer` and decrementing it in every transfer callback. The libusb runloop must then be pumped until the outstanding counter is 0, and not until then can `libusb_close` be called.

I'll close this ticket now since this is clearly by design. I do think that there is an opportunity to improve the library here though. One library that could serve as a good example here is libuv. It's `uv_loop_close` simply returns an error code if there are outstanding things on the event loop. That way the user can pump it until it's empty, instead of being forced to keep an external record of all the outstanding operations that must mirror the hidden internal state.

closed 2015-08-06 by pereckerdal