libusb/libusb #86: libusb Windows hotplug support

mcuee, 2015-08-10:
This is one of the long standing missing feature for Windows.

mcuee, 2015-11-27:
Last known tree back in the libusbx time.
https://github.com/dickens/libusbx-hp/tree/windows-hotplug-3

changyp6, 2016-03-02:
libusb hotplug for Windows is really important.
I have a program running on Linux/Window/MacOS, but Windows doesn't have hotplug support, I have to write my own device detection code for Windows.
But Windows only reports that there is a device arrived or there's a device removed, you have to find which one is added or removed.
So when I receive device remove event, I usb libusb_get_device_list() to get current device present in the system, and compare this list with the old one to find out which one is removed, and I use the same method to find which one is added, after I find the difference, my program will handle the removed/added device.

But if I unplug the device and plug it back to the same USB port very quickly, and Windows's driver searching  program is disabled(to make system recognize device faster), my program can get device remove and device add event correctly, but libusb_get_device_list() will get a usb device list the same as the previous saved one.

Here is the reason:
When my program receives device remove event, and start to list all the USB devices, because the unplug/plug action is so fast, libusb_get_device_list() will give a list after the device is plugged back. So this USB device list is the same as the list before unplug/plug action, which means my program cannot find out which device is removed and which is added, there will be 2 missing actions for my program.

Because my program is event driven, this causes 2 events missing.

I really hope that hotplug feature can be implemented for Windows.

mcuee, 2016-03-03:
https://github.com/dickens/libusbx-hp/tree/windows-hotplug-3

You may want to try the above old branch to see the hotplug code there
works for your application. I am not so sure how much efforts need to bring
that branch up to date though.

changyp6, 2016-03-03:
I tried that before, I even tried to merge the change into libusb 1.0.19, but I failed to merge it to newer code. I still need to use latest branch, because there are tons of bug/leak fixes for Windows platform.

Hope Windows hotplug support can be implemented soon.

dickens, 2016-03-03:
Will be part of the next release. Unfortunately it is a complicated set of
changes.

Regards,
Chris

changyp6, 2016-03-03:
This is a great news!

chrisba77, 2016-09-02:
were there any changes regarding the hotplug issue since then? Currently having the problem with LibUsbDotNet that communication with an unplugged-replugged device is only possible again after reinstalling the driver using the Inf-Wizard OR restarting Windows. This device is a CardReader. On the other hand, i dont have issues hotplugging a USB Keyboard. Has anyone suggestions?

diabolo38, 2016-09-03:
@chrisba77  strange w/o any "hotplug" support you should be able to re-use the device without driver re-install reboot etc 
if:
-release any  claimed interface 
-close the device 
-call exit on the libsusb context 

then run again lib usb init , device detect ,re-open re claim etc ..

That is what i do to get control again of unplugged and re-pluged device 
not  real nice but the simplest i did find.

chrisba77, 2016-09-05:
@diabolo38 I do all this. And actually after plugging in the device again, communication is possible. BUT the read values coming from the device are not valid anymore.
After reading in a Tag / Card the device sends some HID reports with Number-Keys including a Return Hit. After unplug / replug, these values seem to be random. Checked with Port Sniffer that I read the right values coming from the device. But sent Key Hits are not valid anymore and change every time i read in the same Tag. Reconnecting or Restarting my Test-SW doesnt help. Only thing that helps is rerunning Inf-Wizard or a reboot.

diabolo38, 2016-09-05:
@chrisba77  if device is hid  using standard os hid device driver then the issue could be os driver or the device itlsef.  I can't see such issue with non hid device at least.

tony--, 2017-01-05:
It looks like this didn't make it into 1.0.21 [as planned](https://github.com/libusb/libusb/issues/86#issuecomment-191559150).
Any plans to include this in 1.0.22?
Thanks in advance!

Shift34, 2017-09-13:
Hi guys, what's the status of this issue ?

leileigong, 2017-09-25:
Did the hotplug supported on windows?

jazzy-crane, 2017-11-15:
Has this now been moved to 1.0.23? :(

changyp6, 2018-01-02:
I just checked the latest code, it seems that windows hotplug hasn't been implemented.
Windows libusb hotplug definitely is my first 2018 new year wish.

njk42, 2018-01-05:
+1 for Windows hotplug support.

Without hotplug support, what are the standard ways of determining when a device connects or disconnects on Windows?  Does one just repeatedly poll a device list and determine when there are changes detected?

ryxuli, 2018-01-11:
It's really inconvenient that libusb doesn't support hotplug capability on Windows. I finely turn to libusbK. It can be downloaded from https://sourceforge.net/projects/libusbk/, after setting up, you can download the whole src zip in https://sourceforge.net/projects/libusbk/files/libusbK-release/3.0.7.0/libusbK-3.0.7.0-src.zip/download which includes an example hot-plug-monitor.c shows the hotplug ability.

rjgc2015, 2018-03-20:
Hi all, did it support on Windows now？

mcuee, 2018-03-20:
Unfortunately not. Right now only Chris is working on the Windows side and
it will take some time to get this.
-- 
Xiaofan

Timmmm, 2019-01-30:
Is anybody working on this?

changyp6, 2019-05-21:
Now, this Hotplug has become my 2019 new year wish.

wanpp, 2019-06-06:
Now, this Hotplug has become my 2019 new year wish.

huigzi, 2019-06-14:
Now, this Hotplug has become my 2019 new year wish.

X-Crack, 2019-08-03:
Now, this Hotplug has become my 2019 new year wish.

akallabeth, 2020-03-12:
Is anyone actively working on this?
I'd really be interested in this feature and can help out with testing and some coding.
Lacking the deep internal knowledge of the various pitfalls though, so likely some guidance required.

Timmmm, 2020-03-12:
I have implemented this before (not for libusb), although sadly I can't find the code now. :-/ As I recall, it's awkward on Windows because the way you detect device plugs and unplugs is to create a hidden window, register it somehow, and then Windows sends messages to your window.

akallabeth, 2020-03-12:
@Timmmm doing something similar for an application I´m working on, but no idea if that can be easily used for libusb internally?

a-v-s, 2020-03-12:
I have some implementation of this I use with libusb when it returns "not supported" on windows. Then I extract the vid, pid and serialnumber, and match it against the the result of libusb_get_device_list to get the corresponding libusb device. This means a unique serial number must be present, therefore this cannot be used as a generic implementation. 

I haven't looked into the internals of libusb. In order to make a generic implementation, one must be able to match device string in dbcc_name inside libusb directly.

akallabeth, 2020-03-12:
@a-v-s don't know too much about windows USB api, but is it not possible to extract `bus` and `address` from the `dbcc_name` (or some function that takes the path and returns the address)?

akallabeth, 2020-03-12:
@a-v-s should it not be sufficient to use the `dbcc_name` according to https://stackoverflow.com/questions/994774/acessing-dbcc-name-from-dev-broadcast-deviceinterface-win32-programming to get these?

a-v-s, 2020-03-12:
I'm mainly a firmware developer. I'm not that familiar with windows internals. I've run an example in my virtual machine. I've got:

` "\\?\USB#VID_DEAD&PID_BEEF#5075BC12#{a5dcbf10-6530-11d2-901f-00c04fb951ed}" `

Where my VID=DEAD, PID=BEEF, SERIAL=5075BC12 and that GUID is GUID_DEVINTERFACE_USB_DEVICE.

So, this ain't enough, and won't work without unique serial number.

akallabeth, 2020-03-12:
@a-v-s as far as I can see this is only one form of `path` that can be returned, see https://www.silabs.com/community/interface/knowledge-base.entry.html/2013/11/21/windows_usb_devicep-aGxD (no idea how credible, but sounds enough like what I´ve seen so far)

Timmmm, 2020-03-12:
> @Timmmm doing something similar for an application I´m working on, but no idea if that can be easily used for libusb internally?
> 
> edit: you mean this: https://docs.microsoft.com/en-us/windows/win32/devio/registering-for-device-notification
> just with `CreateWindowEx(0, CLASS_NAME, INST_NAME, 0,0,0,0,0,HWND_MESSAGE, NULL, NULL, userdata)`

Yes I think that was it! Really annoying that I've lost the code. If you google `WM_DEVICECHANGE USB` you get a load of helpful results, e.g. [this application note](https://www.ftdichip.com/Support/Documents/AppNotes/AN_152_Detecting_USB_%20Device_Insertion_and_Removal.pdf).

dickens, 2020-03-13:
Yes, I am actively working on this.  I think many people oversimplify the caveats and hidden gotchas of hotplug support for Windows.

akallabeth, 2020-03-13:
@dickens thank you for that. After having a look at the available API I can understand that is not an easy task.

maltfield, 2020-06-26:
Does anyone know of *any* solution at all for getting a python function to be called on a hotplug event in Windows?

I'm happy to hear that this is being actively worked on, but it's been open for over 5 years. Is there any other solution besides libusb to register a python callback function to be called when a usb device is added or removed?

bparker06, 2020-06-26:
@maltfield Yes this has been possible since at least Windows 9x using RegisterDeviceNotification and looking for WM_DEVICECHANGE message:

https://docs.microsoft.com/en-us/windows/win32/devio/registering-for-device-notification

maltfield, 2020-06-27:
> @maltfield Yes this has been possible since at least Windows 9x using RegisterDeviceNotification and looking for WM_DEVICECHANGE message:
> 
> https://docs.microsoft.com/en-us/windows/win32/devio/registering-for-device-notification

Thank you! In case anybody else ends-up here looking for ways to register a callback function on usb hotplugging events in Windows, I succeeded in doing this on Windows 10 using python, ctypes, and the pypiwin32 module.

 * https://pypi.org/project/WMI/

My code can be found here:

 * https://github.com/BusKill/buskill-app/blob/6c84587b25f7ef3b5300d8c0eda354b11051215e/src/buskill.py#L212-L275

Which is largely based on the code found at these links:

 * http://timgolden.me.uk/python/win32_how_do_i/detect-device-insertion.html
 * https://stackoverflow.com/questions/38689090/detect-media-insertion-on-windows-in-python

mcuee, 2020-07-29:
Reference:
https://github.com/libusb/libusb/pull/617

coffeenotfound, 2020-12-03:
Kinda a massive deal breaker for a cross platform usb library to not actually be cross platform. Having to manually install a custom device driver using zadig and now this... Come on

whitequark, 2020-12-03:
> Having to manually install a custom device driver using zadig

To the extent that this is necessary, it is purely a result of Microsoft's policies. (However, libusb can use the WinUSB backend, and devices that opt into it with a special descriptor do not require using Zadig.)

Timmmm, 2020-12-04:
> To the extent that this is necessary, it is purely a result of Microsoft's policies.

True, though it would be nice if libusb integrated libzadig so it could provide an easy way to install the WinUSB drivers if necessary. It could do the same for the udev rules you have to set up on Linux.

Admittedly that would require some new APIs, design work, etc. Given that we still don't have hotplug on Windows after all these years - and I don't recall it being too difficult to implement - I won't hold my breath!

dickens, 2020-12-04:
> Given that we still don't have hotplug on Windows after all these years - and I don't recall it being too difficult to implement - I won't hold my breath!

Registering for notifications is trivial--it is fitting those notifications into the library that is "difficult".  The whole situation is complicated by the fact that each interface of a composite device has its own driver and associated device interface path.  The bookkeeping of mapping the individual interfaces to libusb's representation of a device while also maintaining the topology of the USB bus is why this is not in yet.

thegecko, 2021-01-13:
> mcuee modified the milestones: 1.0.25, 1.0.26 12 hours ago

Shame this has been bumped again.

mcuee, 2021-01-14:
No I have not changed the target date of the feature, rather I have added a bug-fix 1.0.25 release first. 

But anyway, just for info, the milestone target date is not really binding, it is just a quick estimation. The projects are running by volunteers so it will always be dependent on the time availability of the developer.

mcuee, 2021-01-14:
Just FYI on the recent libusb release:
libusb-1.0.22: Mar 25, 2018
libusb-1.0.23: Aug 29, 2019
libusb-1.0.24: Dec 11, 2020

mcuee, 2021-01-14:
@Timmmm BTW, the idea of integrate libwdi was discussed and the conclusion is that it is not part of libusb Windows backend. Anyway now libwdi and Zadig are pretty mature and alive with Pete Batard (main libusb Windows developer before Chris) in charge (even though he has more focus on the very popular rufus project).

mcuee, 2021-03-01:
Basically we need people to help on Windows side as discussed in other issues. Windows backend is a complicated. Chris has already done a great job maintaining the Windows backend after Pete left. And he also deals with Linux side. So hopefully some people will come and help on the Windows side.

yyk123, 2021-03-29:
for me, on Windows 10, I use one thread to run monitor and find device every 100ms. and device_attach_process/device_dettach_process function would be invoked in hotplug callback function when on linux  platform
in device_attach_process/device_dettach_process , we can register user callback function.
monitor()
{
while (running)
	{
		if (isSupportHotplug)
		{
			libusb_handle_events_completed(context, NULL);
		}
		else
		{
			if (find_quiet() == USB3_SUCCESS)
			{
				device_attach_process(this);
			}
			else
			{
				device_dettach_process(this);
			}
		}	
		sleep_ms(100);
	}
}

int LIBUSB_CALL hotplug_callback(struct libusb_context* ctx, struct libusb_device* dev, libusb_hotplug_event event, void* user_data)
{
	MyClass* class_ptr = (MyClass*)user_data;
	switch (event)
	{
	case LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED:
		class_ptr->device_attach_process(class_ptr);
		break;
	case LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT:
		class_ptr->device_dettach_process(class_ptr);
		break;
	default:
		log::logger(verbose::error, "Unhandled usb event.");
		break;
	}
	return 0;
}

hli-luna, 2021-05-15:
> Registering for notifications is trivial--it is fitting those notifications into the library that is "difficult". The whole situation is complicated by the fact that each interface of a composite [...]

Is that the sticking point? The translation from one "interface"/"representation" to another?

mcuee, 2022-01-22:
Unfortunately Chris seems to be busy and has not been active in libusb development for a while. So I have to remove this from libusb 1.0.26 milestones unless he is back or a new Windows developer can join the project and look at this feature.

mcuee, 2022-01-25:
Ref: https://github.com/libusb/libusb/issues/1047#issuecomment-1019279948
From Tormod:
> For something like the Windows hotplug support we must be sure there are developers around that will stay available to timely fix any issues that arises after it lands. Anyway, until there is at least a draft of a pull request for it, it remains just a wishlist item with no relevance for our release milestones.

douglasg14b, 2022-02-08:
Is this why Talon is working for me?

> Activating Microphone: "Microphone (MH670)"
> Falling back to polling for USB hotplug: failed to register hotplug callback

Searching for this lead me here.

mcuee, 2022-02-08:
> Is this why Talon is working for me?
> 
> > Activating Microphone: "Microphone (MH670)"
> > Falling back to polling for USB hotplug: failed to register hotplug callback
> 
> Searching for this lead me here.

Please provide more details. What is your question? Do you have some issues with regard to Talon, then you have to get support from them.
Ref: https://github.com/talonvoice/talon/issues

If Talon is using libusb, then they have to fall back to polling for USB hotplug under Windows, since libusb Windows does not support hotplug yet.

Aii-Wei, 2022-05-31:
Polling for USB hotplug under Windows,  Will it affect the  device communication? I tested and found that the device change can be detected according to idVendor, idProduct, port from libusb_get_port_number, address from libusb_get_device_address.

xhwang-China, 2023-09-14:
unfortunately! 2023 is coming to an end and this feature is still not supported

mcuee, 2023-09-14:
> unfortunately! 2023 is coming to an end and this feature is still not supported

This feature will not be supported, unless we get an active Windows developer for libusb project who can actively working on this feature. 

You want to volunteer?

whitequark, 2023-09-14:
It is really unfortunate that the payoff from having hotplug (not needing to do like a loop or something) is absolutely not worth the person-weeks of work of someone highly competent to get it in upstream libusb. I know I cannot justify it for any of my projects, and at least one would decently benefit from having hotplug. Just not enough.

mcuee, 2023-09-17:
Contributors are more than welcome for libusb projects now that we are short of developers and reviewers which makes 1.0.27 release not happening yet.

* https://github.com/libusb/libusb/issues/947
* https://github.com/libusb/libusb/issues/1094

As for this paticular issue (Windows Hotplug support), it may be even more difficult as we do not have an active Windows developer for quite some time (last commit done by Chris Dickens was on 9-Feb-2021).

xhwang-China, 2023-09-19:
> > unfortunately! 2023 is coming to an end and this feature is still not supported
> 
> This feature will not be supported, unless we get an active Windows developer for libusb project who can actively working on this feature.
> 
> You want to volunteer? PRs are welcome.

My current solution is to listen for WM_DEVICECHANGE events, but this solution is not a good idea for cross-platform library because it requires binding Windows and you have to create a virtual window to do this.

Finally, I added a usb_event_cb api to the libusb library to be called by the upper layer to handle device hotplug，Device monitoring is implemented by the upper layer

This is an approach that hopefully will help other developers, although it's not a perfect solution.

[11 more replies]

closed 2025-08-26 by sonatique