libusb/libusb #88: [QUESTION] libusb_reset_device always returns -99

reverendhomer, 2015-08-11:
Hi all!

I have a small embedded linux with one USB port ant without any hal or udev. I need to avoid the emergence of `/dev/sdc` device, if user plugged and mounted the device and unplugged **without** unmounting after them. To do this, Ive' found one nice program using libusb_reset_device. Here it is:
``` C
#include <stdio.h>
#include <stdlib.h>
#include <libusb-1.0/libusb.h>

struct libusb_device **devices;

struct libusb_device* search_device(int bus, int dev) { 
    int i = 0;
    struct libusb_device *device = devices[i];
    int busNum, devNum;

    while (device != NULL) {
        busNum =(int) libusb_get_bus_number(device);
        devNum =(int) libusb_get_device_address(device);
        printf("bus number: %d; device number: %d\n", busNum, devNum);
        if (busNum == bus && devNum == dev) {
            printf("found device\n");
            return device;
        }
        device = devices[++i];
    }
    return NULL;

}

int main(int argc, char **argv) {

    if (argc < 3) {
        printf("not enough arguments!\n");
        printf("usage: ./usbreset <bus number> <dev number>\n");
        return 0;
    }

    struct libusb_device *dev;
    struct libusb_device_handle *devh;
    int busnum = atoi(argv[1]);
    int devnum = atoi(argv[2]);
    printf("searching for bus number = %d, dev number = %d\n", busnum, devnum);

    int r;
    r = libusb_init(NULL);
    if (r < 0) {
        printf("libusb init failed!\n");
        return 1;
    }
    r = libusb_get_device_list(NULL, &devices);
    if (r < 0) {
        printf("libusb_get_device_list failed!\n");
        return 1;
    }
    dev = search_device(busnum, devnum);
    if (dev == NULL) {
        printf("device not found\n");
        return 0;
    }

    r = libusb_open(dev, &devh);    
    if (r == 0)
        printf("got the usb handle successfully.\n");
    else {
        printf("error getting usb handle.\n");
        return 1;
    }

    //reset the usb device
    r = libusb_reset_device(devh);
    printf("libusb_reset_device returned %d\n", r);

    //free the device list
    libusb_free_device_list(devices, 1);
    libusb_exit(NULL);
    return 0;
}
```
And that program works just fine! `/dev/sdc` turns into `/dev/sdb` as expected, but calling libusb_reset_device to bus number 1 and device number 1 always returns -99. 
What am I doing wrong?

diabolo38, 2015-08-11:
What is this device?  If some hub then it may be normal

not that you can identity mass storagedevice by looking at class code that will avoid reset of device that will not need it. 

A look at log could also help identifying last usb connected device

mcuee, 2015-08-26:
Please use mailing list for technical support. Closing this ticket.

closed 2015-08-26 by mcuee