libusb/libusb #90: libusb_kernel_driver_active() and _detach() are racy

mcuee, 2015-08-17:
Ref:  http://libusb.org/ticket/125

"Alan is correct in stating that fixing this race needs a kernel change. Which is why I've written a kernel patch to allow to do these things in a race free manner, see:
​http://permalink.gmane.org/gmane.linux.usb.general/69725
This patch is upstream now, but libusb still needs to be modified to use it. This requires adding a new API for this, as the old API is inherently racy.
I've this on my todo list. I hope to be able to work on this soonish.
Regards,
Hans"

dickens, 2015-09-07:
I think Hans has done this via the following two commits:

d4e993192aa81a1867bddf90d53cf750ec21e7b1
fc51f19341af70508b94c06ff763842b396ea72e

Or is there something else to be done?

mcuee, 2015-09-07:
I guess this has been done and we can close it.

jwrdegoede, 2015-09-07:
Ack, closing.

closed 2015-09-07 by jwrdegoede