libusb/libusb #94: Windows: libusb0.sys support improvement

mcuee, 2015-08-17:
The current support of libusb-win32 kernel driver libusb0.sys through libusbK.dll is not good.

For example, there is a warning about control transfer. It may not be a libusb problem though.

````c
Mcuee@Win8X64VM MINGW64 /c/work/libusb/msys64/libusb
$ ./examples/xusb.exe 0403:cff8
Using libusb v1.0.20.10997

Opening device 0403:CFF8...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:CFF8
         bcdDevice: 0700
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 2
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 00
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 83
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 04
           max packet size: 0200
          polling interval: 00

Claiming interface 0...

Claiming interface 1...

Reading string descriptors:
   String (0x01): "Amontec"
   String (0x02): "Amontec JTAGkey-2"
   String (0x03): "53T9XDR4"

Releasing interface 0...
Releasing interface 1...
Closing device...
libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)
libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
````

mcuee, 2015-08-17:
This will be a low priority as WinUSB will be the main driver supported and the support of libusb0.sys is considered as beta.

fakedrake, 2015-10-14:
I encountered issues with libusb0 support through chrome, when trying to use the USB javascript api to communicate with arduino programmers. It looks like libusb0 is the main driver used for USBASP and USBTinyISP devices, which means the leading ones. Here is a detailed description of the encountered issue, and please correct any misunderstandings.

libusb doesn't *actually* use the libusb0 filter driver. It rather just [loads the functions of WinUSB](https://github.com/libusb/libusb/blob/master/libusb/os/windows_usb.c#L2678L2691). [Apparently](https://social.msdn.microsoft.com/Forums/en-US/be7f9aa0-1694-42fe-932a-c49172377c45/winusbinitialize-fails-for-error-87-invalid-parameter?forum=windowssdk) when winusb tries to initialize a device that is configured with linusb0 the [undocumented](https://msdn.microsoft.com/en-us/library/windows/hardware/ff540277(v=vs.85).aspx?f=255&MSPPError=-2147217396) `ERROR_INVALID_PARAMETER (87)` gets raised.

The device initialization does not happen on [device open](https://github.com/libusb/libusb/blob/master/libusb/os/windows_usb.c#L2750L2783) but rather on [auto interface claim](https://github.com/libusb/libusb/blob/master/libusb/os/windows_usb.c#L3078) during the first control tranfer. Claiming the interfaces by hand doesn't help because we fail on our attempt to claim interface 0 [which is a prerequisite for claiming other interfaces](https://github.com/libusb/libusb/blob/master/libusb/os/windows_usb.c#L2899L2901)

mcuee, 2015-10-16:
I do not think usbasp and usbtinyisp uses libusb Windows (libusb-1.0 API),
rather they use libusb-win32 (enhanced libusb-0.1 API) which requires
libusb0.sys driver.

On the other hand, usbasp and usbtinyisp can probably use libusb-compat-0.1
as well to replace libusb-win32. In that case, just use WinUSB driver to
replace the existing libusb0.sys driver to have a better support.

Take note libusb-compat-0.1 does not officially support Windows right now.

fakedrake, 2015-10-20:
That is true, but windows install libusb0 (ie libusb-win32) by default.
That works fine with avrdude but not so well with chrome, backed by libusb.
An interesting point is that once one has switched to WinUSB or libusbK,
gaining access to their device via libusb, switching back to libusb0 does
not break the API. My best guess is that once WinUSB gains access to a
device, it can access it forever, regardless of the assigned driver.

The issue about all this is that while one *can* make their device work,
certain knowledge of the libusb library/driver stack is required. Higher
level API developers should not have to worry about these kinds of things,
let alone end users.

mcuee, 2015-10-21:
I do not quite understand what do you mean it works with avrdude and not
well with Chrome. What do you mean by Chrome here? Chrome OS? That is Linux
and has nothing to do with libusb0.

As for the driver, there is no such thing as Windows install libusb-win32
driver by default. It depends on the driver provided by the user first time
the device is plugged-in. If the device is supposed to be using
libusb-win32, the user needs to provide the libusb-win32 based driver
package. If the device is supposed to be using libusb Windows (or the
unsupported libusb-compat-0.1 under Windows), the WinUSB is the way to go.

Zadig is the preferred tool to install the driver. Alternatively, you can
write a driver installer using libwdi.

It seems to be a hassle but that is the reality for libusb under Windows.
If you want to write a program based on libusb under Windows, you have to
deal with Windows driver.

>
>

mcuee, 2018-04-20:
`c:\work\libusb\bin-1.0.22\examples\bin64>xusb 0403:6001
Using libusb v1.0.22.11312

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)
no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"
libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.

Releasing interface 0...
Closing device...

c:\work\libusb\bin-1.0.22\examples\bin64>xusb 0403:6001
Using libusb v1.0.22.11312

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"

Releasing interface 0...
Closing device...

c:\work\libusb\bin-1.0.22\examples\bin64>xusb 0403:6001
Using libusb v1.0.22.11312

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"

Releasing interface 0...
Closing device...

c:\work\libusb\bin-1.0.22\examples\bin64>xusb 0403:6001
Using libusb v1.0.22.11312

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"

Releasing interface 0...
Closing device...`

mcuee, 2018-04-20:
Using libusb-win32 filter driver will produce many warning messages.

<pre>
c:\work\libusb\bin-1.0.22\examples\bin64>xusb 0403:6001
Using libusb v1.0.22.11312

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: libusb: error [winusbx_claim_interface] could not access interface 0: [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 1 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 2 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 3 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 4 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 5 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 6 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 7 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 8 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 9 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 10 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 11 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 12 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 13 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 14 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 15 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 16 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 17 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 18 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 19 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 20 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 21 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 22 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 23 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 24 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 25 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 26 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 27 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 28 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 29 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 30 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 31 with WinUSB): [1] Incorrect function.
libusb: error [auto_claim] could not auto-claim any interface
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-5)
no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Claiming interface 0...
libusb: error [winusbx_claim_interface] could not access interface 0: [1] Incorrect function.
   Failed.

Reading string descriptors:
[... 111 lines cut ...]
[... cut ...]
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 29 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 30 with WinUSB): [1] Incorrect function.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 31 with WinUSB): [1] Incorrect function.
libusb: error [auto_claim] could not auto-claim any interface

Releasing interface 0...
Closing device...
</pre>

mcuee, 2020-08-31:
This will be a low priority work. Please use other supported driver (WinUSB, libusbk or usbdk).

mcuee, 2021-03-31:
Ref:
https://github.com/libusb/libusb/issues/741

mcuee, 2021-07-24:
I have closed #741 since there are still use case of libusbk and even libusb0.sys.

mcuee, 2021-07-24:
With the merging of #844 and #912, libusbk (and probably libusb0.sys) are working better now. Still libusb0.sys should be the last choice if you want to use libusb-1.0 Windows.

mcuee, 2021-07-24:
1.0.24 release:
```

C:\work\libusb\libusb-1.0.24_bin> .\examples\bin64\xusb 0483:3748
Using libusb v1.0.24.11584

Opening device 0483:3748...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0483:3748
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)
no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 3
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00
       endpoint[2].address: 83
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
libusb: warning [handle_timeout] async cancel failed -5
```

mcuee, 2021-07-24:
Latest git head with libusb0.sys
```

C:\work\libusb\libusb\x64\Release [master ≡ +1 ~0 -0 !]> .\examples\xusb.exe 0483:3748
Using libusb v1.0.24.11644

Opening device 0483:3748...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0483:3748
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)
no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 3
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00
       endpoint[2].address: 83
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
libusb: warning [handle_timeout] async cancel failed -5
```

mcuee, 2021-07-24:
The device works well with WinUSB driver.
```

C:\work\libusb\libusb\x64\Release [master ≡ +1 ~0 -0 !]> .\examples\xusb.exe 0483:3748
Using libusb v1.0.24.11644

Opening device 0483:3748...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0483:3748
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 3
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00
       endpoint[2].address: 83
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
   String (0x01): "STMicroelectronics"
   String (0x02): "STM32 STLink"
   String (0x03): "I?s♠I?QQ7↨◄?"

Releasing interface 0...
Closing device...
```

mcuee, 2021-07-24:
1.0.23 release actually works a bit better with libusb0.sys.

```

C:\work\libusb\libusb-1.0.23_bin> .\examples\bin64\xusb.exe 0483:3748
Using libusb v1.0.23.11397

Opening device 0483:3748...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0483:3748
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)
no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 3
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00
       endpoint[2].address: 83
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "STMicroelectronics"
   String (0x02): "STM32 STLink"
   String (0x03): "I?s♠I?QQ7↨◄?"
libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.

Releasing interface 0...
Closing device...
```

mcuee, 2021-07-24:
https://github.com/libusb/libusb/issues/761 reading string descriptor is broken for libusb0.sys in 1.0.24 and latest git. So this is a regression from 1.0.23.

mcuee, 2021-07-24:
But anyway, it is not recommended to use libusb0.sys with libusb-1.0 Windows as of now.

mcuee, 2021-07-24:
Up to this commit, libusb0.sys support is not good but not that broken, https://github.com/libusb/libusb/commit/4261cbefc716e49d459426593cef0104482ec43b

```

C:\work\libusb\libusb_test\x64\Release [(4261cbe...)]> .\examples\xusb.exe 0483:3748
Using libusb v1.0.23.11523

Opening device 0483:3748...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0483:3748
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)
no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 3
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00
       endpoint[2].address: 83
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.

Releasing interface 0...
Closing device...

```

mcuee, 2021-07-24:
From this release https://github.com/libusb/libusb/commit/ba6b8bcb7ea204e65a3deec3be81aacc9f4b6d5a onwards, it is pretty much broken for libusb0.sys (also libusbk and hid but then they got fixed later).

```

C:\work\libusb\libusb_test\x64\Release [(ba6b8bc...)]> .\examples\xusb.exe 0483:3748
Using libusb v1.0.23.11531

Opening device 0483:3748...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0483:3748
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)
no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 3
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00
       endpoint[2].address: 83
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
```

mcuee, 2021-10-13:
Take note libusb0.sys 1.2.7.1 is not signed so it can not be used by normal users.

mcuee, 2021-10-13:
So all in all, I will discourage the use of libusb0.sys with libusb-1.0 Windows.

mcuee, 2021-10-18:
@tormodvolden 
> (I wish there was a testing infrastructure where I could submit a patch or build and have it tested against some real devices :) I have a Windows computer in the drawer, but it has only XP for which libusb recently dropped all support, so I cannot test myself.)

Maybe you can give this a try: 60 days Windows 10 virtual machine with Visual Studio 2019 and WSL/Ubuntu which you can use under Linux.
https://developer.microsoft.com/en-us/windows/downloads/virtual-machines/

tormodvolden, 2021-10-29:
@qmfrederik So do I understand correctly that most control transfers (getting descriptors etc) were working at the time, but not setting configuration? I just try to get a picture of what was working when, and at what point did it break.

@mcuee Thanks for the debug logs. I guess the BOS fetching fails in all cases because the device has none and probably replies with stalling the pipe. But with libusb0.sys it leaves something dangling that messes up later. Can you try commenting out the BOS fetching?

In the libusbK.sys log there is however one last failed XferCtrlComplete that I don't understand where comes from.

mcuee, 2021-10-30:
> > Did you use the snapshot Zadig to install the driver?
> 
> Yes, I did, the file zadig_git_99a38ae62ba0.exe It works fine on W10, but there is a trouble on W7.

You can check my prior comments, but there should be an easier work-around for Windows 7 if it does not work.
1) Using Zadig 2.6 release to install libusb0.sys 1.2.6.0 device driver.
2) Using the libusb-win32 1.2.7.2 installer to upgrade libusb0.sys and libusb0.dll

mcuee, 2021-10-30:
> EDIT: BTW your test, with libusb0.sys from 1.2.6.0 doesn't include Frederik's fixes. We'll need a signed debug version of 1.2.7.2...

@tormodvolden Let me check with Peter whether it is possible to get signed debug version or not.

mcuee, 2021-10-31:
> Thanks for the debug logs. I guess the BOS fetching fails in all cases because the device has none and probably replies with stalling the pipe. But with libusb0.sys it leaves something dangling that messes up later. Can you try commenting out the BOS fetching?

@tormodvolden 
After I commenting out, it still hangs even with your patch, with both libusb0.sys 1.2.6.0 and 1.2.7.2.

```
diff --git a/examples/xusb.c b/examples/xusb.c
index 61ce4dd..5dc0cd5 100644
--- a/examples/xusb.c
+++ b/examples/xusb.c
@@ -851,6 +851,7 @@ static int test_device(uint16_t vid, uint16_t pid)
        string_index[1] = dev_desc.iProduct;
        string_index[2] = dev_desc.iSerialNumber;

+       /*
        printf("\nReading BOS descriptor: ");
        if (libusb_get_bos_descriptor(handle, &bos_desc) == LIBUSB_SUCCESS) {
                printf("%d caps\n", bos_desc->bNumDeviceCaps);
@@ -860,6 +861,7 @@ static int test_device(uint16_t vid, uint16_t pid)
        } else {
                printf("no descriptor\n");
        }
+       */

        printf("\nReading first configuration descriptor:\n");
        CALL_CHECK_CLOSE(libusb_get_config_descriptor(dev, 0, &conf_desc), handle);
```

Debug log from libusbk.dll (libusb0.sys 1.2.7.2)
```

00000374	11.14433002	[13656] 	
00000375	11.14444828	[13656] AllK required contiguous memory = 675016 (64bit)	
00000376	11.14447117	[13656]   8 HotK Handles: HandleSize 2112 PoolSize 16912 (bytes)	
00000377	11.14453793	[13656]   64 LstK Handles: HandleSize 64 PoolSize 4112 (bytes)	
00000378	11.14463043	[13656]   2048 LstInfoK Handles: HandleSize 64 PoolSize 131088 (bytes)	
00000379	11.14466095	[13656]   128 UsbK Handles: HandleSize 96 PoolSize 12304 (bytes)	
00000380	11.14474201	[13656]   64 DevK Handles: HandleSize 112 PoolSize 7184 (bytes)	
00000381	11.14483356	[13656]   2048 OvlK Handles: HandleSize 104 PoolSize 213008 (bytes)	
00000382	11.14486408	[13656]   64 OvlPoolK Handles: HandleSize 96 PoolSize 6160 (bytes)	
00000383	11.14493179	[13656]   32 StmK Handles: HandleSize 176 PoolSize 5648 (bytes)	
00000384	11.14502525	[13656]   2048 IsochK Handles: HandleSize 136 PoolSize 278544 (bytes)	
00000385	11.14511871	[13656] 	
00000386	11.14516830	[13656] Dynamically allocated as needed:	
00000387	11.14523983	[13656]  KLST_DEVINFO = 2596 bytes each	
```

tormodvolden, 2021-10-31:
Thanks! This time there was some libusb0.sys debug output. Maybe you can put the BOS fetching back to compare again.

`libusb: warning [handle_timeout] async cancel failed -5`
I wonder if the libusb timeout warning and the hangs are related to issue I mentioned in the libusbk.dll pull requests (that it always uses sync mode and ignores the overlapped from libusb).

```
00000800	4.03162146	libusb0-sys:[get_descriptor] buffer size: 41 type: 0002 recipient: 0000 index: 0000 language id: 0000 timeout: 0	
00000801	4.03180456	libusb0-sys:[get_descriptor] buffer size: 4 type: 0003 recipient: 0000 index: 0000 language id: 0000 timeout: 1000	
00000803	4.03202057	libusb0-sys:dbg [call_usbd_ex] status = 00000000h	
```
I think, based on the sizes, the first get_descriptor is getting the device configuration descriptor (although this should have been cached by the OS and not require a USB request, and BTW maybe we can improve xusb to print the length) and the second is the language descriptor for the first string descriptor to be retrieved. The one call_usb_ex seems successful, probably the configuration descriptor that  we also see successfully displayed. The mystery is why there isn't a second one. Why the timeout limits are different I am not sure, but it could be normal.
`00000775	2.91944551	(libusbK.sys)[XferCtrl] bmDir=DeviceToHost(1) bmType=Standard(0) bmRecipient=Device(0) bmReserved=000 bRequest=6 wIndex=0 wValue=1006 wLength=18`
Is this the request causing error in the libusbK.sys case? What is xusb doing here? Well, I'll check the code later.

tormodvolden, 2021-10-31:
Another thing I'd like someone to try out is what it looks like with libusb just prior to (and after 9c28ad2).

tormodvolden, 2021-10-31:
I am starting to think most of the libusb0.sys breakage stems from the issues I mention in https://github.com/mcuee/libusbk/pull/33. Unfortunately I don't know the Windows internals well enough to come up with a good fix, but someone who does (Peter, Travis, ?) will probably figure it out. I was previously confused by the fact that some control transfers seemed to work, but that was before I realized that the libusb Windows backend retrieves device and configuration descriptors from the OS directly through DeviceIoControl() calls without using winusbx_submit_control_transfer ( -> ControlTransfer). Everything using libusb_submit_transfer for control transfers (e.g. libusb_control_transfer and libusb_get_string_descriptor) seems consistently broken with libusb0.sys since libusb expects async to work.

tormodvolden, 2021-10-31:
I see that the libusb HID backend is also using windows_force_sync_completion() in some cases where the transfer is synchronous, and it can be seen in the HID logs above. With my 0001-windows-Allow-synchronous-control-transfer-libusb0.patch.txt above I was expecting this to show up for libusb0 as well. Do we have a libusb debug log with this patch applied?

Note that this patch doesn't affect the "ControlTransfer failed" messages but those are relatively cosmetic, they appear when requesting a non-existing descriptor. AFAICS they were an annoyance on 1.0.23 and the original reports from 2015 but not an important issue. I think it is best to disable the BOS fetching in xusb for now, since that is not important for debugging the control transfer issue. Finally there is the "Kernel driver attached for interface 0: -12" issue which might be independent as well.

mcuee, 2021-11-01:
> I see that the libusb HID backend is also using windows_force_sync_completion() in some cases where the transfer is synchronous, and it can be seen in the HID logs above. With my [...]

The recent logs are with the patch applied as mentioned in the comment.
Eg: https://github.com/libusb/libusb/issues/94#issuecomment-952481115

mcuee, 2021-11-01:
> Note that this patch doesn't affect the "ControlTransfer failed" messages but those are relatively cosmetic, they appear when requesting a non-existing descriptor. AFAICS they were an annoyance on [...]

Yes I agree. 1.0.23 at least works even with the annoyances, but 1.0.24/git head are totally not working.

mcuee, 2021-11-01:
@tormodvolden 
The following tests will use the two patched written by you, and also with BOS fetching disabled.

```
$ git diff
diff --git a/examples/xusb.c b/examples/xusb.c
index 61ce4dd..5dc0cd5 100644
--- a/examples/xusb.c
+++ b/examples/xusb.c
@@ -851,6 +851,7 @@ static int test_device(uint16_t vid, uint16_t pid)
        string_index[1] = dev_desc.iProduct;
        string_index[2] = dev_desc.iSerialNumber;

+       /*
        printf("\nReading BOS descriptor: ");
        if (libusb_get_bos_descriptor(handle, &bos_desc) == LIBUSB_SUCCESS) {
                printf("%d caps\n", bos_desc->bNumDeviceCaps);
@@ -860,6 +861,7 @@ static int test_device(uint16_t vid, uint16_t pid)
        } else {
                printf("no descriptor\n");
        }
+       */

        printf("\nReading first configuration descriptor:\n");
        CALL_CHECK_CLOSE(libusb_get_config_descriptor(dev, 0, &conf_desc), handle);
diff --git a/libusb/os/windows_winusb.c b/libusb/os/windows_winusb.c
index df40773..387a7d7 100644
--- a/libusb/os/windows_winusb.c
+++ b/libusb/os/windows_winusb.c
@@ -2352,7 +2352,7 @@ cleanup_winusb:

                // NB: The below for loop works because the sub_api value for WinUSB
                // is a higher value than that of libusbK and libusb0
-               for (; sub_api < SUB_API_WINUSB; sub_api++) {
+               for (; sub_api <= SUB_API_WINUSB; sub_api++) { // try both original < and for test <=
                        libusbK_Set(sub_api, AbortPipe, true);
                        libusbK_Set(sub_api, ControlTransfer, true);
                        libusbK_Set(sub_api, FlushPipe, true);
@@ -2520,7 +2520,7 @@ static int winusbx_configure_endpoints(int sub_api, struct libusb_device_handle
                endpoint_address = (i == -1) ? 0 : priv->usb_interface[iface].endpoint[i];
                if (!WinUSBX[sub_api].SetPipePolicy(winusb_handle, endpoint_address,
                        PIPE_TRANSFER_TIMEOUT, sizeof(ULONG), &timeout))
-                       usbi_dbg(HANDLE_CTX(dev_handle), "failed to set PIPE_TRANSFER_TIMEOUT for control endpoint %02X", endpoint_address);
+                       usbi_dbg(HANDLE_CTX(dev_handle), "failed to set PIPE_TRANSFER_TIMEOUT for endpoint %02X", endpoint_address);

                if ((i == -1) || (sub_api == SUB_API_LIBUSB0))
                        continue; // Other policies don't apply to control endpoint or libusb0
@@ -2593,6 +2593,10 @@ static int winusbx_claim_interface(int sub_api, struct libusb_device_handle *dev
                                // it may be that we're using the libusb0 filter driver.
                                // TODO: can we move this whole business into the K/0 DLL?
                                r = LIBUSB_SUCCESS;
+
+                               usbi_dbg(ctx, "Initialize for interface %u (subapi %d) returned: %s", iface, sub_api, windows_error_str(err));
+                               usbi_dbg(ctx, "trying to find filter match for: %s", priv->usb_interface[iface].path);
+
                                for (_index = 0; ; _index++) {
                                        safe_free(dev_interface_path);

@@ -2603,6 +2607,8 @@ static int winusbx_claim_interface(int sub_api, struct libusb_device_handle *dev
                                        if ((r != LIBUSB_SUCCESS) || (dev_interface_path == NULL))
                                                break;

+                                       usbi_dbg(ctx, "- candidate (ignore if no guid): %s", dev_interface_path);
+
                                        // ignore GUID part
                                        dev_interface_path_guid_start = strchr(dev_interface_path, '{');
                                        if (dev_interface_path_guid_start == NULL)
@@ -2652,6 +2658,7 @@ static int winusbx_claim_interface(int sub_api, struct libusb_device_handle *dev
                }
                if (!WinUSBX[sub_api].GetAssociatedInterface(winusb_handle, (UCHAR)(iface - 1),
[... 14 lines cut ...]
@@ -2792,7 +2799,7 @@ static int winusbx_submit_control_transfer(int sub_api, struct usbi_transfer *it
                        return LIBUSB_ERROR_NOT_FOUND;
        }

-       usbi_dbg(ITRANSFER_CTX(itransfer), "will use interface %d", current_interface);
+       usbi_dbg(ITRANSFER_CTX(itransfer), "will use interface %d (subapi %d)", current_interface, sub_api);

        winusb_handle = handle_priv->interface_handle[current_interface].api_handle;
        set_transfer_priv_handle(itransfer, handle_priv->interface_handle[current_interface].dev_handle);
@@ -2808,11 +2815,13 @@ static int winusbx_submit_control_transfer(int sub_api, struct usbi_transfer *it
                }
                windows_force_sync_completion(itransfer, 0);
        } else {
-               if (!WinUSBX[sub_api].ControlTransfer(winusb_handle, *setup, transfer->buffer + LIBUSB_CONTROL_SETUP_SIZE, size, NULL, overlapped)) {
+               if (!WinUSBX[sub_api].ControlTransfer(winusb_handle, *setup, transfer->buffer + LIBUSB_CONTROL_SETUP_SIZE, size, &transferred, overlapped)) {
                        if (GetLastError() != ERROR_IO_PENDING) {
                                usbi_warn(TRANSFER_CTX(transfer), "ControlTransfer failed: %s", windows_error_str(0));
                                return LIBUSB_ERROR_IO;
                        }
+               } else {
+                       windows_force_sync_completion(itransfer, transferred);
                }
        }

```

mcuee, 2021-11-01:
Debug log with HID driver.

<details>
<summary>click for the debug log</summary>

```
C:\work\libusb\libusb\x64\Release\examples [master ↓7 +4 ~2 -0 !]> .\xusb.exe 0925:7001
Using libusb v1.0.24.11656

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000290] [000059c0] libusb: debug [libusb_init] created default context
[ 0.000413] [000059c0] libusb: debug [libusb_init] libusb v1.0.24.11656
[ 0.000538] [000059c0] libusb: debug [usbi_add_event_source] add HANDLE 000000000000008C events 0
[ 0.000655] [000059c0] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000758] [000059c0] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000090 events 0
[ 0.000866] [000059c0] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.000949] [000059c0] libusb: debug [htab_create] using 1021 entries hash table
[ 0.007517] [000059c0] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.013257] [000059c0] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.019383] [000059c0] libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
[ 0.019509] [000059c0] libusb: info [windows_init] UsbDk backend is not available
Opening device 0925:7001...
[ 0.019877] [000059c0] libusb: debug [libusb_get_device_list]
[ 0.020898] [000054d4] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.041406] [000059c0] libusb: debug [winusb_get_device_list] allocating new device for session [A1]
[ 0.041606] [000059c0] libusb: debug [winusb_get_device_list] allocating new device for session [A2]
[ 0.042023] [000059c0] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.042206] [000059c0] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.042317] [000059c0] libusb: debug [winusb_get_device_list] allocating new device for session [16]
[ 0.042938] [000059c0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.043086] [000059c0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.043194] [000059c0] libusb: debug [winusb_get_device_list] allocating new device for session [7E]
[ 0.043523] [000059c0] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.043674] [000059c0] libusb: debug [get_api_type] matched driver name against HID API
[ 0.043869] [000059c0] libusb: debug [winusb_get_device_list] allocating new device for session [7F]
[ 0.044221] [000059c0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.044408] [000059c0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.044560] [000059c0] libusb: debug [winusb_get_device_list] allocating new device for session [80]
[ 0.044928] [000059c0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.045156] [000059c0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.045290] [000059c0] libusb: debug [winusb_get_device_list] allocating new device for session [81]
[ 0.045659] [000059c0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.045848] [000059c0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.045965] [000059c0] libusb: debug [winusb_get_device_list] allocating new device for session [82]
[ 0.046347] [000059c0] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A36D&SUBSYS_091A1028&REV_10\3&11583659&0&A0' bus number 1
[ 0.046546] [000059c0] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_15DB&SUBSYS_091A1028&REV_02\71D9F51854B3020000' bus number 2
[ 0.047950] [000059c0] libusb: debug [winusb_get_device_list] extra GUID: {865C4DDE-10B6-6373-FAEE-1506B321A619}
[ 0.048061] [000059c0] libusb: debug [winusb_get_device_list] found existing device for session [82]
[ 0.048310] [000059c0] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.048460] [000059c0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.048623] [000059c0] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 1): 'USB\VID_046D&PID_C534\5&E9F3E45&0&1'
[ 0.048906] [000059c0] libusb: debug [winusb_get_device_list] found existing device for session [16]
[ 0.049092] [000059c0] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.049232] [000059c0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.049386] [000059c0] libusb: debug [init_device] (bus: 1, addr: 6, depth: 1, port: 14): 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
[ 0.049926] [000059c0] libusb: debug [winusb_get_device_list] found existing device for session [81]
[ 0.050126] [000059c0] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.050236] [000059c0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 288 bytes)
[ 0.050431] [000059c0] libusb: debug [init_device] (bus: 1, addr: 15, depth: 1, port: 2): 'USB\VID_047F&PID_C056\D1CEC32927974D5F9BD6B2AEBF2EA8E3'
[ 0.050920] [000059c0] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&36020D6F&0&0' reports 26 ports
[ 0.051097] [000059c0] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&36020D6F&0&0'
[ 0.051429] [000059c0] libusb: debug [winusb_get_device_list] found existing device for session [80]
[ 0.051617] [000059c0] libusb: debug [init_device] found 1 configurations (current config: 1)
[... cut ...]
[ 0.132197] [000059c0] libusb: debug [libusb_unref_device] destroy device 1.17
[ 0.132250] [000059c0] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.132308] [000059c0] libusb: debug [libusb_exit] destroying default context
[ 0.132402] [000054d4] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.134437] [000059c0] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000090
[ 0.134551] [000059c0] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000008C
```

</details>

mcuee, 2021-11-01:
No issues even with the BOS descriptor fetching enabled, tested with libusb0.sys 1.2.6.0 version and 1.2.7.2 version.

```
C:\work\libusb\libusb\x64\Release\examples [master ↓7 +4 ~2 -0 !]> $Env:LIBUSB_DEBUG=0
C:\work\libusb\libusb\x64\Release\examples [master ↓7 +4 ~2 -0 !]> .\xusb.exe 0925:7001
Using libusb v1.0.24.11656

Opening device 0925:7001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 0925:7001
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 01
       endpoint[1].address: 01
           max packet size: 0040
          polling interval: 01

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Lakeview Research"
   String (0x02): "Generic HID"

Releasing interface 0...
Closing device...
```

tormodvolden, 2021-11-01:
That's wonderful! Thanks for all the testing and the clear reporting.

The "debug" patch should not be needed, what counts is the 0001-windows-Allow-synchronous-control-transfer-libusb0.patch.txt. I will make a merge request for this one, and hopefully Chris can take a look.

tormodvolden, 2021-11-01:
In the debug log with BOS fetching re-enabled, there are still the BOS warning and error messages, but they are suppressed by LIBUSB_DEBUG=0. I'd suggest separate bug reports for:
- result/error propagation when control transfer fails (e.g. request for missing descriptor)
- Kernel driver attached for interface 0: -12
- any problems with filter driver (use my debug patch in that case)

mcuee, 2021-11-02:
I will test libusb0.sys in filter mode again to see if that has been improved or not. It had more problems than device driver mode.

mcuee, 2021-11-02:
I add the debug log for HID and WinUSB in the comments above.
https://github.com/libusb/libusb/issues/94#issuecomment-955999279

mcuee, 2021-11-02:
Test https://github.com/libusb/libusb/pull/1016 as well and it seems to work fine.
Test results with debug logs: https://github.com/libusb/libusb/pull/1016#issuecomment-957033908

mcuee, 2021-11-02:
> In the debug log with BOS fetching re-enabled, there are still the BOS warning and error messages, but they are suppressed by LIBUSB_DEBUG=0. I'd suggest separate bug reports for:
> 
> * result/error propagation when control transfer fails (e.g. request for missing descriptor)

@tormodvolden Just wondering what kind of improvement you are looking at in this area. Thanks.

mcuee, 2021-11-02:
Once the following pull request is merged, I will close this issue.
- #1016

mcuee, 2021-11-02:
>  You can check my prior comments, but there should be an easier work-around for Windows 7 if it does not work.
> 
> 1. Using Zadig 2.6 release to install libusb0.sys 1.2.6.0 device driver.
> 2. Using the libusb-win32 1.2.7.2 installer to upgrade libusb0.sys and libusb0.dll

@pazourek

Have you tried to the above trick?
It could be true that you can not get 1.2.7.2 to work under Windows 7 as per the comments here. 
https://github.com/mcuee/libusb-win32/issues/24#issuecomment-955795763

But with #1016, you should be able to resolve #761without using libusb0.sys 1.2.7.2 version.

tormodvolden, 2021-11-02:
> > * result/error propagation when control transfer fails (e.g. request for missing descriptor)
> 
> @tormodvolden Just wondering what kind of improvement you are looking at in this area. Thanks.

I have filed this in #1017.

mcuee, 2021-11-09:
This issue will be closed once #1016 is merged.
- #1016 

Then it will split into two issues.
1) libusb0.sys filter driver mode issue 
- #1019
2) libusb0.sys result/error propagation when control transfer fails (e.g. request for missing descriptor)
- #1017

closed 2021-11-09 by tormodvolden

mcuee, 2021-11-11:
@pazourek BTW, it is confirmed that libusb0.sys 1.2.7.2 snapshot release does not work under Windows 7.
https://github.com/mcuee/libusb-win32/issues/24

mcuee, 2021-11-15:
https://github.com/mcuee/libusb-win32/issues/24
libusb-win32 snapshot release 1.2.7.3 works under Windows 7.

Snapshot release including Zadig: https://github.com/mcuee/libusb-win32/releases/tag/snapshot_1.2.7.3

commit 5b408d10 (2021-11-09): Windows: Allow synchronous control transfers (for libusb0)

Some of the changes in commit 9c28ad2 rely on all transfers having (or
appearing to have) asynchronous completion.

However, the libusb0.sys backend of libusbk.dll performs all control
transfers synchronously and ignores any "overlapped" structure handed to
it. Our asynchronous handling will in this case be pending and
eventually time out although the USB request itself was successful.

Therefore restore the possibility of synchronous completion of control
transfers, by forcing the completion handling on a successful return
from reque [... cut]