libusb/libusb #96: Broken On OS X El Capitan

iAdam1n, 2015-08-29:
I've tried running greenpois0n while on OS X El Capitan in the way that is required for accessing the AES engine on my iPhone 4 but it gets stuck on injecting, freezes and the bar is all fuzzy. Is this a known issue? Are there plans to fix it? I'm aware El Capitan is in beta so it might not be right now.

mcuee, 2015-08-30:
Your log does not really tell much. You can set env variable LIBUSB_DEBUG=4 and then print the logs.

iAdam1n, 2015-08-30:
I'm afraid it didn't really change anything even though I'd done it but here is the log - https://ghostbin.com/paste/kszbx. When it says "Waiting 2 seconds for the device to pop up..." unless I unplug the device and plug it back in again, I constantly get "Connection failed. Waiting 1 sec before retry." and that did not happen on Yosemite.

hjelmn, 2015-08-30:
The first run is getting kIOUSBNoAsyncPortErr which shouldn't be possible if the device is open. I have 10.11 installed. Let me see if I can reproduce the error.

hjelmn, 2015-08-30:
Keep in mind that 10.11 is still in beta. It could be a IOKit bug.

iAdam1n, 2015-08-31:
The device was in DFU mode because it has to be while greenpois0n is used which will then take it to a white screen due to how running the AES engine works.

mcuee, 2015-10-17:
Please try the release version of 10.11 to see if it is still there or not.

iAdam1n, 2015-10-17:
Sadly that makes no difference. I still have to unplug and reconnect the device as well as still getting this multiple times.

libusb:error [darwin_transfer_status] transfer error: timed out

mcuee, 2015-10-23:
This may not be libusb problem as well since it could be the host program needs to wait a bit longer. BTW, what is your Mac model?

iAdam1n, 2015-10-23:
MacBookPro8,2. MacBook Pro Early 2011. It worked fine on Yosemite.

hjelmn, 2015-11-27:
Radar bug 23677332..

hjelmn, 2016-01-12:
Can you try again with the latest 10.11.4 beta? This may be fixed.

iAdam1n, 2016-01-14:
No it's still the same.

hjelmn, 2016-01-14:
Damn, this is a hard one to get Apple to look at directly considering it involves greenpois0n. Was hoping it was a similar issue to #111 and #123.

iAdam1n, 2016-01-14:
So this is definitely an Apple bug?

hjelmn, 2016-01-14:
Probably. They rewrote the usb stack in 10.11 and there have been numerous bugs. I was hoping this was related to one of the other ones we are tracking.

Cheong2K, 2016-01-29:
Confirmed, fixed in OSX 10.11.4 beta.

iAdam1n, 2016-01-29:
@Cheong2K It can't be fixed. Still happens here.

mcuee, 2016-09-20:
Just wondering if the issue is still there or not.

mcuee, 2016-12-02:
Just wondering if this issue is still there under macOS or not? Will close this issue if no further info is provided in another month.

iAdam1n, 2016-12-02:
@mcuee Yes, the issue is still there. I just tested it.

mcuee, 2016-12-02:
Is it still the same debug log?
`libusb:error [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
libusb:error [submit_control_transfer] control request failed: no async port has been opened for interface'

That seems to mean that another program is accessing the device.

mcuee, 2016-12-02:
If that is the case, I am thinking this is not a libusb problem. You may need to find the other program which access the device and block that program. Or if it is another system driver, then you may need to use a codeless kext (not that easy under newer generation of OS X including macOS) to block the systemd driver.

iAdam1n, 2016-12-02:
@mcuee I'm still trying to get the debug to work, it isn't at the moment. It seems it is a libusb issue, see [this ghostbin](https://ghostbin.com/paste/rkjwc).

mcuee, 2016-12-02:
Thanks. A debug log will help.

On the other hand, the tool seems to use older version of libirecovery. Have you tried to update to the latest version of libirecovery?
https://github.com/libimobiledevice/libirecovery

iAdam1n, 2016-12-02:
@mcuee How do I create a debug log? I've posted [this ghostbin](https://ghostbin.com/paste/84vzp) with the latest version of libirecovery that I just compiled and it still doesn't work.

mcuee, 2016-12-02:
You generate the debug for libusb when you set the environment variable LIBUSB_DEBUG=4.

Example:
`
MyMacMini:libusb xiaofanc$ export LIBUSB_DEBUG=4
MyMacMini:libusb xiaofanc$ ./examples/listdevs 
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000013] [00000307] libusb: debug [libusb_init] created default context
[ 0.000079] [00000307] libusb: debug [libusb_init] libusb v1.0.21.11157
[ 0.000374] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x51dba752
[ 0.000406] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x51dba752
[ 0.001801] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
`

iAdam1n, 2016-12-02:
@mcuee ok thanks. [Here](https://ghostbin.com/paste/r7hzv) is the log.

dickens, 2017-08-25:
@mcuee, @hjelmn, any update to provide on this?

mcuee, 2017-09-10:
I do not have any updates for this. We probably need to ask the OP to try
latest version of Mac OS X to see if the problem is still there or not. As
this involved Jailbreak, Apple is not going to help if it is related to Mac
OS X USB stack.

iAdam1n, 2017-09-18:
@mcuee I'm running macOS High Sierra GM and the issue is still there.

mcuee, 2020-07-29:
@iAdam1n Just wondering if this issue still exist or not.

catap, 2020-07-29:
@mcuee I've "enjoyed" this issue when I started to create http://github.com/catap/swrng-provider

So, it is existed at May'20 :)

mcuee, 2020-07-29:
Thanks for the updates.

mcuee, 2020-12-07:
@catap Just wondering if you can provide more details on the issues you encounter. Which macOS version are you using? Do you have any debug logs to share?

catap, 2020-12-07:
@mcuee it was at Mojave and I don't think that I can provide any debug logs because it was too long ago.

Sorry.

mcuee, 2020-12-07:
@catap Just want to make sure, so you are saying that now you do not encounter the issue with later macOS version like Catalina and Big Sur, right?

catap, 2020-12-07:
@mcuee I rewritten code to another way and I haven't reproduce this issue.

Why? Because code that was encountered it was rewritten long ago and I haven't got it anymore.

mcuee, 2020-12-10:
I see. No problem. I will keep this issue as open.

mcuee, 2021-02-26:
In this case, I think it is better to close the issue and reopen if the issue surfaces again.

closed 2021-02-26 by mcuee