libusb/libusb #103: libusb_cancel_transfer requires multiple calls to libusb_handle_events_timeout_completed

radioflash, 2015-09-25:
Under some circumstances, after calling ``libusb_cancel_transfer`` *multiple* (in my case, two) calls to ``libusb_handle_events_timeout_completed`` are required before the transfer callback is triggered.

This is unexpected behavior. Even waiting for 100ms after cancelling the transfer, and then calling ``libusb_handle_events_timeout_completed`` *once* will **not** trigger the callback in my case. I need two consecutive calls to ``libusb_handle_events_timeout_completed``, or my "transfer cancelled" callback won't arrive.

If this behavior is as intended, **please** mention it in the documentation of the ``libusb_cancel_transfer function`` (and maybe also on the ``Asynchronous device I/O``- page).

If you want to fix this, just reply and I will provide additional information on how to trigger this.

jwrdegoede, 2015-09-25:
Hi,

With which version of libusb are you seeing this ? And on which operating system ?

The just released libusb-1.0.20 contains a Linux backend bug-fix which makes libusb_handle_events_* process all it events it can, rather then only the first one.

Regards,

Hans

radioflash, 2015-09-25:
OS: Windows 8.1
libusb-v1.0.19.10905 (now libusb-v1.0.20.11003, but I still have the same problem)
Custom device based on STM32F4, WinUSB descriptor, control and bulk transfers.

This behavior only happens in specific circumstances. In my case, I request data from my device (with a synchronous control package). Then I repeatedly submit a BULK-IN transfer, wait for its completion and resubmit it until all available data is received. When cancelling the last bulk transfer, multiple handle_event calls become necessary as described.

This is not how I would normally communicate with the device, but I was investigating a way to read all the currently available data from a bulk endpoint (in order to make sure that all future bulk-reads only return "fresh" data).

PS: Multiple ``libusb_handle_events_timeout_completed`` also seem necessary when checking for timed-out-transfers:
Submitting a bulk in transfer with a 10ms timeout when no data is available, then calling ``libusb_handle_events_timeout_completed`` with a 80000µs tv_usec value will not trigger the transfer callback, but calling ``libusb_handle_events_timeout_completed`` three times with a 10000µs tv_usec value will.

jwrdegoede, 2015-09-25:
"OS: Windows 8.1" Ah, that is not really my thing. Then I'll leave your problem in the capable hands of the windows maintainers.

dickens, 2015-10-26:
The application should not make any assumptions about how libusb will handle events internally. If there are any outstanding transfers, the application must wait until they are all completed, calling the event handling functions as many times as necessary. The sync API does this internally, and users of the async API are required to do this as well. I don't think there is anything to "fix" here, though as you suggest it may be prudent to add my first sentence in this reply to the docs.

Now, just by reading the description, I cannot see why multiple event handling calls would be required for a single outstanding transfer. If you'd like to provide a debug log of this situation I'll gladly look it over and provide feedback.

Dimonsky, 2016-11-08:
Can I just update this topic, not to create one more?

There is an example coming with the libusb, called dpfp_threaded.c (and dpfp.c without threads)

There is a function in it, called init_capture(void)

It is a subject of the same problem, discussed in this topic.
When the second libusb_submin_transfer returns < 0, the code goes into **if ( r<0)** block, and tries to "cancel" the transfer.

And it never makes the callback receiving _CANCELED status.
Even if I unplug the device, the function always returns 0, and a callback is not called. Original code uses libusb_handle_events(), which inside calls libusb_handle_events_timeout with 60seconds timer (!!!) so, i wait 60 seconds this function to return when device unplugged.

I use lib 1.0.20.
Original code from the example:
static int init_capture(void)
{
	int r;

	r = libusb_submit_transfer(irq_transfer);
	if (r < 0)
		return r;

	r = libusb_submit_transfer(img_transfer);
	if (r < 0) {
		libusb_cancel_transfer(irq_transfer);
		while (irq_transfer)
			if (libusb_handle_events(NULL) < 0)
				break;
		return r;
	}

	/* start state machine */
	state = STATE_AWAIT_IRQ_FINGER_REMOVED;
	return next_state();
}

My modification not to wait 60seconds:
static int init_capture(void)
{
	int r;

	r = libusb_submit_transfer(irq_transfer);
	if (r < 0)
		return r;

	r = libusb_submit_transfer(img_transfer);
	if (r < 0) 	
	{
		boost::unique_lock<boost::mutex> lock(m_mutexHandleEvents);

		libusb_cancel_transfer(irq_transfer);
		while (irq_transfer)
		{			
			struct timeval tv;
			tv.tv_sec = 1;
			tv.tv_usec = 0; 			
			if (libusb_handle_events_timeout(NULL, &tv) < 0)
				break;			
		}
		return r;
	}

	/* start state machine */
	state = STATE_AWAIT_IRQ_FINGER_REMOVED;
	return next_state();
}

Pls. resonse if there any workaround...

dickens, 2020-08-18:
Can anyone who has encountered this please confirm whether this is still an issue?

mcuee, 2021-01-23:
Close this for now due to inactivity. Please reopen if the issue is still there with the latest codes.

closed 2021-01-23 by mcuee