libusb/libusb #105: manage doc/ better

hiyuh, 2015-09-30:
current `doc/Makefile.am` is simply not cool.
```
EXTRA_DIST = doxygen.cfg.in

docs: doxygen.cfg
        doxygen $^

docs-upload: docs
        ln -s html api-1.0
        scp -r api-1.0 pbatard@web.sourceforge.net:/home/project-web/libusb/htdocs
        rm -f api-1.0
```
* `docs` is used, why not `doc`...
*  `scp` command has hardcoded dev username and sf.net URI.
* users can not run `doxygen` and `install` via top Makefile.

a case of users optionally install doc via top Makefile is worth to consider.
* uBlock Origin recently blocks sf.net b/c some their scam.
* a few dev scene needs libusb doc w/o online connection.

hiyuh, 2015-10-20:
note, `doc/doxygen.cfg.in` w/ `GENERATE_MAN = YES` outputs,
```
hiyuh@lynx ~/git-repos/libusb/doc
$ ls man/man3/
_home_hiyuh_git-repos_libusb_libusb_.3  libusb_endpoint_descriptor.3
api.3                                   libusb_interface.3
asyncio.3                               libusb_interface_descriptor.3
caveats.3                               libusb_iso_packet_descriptor.3
contexts.3                              libusb_pollfd.3
desc.3                                  libusb_ss_endpoint_companion_descriptor.3
dev.3                                   libusb_ss_usb_device_capability_descriptor.3
hotplug.3                               libusb_transfer.3
io.3                                    libusb_usb_2_0_extension_descriptor.3
lib.3                                   libusb_version.3
libusb_bos_descriptor.3                 misc.3
libusb_bos_dev_capability_descriptor.3  mtasync.3
libusb_config_descriptor.3              packetoverflow.3
libusb_container_id_descriptor.3        poll.3
libusb_control_setup.3                  syncio.3
libusb_device_descriptor.3
```
as you can see, some summarized docs have no `libusb_` prefix.
consider possible conflict in `/usr/share/man`.

hiyuh, 2015-12-08:
current libusb doc inherits following unsensible function decls,
```
$ grep -e 'libusb_device_handle \*dev' libusb/libusb.h
        libusb_device_handle *dev_handle;
int LIBUSB_CALL libusb_get_configuration(libusb_device_handle *dev,
void LIBUSB_CALL libusb_close(libusb_device_handle *dev_handle);
libusb_device * LIBUSB_CALL libusb_get_device(libusb_device_handle *dev_handle);
int LIBUSB_CALL libusb_set_configuration(libusb_device_handle *dev,
int LIBUSB_CALL libusb_claim_interface(libusb_device_handle *dev,
int LIBUSB_CALL libusb_release_interface(libusb_device_handle *dev,
int LIBUSB_CALL libusb_set_interface_alt_setting(libusb_device_handle *dev,
int LIBUSB_CALL libusb_clear_halt(libusb_device_handle *dev,
int LIBUSB_CALL libusb_reset_device(libusb_device_handle *dev);
int LIBUSB_CALL libusb_alloc_streams(libusb_device_handle *dev,
int LIBUSB_CALL libusb_free_streams(libusb_device_handle *dev,
int LIBUSB_CALL libusb_kernel_driver_active(libusb_device_handle *dev,
int LIBUSB_CALL libusb_detach_kernel_driver(libusb_device_handle *dev,
int LIBUSB_CALL libusb_attach_kernel_driver(libusb_device_handle *dev,
        libusb_device_handle *dev, int enable);
        struct libusb_transfer *transfer, libusb_device_handle *dev_handle,
        libusb_device_handle *dev_handle, unsigned char endpoint,
        struct libusb_transfer *transfer, libusb_device_handle *dev_handle,
        struct libusb_transfer *transfer, libusb_device_handle *dev_handle,
        libusb_device_handle *dev_handle, unsigned char endpoint,
int LIBUSB_CALL libusb_control_transfer(libusb_device_handle *dev_handle,
int LIBUSB_CALL libusb_bulk_transfer(libusb_device_handle *dev_handle,
int LIBUSB_CALL libusb_interrupt_transfer(libusb_device_handle *dev_handle,
static inline int libusb_get_descriptor(libusb_device_handle *dev,
static inline int libusb_get_string_descriptor(libusb_device_handle *dev,
int LIBUSB_CALL libusb_get_string_descriptor_ascii(libusb_device_handle *dev,
```
there is no reason to use insane named arg, `dev` or `dev_handle` for `libusb_device_handle`.
b/c it looks simply eyesore due to many `libusb_device *dev` are w/ them.
just use `h`, `hnd` or `handle` looks simply better.

LudovicRousseau, 2015-12-08:
@hiyuh you should not mix different issues in the same bug report. Create a different issue for each different problem.

hiyuh, 2015-12-09:
@LudovicRousseau 
ok, sorry for noise.
i'll split them ASAP.

hiyuh, 2015-12-09:
done, now closing this issue...

closed 2015-12-09 by hiyuh