libusb/libusb #107: Device no longer detected after multiple disconnects with 1.0.16

AlexandruCostache, 2015-10-05:
Hi,

We've been using libusb 1.0.16 with the following patch included:
d8a33df linux: Use a separate lock to serialize start/stop vs hotplug events

The problem here is that after a large number of connects and disconnects (200 or more), the device is no longer detected by libusb. I know that 1.0.16 is quite old and have tested with 1.0.20 and it looks like this issue has been fixed. Unfortunately updating is not allowed so we need to bring in the fix in the form of patches.

I have uploaded debug logs here: 
https://www.dropbox.com/s/ln434pffkf8d35b/libusb_log_no_device_detected_disconnects.zip?dl=0

and would like to ask if anyone has any idea of what changes may fix this behavior. 

The log is quite large and I noted that the issue starts to happen from here:

[1395.985080] [0000081f] libusbx: debug [libusb_init] created default context
[1395.985337] [0000081f] libusbx: debug [libusb_init] libusbx v1.0.16.10774
[1395.985673] [0000081f] libusbx: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[1395.985971] [0000081f] libusbx: debug [op_init] bulk continuation flag supported
[1395.986155] [0000081f] libusbx: debug [op_init] zero length packet flag supported
[1395.986340] [0000081f] libusbx: debug [op_init] sysfs can relate devices
[1395.986511] [0000081f] libusbx: debug [op_init] sysfs has complete descriptors
[1395.986702] [0000081f] libusbx: error [op_init] error starting hotplug event monitor
[1400.341832] [0000081f] libusbx: debug [libusb_init] created default context
[1400.341886] [0000081f] libusbx: debug [libusb_init] libusbx v1.0.16.10774
[1400.341936] [0000081f] libusbx: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[1400.341946] [0000081f] libusbx: debug [op_init] bulk continuation flag supported
[1400.341952] [0000081f] libusbx: debug [op_init] zero length packet flag supported
[1400.341973] [0000081f] libusbx: debug [op_init] sysfs can relate devices
[1400.341981] [0000081f] libusbx: debug [op_init] sysfs has complete descriptors
[1400.342015] [0000081f] libusbx: error [op_init] error starting hotplug event monitor

Thank you!

AlexandruCostache, 2015-10-13:
Hi,

Maybe I could provide some more detailed logs for this?

Thanks

dickens, 2015-10-14:
Hi,

I would minimally apply the following patches to see if it helps:

https://github.com/libusb/libusb/commit/9afce4282ecbb32e0cef9282d7f97f2353d7eca1
https://github.com/libusb/libusb/commit/3107f30baddabf18039a4ca99c8a50b7f13f4de8
https://github.com/libusb/libusb/commit/47e1141f6b064a99fad7956b9b00b335bd90476d

Regards,
Chris

AlexandruCostache, 2015-10-16:
Thank you very much Chris! 

Had more than 700 connect-disconnect cycles with the patches you suggested and the device is still detected

closed 2015-10-16 by AlexandruCostache