libusb/libusb #108: Slow enumeration on el Capitan

ravens, 2015-10-07:
Hi,

I noticed since the upgrade to el Capitan my libusb-based program (example, hackrf utilities) are extremely slow. I join logs of execution of one of the hackrf program and the same log on another machine. The difference is noticeable (4s >> 0,007). This happen to other libusb dependent tools.

On the el Capitan machine:
hackrf_info 
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000008] [0000050f] libusb: debug [libusb_init] created default context
[ 0.000060] [0000050f] libusb: debug [libusb_init] libusb v1.0.20.11004
[ 0.000189] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.000227] [0000050f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x0
[ 0.001558] [0000050f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.001576] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.001584] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0100
[ 0.001588] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.001593] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0xff
[ 0.001597] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x00
[ 0.001601] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x08
[ 0.001605] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.001609] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8005
[ 0.001613] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0198
[ 0.001619] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.001623] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.001627] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.001631] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 1.002889] [0000050f] libusb: debug [darwin_check_configuration] active config: 0, first config: 1
[ 1.002946] [0000050f] libusb: debug [process_new_device] allocating new device in context 0x7fd8c9c06b00 for with session 0x0
[ 1.002972] [0000050f] libusb: debug [process_new_device] found device with address 0 port = 0 parent = 0x0 at 0x7fd8c9c0704a
[ 1.003028] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 1.003056] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x0 against cached device with sessionID 0x0
[ 1.003061] [0000050f] libusb: debug [darwin_get_cached_device] using cached device for device
[ 2.004243] [0000050f] libusb: debug [darwin_check_configuration] active config: 0, first config: 1
[ 2.004301] [0000050f] libusb: debug [process_new_device] allocating new device in context 0x7fd8c9c06b00 for with session 0x0
[ 2.004383] [0000050f] libusb: debug [process_new_device] found device with address 0 port = 0 parent = 0x0 at 0x7fd8c9c0704a
[ 2.004426] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 2.004453] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x0 against cached device with sessionID 0x0
[ 2.004458] [0000050f] libusb: debug [darwin_get_cached_device] using cached device for device
[ 3.005674] [0000050f] libusb: debug [darwin_check_configuration] active config: 0, first config: 1
[ 3.005733] [0000050f] libusb: debug [process_new_device] allocating new device in context 0x7fd8c9c06b00 for with session 0x0
[ 3.005756] [0000050f] libusb: debug [process_new_device] found device with address 0 port = 0 parent = 0x0 at 0x7fd8c9c0704a
[ 3.005894] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x266aa5fb0084
[ 3.005937] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x266aa5fb0084 against cached device with sessionID 0x0
[ 3.005944] [0000050f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x266aa5fb0084
[ 3.006782] [0000050f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 3.006797] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 3.006802] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 3.006806] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 3.006810] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 3.006813] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x00
[ 3.006817] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x08
[ 3.006821] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0a5c
[ 3.006827] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x4500
[ 3.006831] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0100
[ 3.006834] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 3.006838] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 3.006841] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 3.006845] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 3.006888] [0000050f] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 3.006894] [0000050f] libusb: debug [process_new_device] allocating new device in context 0x7fd8c9c06b00 for with session 0x266aa5fb0084
[ 3.006935] [0000050f] libusb: debug [process_new_device] found device with address 2 port = 0 parent = 0x0 at 0x7fd8c9c08bfa
[ 3.006996] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x266aba7ccc38
[ 3.007044] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x266aba7ccc38 against cached device with sessionID 0x266aa5fb0084
[ 3.007051] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x266aba7ccc38 against cached device with sessionID 0x0
[ 3.007055] [0000050f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x266aba7ccc38
[ 3.007891] [0000050f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 3.007907] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 3.007914] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 3.007920] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0xff
[ 3.007925] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x01
[... 196 lines cut ...]
[ 0.007065] [00002384] libusbx: debug [linux_get_device_address] getting address for device: 4-1 detached: 0
[ 0.007092] [00002384] libusbx: debug [linux_get_device_address] scan 4-1
[ 0.007187] [00002384] libusbx: debug [linux_get_device_address] bus=4 dev=2
[ 0.007212] [00002384] libusbx: debug [linux_enumerate_device] busnum 4 devaddr 2 session_id 1026
[ 0.007235] [00002384] libusbx: debug [linux_enumerate_device] allocating new device for 4/2 (session 1026)
[ 0.007327] [00002384] libusbx: debug [linux_get_parent_info] Dev 0x1453380 (4-1) has parent 0x1452ed0 (usb4) port 1
[ 0.007456] [00002384] libusbx: debug [usbi_add_pollfd] add fd 6 events 1
[ 0.007491] [00002384] libusbx: debug [usbi_add_pollfd] add fd 8 events 1
[ 0.007519] [00002384] libusbx: debug [usbi_io_init] using timerfd for timeouts
[ 0.007538] [00002384] libusbx: debug [usbi_add_pollfd] add fd 10 events 1
[ 0.007562] [00002384] libusbx: debug [libusb_get_device_list] 
[ 0.007592] [00002384] libusbx: debug [discovered_devs_append] need to increase capacity
[ 0.007622] [00002384] libusbx: debug [libusb_get_device_descriptor] 
[ 0.007646] [00002384] libusbx: debug [libusb_get_device_descriptor] 
[ 0.007668] [00002384] libusbx: debug [libusb_get_device_descriptor] 
[ 0.007689] [00002384] libusbx: debug [libusb_get_device_descriptor] 
[ 0.007714] [00002384] libusbx: debug [libusb_get_device_descriptor] 
[ 0.007739] [00002384] libusbx: debug [libusb_get_device_descriptor] 
[ 0.007762] [00002384] libusbx: debug [libusb_get_device_descriptor] 
[ 0.007787] [00002384] libusbx: debug [libusb_get_device_descriptor] 
[ 0.007811] [00002384] libusbx: debug [libusb_get_device_descriptor] 
[ 0.007835] [00002384] libusbx: debug [libusb_get_device_descriptor] 
[ 0.007859] [00002384] libusbx: debug [libusb_get_device_descriptor] 
[ 0.007884] [00002384] libusbx: debug [libusb_get_device_descriptor] 
No HackRF boards found.

pavelrevak, 2015-10-09:
also same issue, from any application using libusb

hjelmn, 2015-10-09:
I see the problem. Will put together a fix today.

closed 2015-10-09 by hjelmn

pavelrevak, 2015-10-10:
thanks, works well

ravens, 2015-10-11:
Yes, many thanks, problem fixed here too.

commit 5e45e074 (2015-10-09): darwin: ignore root hub simulation devices

This commit fixes #108. In the 10.11.0 release it can take up to 1
second to query the configuration of one of Apple's root hub
simulation devices. Since the devices are not usable anyway go ahead
and mark them as unconfigured.