libusb/libusb #111: Problem with second connection to ST-Link device on OS/X "El Capitan"

pavelrevak, 2015-10-12:
After upgrade Mac to El Capitan I see several problems, one of is solved in #108 but I see another problem. I'm not sure if it is only device specific problem, but in previous version of OS/X was everything OK.

I see this issue with ST-Link devices. In my case I have two different devices ST-Link V2 and ST-Link V2-1 whith different idProduct numbers.

I tested it with following commands (from python over pyusb connected to latest libusb):

    >>> import usb.core
    >>> d = usb.core.find(idVendor=0x0483, idProduct=0x3748)
    >>> d.write(0x02, [0xf1, 0x80, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 1000)
    >>> d.read(0x81, 6, 1000)
    array('B', [37, 192, 131, 4, 72, 55])

When I run this first time it is OK, but If I run this code second time then it timeout in read method.

This problem also occurs in applications written in C, like https://github.com/texane/stlink and https://github.com/ntfreak/openocd where is same issue.

In detailed log I don't see any differences instead of timeout on reading after one second:

    d.read(0x81, 6, 1000)
    [ 5.942978] [00000a0b] libusb: debug [libusb_get_config_descriptor] index 0
    [ 5.943145] [00000a0b] libusb: debug [libusb_get_config_descriptor] index 0
    [ 5.943280] [00000a0b] libusb: debug [libusb_get_config_descriptor] index 0
    [ 5.943462] [00000a0b] libusb: debug [libusb_alloc_transfer] transfer 0x7f8171df6eb8
    [ 5.943473] [00000a0b] libusb: debug [libusb_submit_transfer] transfer 0x7f8171df6eb8
    [ 5.943497] [00000a0b] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
    [ 5.943505] [00000a0b] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
    [ 5.943675] [00000a0b] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
    [ 5.943689] [00000a0b] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
    [ 5.943697] [00000a0b] libusb: debug [handle_events] poll() 1 fds with timeout in 60000ms
    [ 6.943900] [00002207] libusb: debug [darwin_async_io_callback] an async io operation has completed
    [ 6.943984] [00000a0b] libusb: debug [handle_events] poll() returned 1
    [ 6.943995] [00000a0b] libusb: debug [handle_events] caught a fish on the event pipe
    [ 6.944002] [00000a0b] libusb: debug [darwin_handle_transfer_completion] handling bulk completion with kernel status -536854447

    --- to here it is same (instead of time delay) ---

    [ 6.944009] [00000a0b] libusb: warning [darwin_transfer_status] transfer error: timed out
    [ 6.944015] [00000a0b] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f8171df6eb8 has callback 0x10a187f8c
    [ 6.944023] [00000a0b] libusb: debug [sync_transfer_cb] actual_length=0
    [ 6.944033] [00000a0b] libusb: debug [libusb_free_transfer] transfer 0x7f8171df6eb8

It looks like the device not send an answer or the answer was lost....

Actually help only disconnect and connect back the device.

mcuee, 2015-10-17:
Just want to make sure I understand you, you have problems with both
ST-Link V2, and ST-Link V2-1. Right? Have you updated the firmware? What is
the model of your Mac computer?

mcuee, 2015-10-17:
I can not reproduce this issue with my ST-Link V2 on a Mac Mini 2011
running Mac OS X 10.11.

````
MyMacMini:~ xiaofanc$ python
Python 2.7.10 (default, Jul 13 2015, 12:05:58)
[GCC 4.2.1 Compatible Apple LLVM 6.1.0 (clang-602.0.53)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> d = usb.core.find(idVendor=0x0483, idProduct=0x3748)
>>> d.write(0x02, [0xf1, 0x80, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 1000)
16
>>> d.read(0x81, 6, 1000)
array('B', [37, 64, 131, 4, 72, 55])
>>> d.write(0x02, [0xf1, 0x80, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 1000)
16
>>> d.read(0x81, 6, 1000)
array('B', [37, 64, 131, 4, 72, 55])
>>> exit()

MyMacMini:~ xiaofanc$ python
Python 2.7.10 (default, Jul 13 2015, 12:05:58)
[GCC 4.2.1 Compatible Apple LLVM 6.1.0 (clang-602.0.53)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> d = usb.core.find(idVendor=0x0483, idProduct=0x3748)
>>> d.write(0x02, [0xf1, 0x80, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 1000)
16
>>> d.read(0x81, 6, 1000)
array('B', [37, 64, 131, 4, 72, 55])
>>> d.write(0x02, [0xf1, 0x80, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 1000)
16
>>> d.read(0x81, 6, 1000)
array('B', [37, 64, 131, 4, 72, 55])
>>> exit()
````

mcuee, 2015-10-17:
I can not reproduce this issue with my ST-Link V2 on a Mac Mini 2011
running Mac OS X 10.11.

```
MyMacMini:~ xiaofanc$ python
Python 2.7.10 (default, Jul 13 2015, 12:05:58)
[GCC 4.2.1 Compatible Apple LLVM 6.1.0 (clang-602.0.53)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> d = usb.core.find(idVendor=0x0483, idProduct=0x3748)
>>> d.write(0x02, [0xf1, 0x80, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 1000)
16
>>> d.read(0x81, 6, 1000)
array('B', [37, 64, 131, 4, 72, 55])
>>> d.write(0x02, [0xf1, 0x80, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 1000)
16
>>> d.read(0x81, 6, 1000)
array('B', [37, 64, 131, 4, 72, 55])
>>> exit()

MyMacMini:~ xiaofanc$ python
Python 2.7.10 (default, Jul 13 2015, 12:05:58)
[GCC 4.2.1 Compatible Apple LLVM 6.1.0 (clang-602.0.53)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> d = usb.core.find(idVendor=0x0483, idProduct=0x3748)
>>> d.write(0x02, [0xf1, 0x80, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 1000)
16
>>> d.read(0x81, 6, 1000)
array('B', [37, 64, 131, 4, 72, 55])
>>> d.write(0x02, [0xf1, 0x80, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 1000)
16
>>> d.read(0x81, 6, 1000)
array('B', [37, 64, 131, 4, 72, 55])
>>> exit()
```

pavelrevak, 2015-10-17:
I have little older Mac book pro 2010 mid, core2duo with OS/X 10.11 and ST-Link/V2 V2J23S0 (think it is latest firmware)

But, now it is interesting, your code is working also on my Mac!

You have read version two times and in my example only once. If I also read the version two times it will work also in second connection (after disconnect).

But more interesting it is that it works with reading version one time but with 10.10 or any other OS (Windows, Linux)

pavelrevak, 2015-10-18:
Mcuee, why do you try write/read two times and not only once like in my example? Do you know something more about ST-Link protocol? Although this not solve the ticket, this is good workaround. Thanks.

mcuee, 2015-10-18:
I do not know much about the ST-Link protocol. But the way I interpreted
your "run this code second time" is like what I did -- to run the two
commands twice (write once and then read once). So you can say it is by
luck. :-)

This could potentially points to a ST-Link FW limitation if your
write-once-read-twice command sequence is legitimate.

diabolo38, 2015-10-18:
if it happen also on macos 10.6 i may give a try and see what's happening at bus level it than can help. 

if  it show only on macos 10.11  with openocd (that i guess does same things on all os) it is likely  related to what the back end or  os  is doing or timing.

Note that reading 6 byte from a bulk endpoint is not quite correct and safe unless you sure you will get exactly or at most 6 byte back ! you should always initiate read with full bulk packet size (64 byte for this device on ep 0x81) and see how many data you get back. (that will not time due to short packet sent by device)
They are some words about that  on the libusb_bulk_transfer help , also you can have time out and yet have data. 
If you do not proceed that way it is possible to leave data on the device ep fifo that is source of "synch" and time out  issue.

I had a talk with stlink driver developer at st about same type of issue on the virtual com of the nucleo stlink (the issue show on some usb host controller that can't managed non 4 byte multiple xfer and they were reading  2 6 10 ... ie non 64 byte packet from the cdc in ep). That fix has been integrated on the window driver around STSW-LINK007 or 8 release.

for the protocol wr once read twice is certainly to fail if it is wr command rd answer/status.
I also suspect the  f/w does "in ep flush" when it get a new command  wr to ep 0x02 prior it queue any new data to it's in ep 0x81 so that host get "answer right" to that last command. 
But that may not be enough as host can be faster in doing wr  then initiating the rd before device f/w can even react to the initial wr and "flush  recovery" to be effective. 
Placing a small delay after the write could help to get the next read ok  every time  in case the issue is well linked to data left unread in the in end point.

pavelrevak, 2015-10-18:
Don't know about 10.6, but previously I had 10.9, 10.10 and I didn't meet any problems, but in my case and after googling the problem occurs only in 10.11.

Now I trying something else, for example always read 64 bytes and or round to 4 bytes, but still without success. Right now I know only one workaround to avoid timeouts and it is to read version two times during initialization.

I also found another solution which can recover ST-Link device from this 'stupid state':
It is to repeat sequence with write command and read answer more times until return expected data instead of timeout. In my case it is 3 times.

diabolo38, 2015-10-18:
real strange,  something is wrong on the in ep  :( 
i have seen some similar issue on mac earlier but that was constant over all  10.6 10.10   and it was not related to the device (also an stm32 on bulk in ep)  according to what i saw  on the bus (did not resolved it  but work it around as well). 
Since that i revert the mac to 10.6 and want move to 10.11, so can help more sorry.  
Have you report it to ST  ?

mcuee, 2015-10-23:
So this does not seem to be a problem of libusb. I will close this issue.

mcuee, 2015-11-24:
The comment in https://github.com/texane/stlink/issues/331 is interesting that Ubuntu on the same machine does not seem to work as well.

pavelrevak, 2015-11-24:
mcuee, I try to fix this issue in https://github.com/pavelrevak/pystlink project and during this I found some workarounds which can help:
- First I count number of send packets and before disconnecting if this number is **odd** I send one more dummy packet (number of packets must be **even**) then next connection is OK.
- Next if I send or receive longer packet than 64 bytes, then big packets also need to be even number (need count separately from small packets).
- And still it is not all although everything seems OK with **ST-Link/V2** but not with **ST-Link/V2-1** and here is again next workaround: trying to "xfer" command GET_VERSION more times (maximum 3 times) until I get answer from ST-Link.... (but this workaround don't work without previous workarounds)

I don't know much about USB communication, but it is really strange and confusing, especially if it is working on other platforms and older OS/X. may by these workarounds help someone better understand where can be problem.

And also these problems exists not only stlink or openocd but also original firmware updater tool from ST which is also for Linux and OS/X.

mcuee, 2015-11-26:
The work-around seems to be read the ST-Link version twice. Please try that.

As per the report in the OpenOCD mailing list, that seems to sort out the problem.

sajimon, 2015-11-26:
I'm using that workaround for some time now and it's working, but obviously, only with OpenOCD.

pavelrevak, 2015-11-26:
yes, reading version more times help, but not always and still not solve problems on OS/X 10.11

I just sniff communication using oscilloscope and see lot of differences:

Here I try explain it what I sniff: for testing was used python example from first comment of this ticket with possibility to read version more times, changed code is here `stlink_test.py`:

    import sys
    import usb.core

    d = usb.core.find(idVendor=0x0483, idProduct=0x3748)
    try:
        d.write(0x02, [0xf1, 0x80, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 1)
        print(d.read(0x81, 6, 1))
    except usb.core.USBError as e:
        print(e)

if I run this program on windows machine I get this (whats was expected):

    # stlink_test.py
    array('B', [38, 0, 131, 4, 72, 55])

    SNIFFED USB:
    ....
    Time:-0.002710 | PID:SOF   | Frame:1496 | CRC5:07
    Time:-0.001710 | PID:SOF   | Frame:1497 | CRC5:18
    Time:-0.000710 | PID:SOF   | Frame:1498 | CRC5:1a
    Time:-0.000005 | PID:OUT   | Address:26 | Endpoint:2 | CRC5:08
    Time:-0.000002 | PID:DATA0 | Data:f1 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | CRC16:22dd
    Time: 0.000012 | PID:ACK
    Time: 0.000290 | PID:SOF   | Frame:1499 | CRC5:05
    Time: 0.000429 | PID:IN    | Address:26 | Endpoint:1 | CRC5:11
    Time: 0.000432 | PID:DATA0 | Data:26 00 83 04 48 37 | CRC16:e78e
    Time: 0.000439 | PID:ACK
    Time: 0.001290 | PID:SOF   | Frame:1500 | CRC5:1b
    Time: 0.002290 | PID:SOF   | Frame:1501 | CRC5:04
    Time: 0.003290 | PID:SOF   | Frame:1502 | CRC5:06
    ....

but on OS/X 10.11 is after run same code (it's really mess):

    # stlink_test.py
    array('B', [38, 0, 131, 4, 72, 55])

    SNIFFED USB:
    ....
    Time:-0.002913 | PID:SOF   | Frame:944 | CRC5:0d
    Time:-0.001913 | PID:SOF   | Frame:945 | CRC5:12
    Time:-0.000913 | PID:SOF   | Frame:946 | CRC5:10
    Time:-0.000005 | PID:SETUP | Address:1 | Endpoint:0 | CRC5:17
    Time:-0.000002 | PID:DATA0 | Data:80 06 00 01 00 00 12 00 | CRC16:072f
    Time: 0.000007 | PID:ACK
    Time: 0.000012 | PID:IN    | Address:1 | Endpoint:0 | CRC5:17
    Time: 0.000015 | PID:NAK
    Time: 0.000087 | PID:SOF   | Frame:947 | CRC5:0f
    Time: 0.000090 | PID:IN    | Address:1 | Endpoint:0 | CRC5:17
    Time: 0.000093 | PID:DATA1 | Data:12 01 00 02 00 00 00 40 83 04 48 37 00 01 01 02 03 01 | CRC16:ee11
    Time: 0.000108 | PID:ACK
    Time: 0.000114 | PID:OUT   | Address:1 | Endpoint:0 | CRC5:17
    Time: 0.000117 | PID:DATA1 | CRC16:0000 | Errors:CRC16 Error
    Time: 0.000120 | PID:ACK
    Time: 0.001087 | PID:SOF   | Frame:948 | CRC5:11
    Time: 0.002087 | PID:SOF   | Frame:949 | CRC5:0e
    Time: 0.003087 | PID:SOF   | Frame:950 | CRC5:0c
    Time: 0.004087 | PID:SOF   | Frame:951 | CRC5:13
    Time: 0.004434 | PID:OUT   | Address:1 | Endpoint:2 | CRC5:03
    Time: 0.004437 | PID:DATA0 | Data:f1 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | CRC16:22dd
    Time: 0.004450 | PID:ACK
    Time: 0.005087 | PID:SOF   | Frame:952 | CRC5:03
    Time: 0.005643 | PID:IN    | Address:1 | Endpoint:1 | CRC5:1a
    Time: 0.005646 | PID:DATA0 | Data:26 00 83 04 48 37 | CRC16:e78e
[... 134 lines cut ...]
    Time: 0.029579 | PID:SOF   | Frame:1783 | CRC5:18
    Time: 0.030579 | PID:SOF   | Frame:1784 | CRC5:08
    Time: 0.030609 | PID:SETUP | Address:1 | Endpoint:0 | CRC5:17
    Time: 0.030611 | PID:DATA0 | Data:02 01 00 00 81 00 00 00 | CRC16:608b
    Time: 0.030620 | PID:ACK
    Time: 0.030625 | PID:IN    | Address:1 | Endpoint:0 | CRC5:17
    Time: 0.030629 | PID:NAK
    Time: 0.030633 | PID:IN    | Address:1 | Endpoint:0 | CRC5:17
    Time: 0.030637 | PID:DATA1 | CRC16:0000 | Errors:CRC16 Error
    Time: 0.030640 | PID:ACK
    Time: 0.031579 | PID:SOF   | Frame:1785 | CRC5:17
    Time: 0.031710 | PID:SETUP | Address:1 | Endpoint:0 | CRC5:17
    Time: 0.031713 | PID:DATA0 | Data:02 01 00 00 83 00 00 00 | CRC16:e096
    Time: 0.031721 | PID:ACK
    Time: 0.031726 | PID:IN    | Address:1 | Endpoint:0 | CRC5:17
    Time: 0.031730 | PID:NAK
    Time: 0.031734 | PID:IN    | Address:1 | Endpoint:0 | CRC5:17
    Time: 0.031738 | PID:DATA1 | CRC16:0000 | Errors:CRC16 Error
    Time: 0.031741 | PID:ACK
    Time: 0.032579 | PID:SOF   | Frame:1786 | CRC5:15
    Time: 0.033579 | PID:SOF   | Frame:1787 | CRC5:0a
    Time: 0.034579 | PID:SOF   | Frame:1788 | CRC5:14
    ....

This are only examples, I can also measure more cases if someone will interest more...

paulfertser, 2015-11-27:
So sometimes it sends garbage in a CONTROL transaction, sometimes it receives garbage, and all the time it tries to confuse the stlink firmware with useless "clear ENDPOINT_HALT" requests for the endpoints 2 (the one the request version is sent to), 0x81 (the one the request version reply is received from) and 0x83 (seems totally unrelated). Apparently, stlink firmware is crap, but those CRC16 errors are no good either.

Pavel, is it always the same CRC16 error present in one of the CONTROL (endpoint 0) OUT data phases before the bad things start to happen?

I assume when you run OS X on a Mac legally you can expect some customer service, right? Is it possible to show these USB wire-level dumps to the Apple engineers?

hjelmn, 2015-11-27:
It is poosible for developers. We can open a bug on radar.apple.com and link it to this and the other bugs we are tracking. I will do this today.

hjelmn, 2015-11-27:
Radar bug 23677332. Please keep updating this bug with any additional information and lets see what Apple does.

pavelrevak, 2015-11-27:
Error: CRC16 is wrongly reported by my Tektronix, it is not bug, its normal when empty data was send, then CRC16 is zero.

hjelmn, 2015-11-27:
Ah, ok. Then let me update the bug on radar.

pavelrevak, 2015-11-27:
Hello, sorry for this previous confusing with CRC, (please ignore CRC16 error where are no data)
Also today I make more measurements and here my summary...

If I compare communication on Windows with OS/X 10.11 (I don't have MAC with older system to compare it with this) then exists only these differences:
- after open USB device with `libusb` MAC always read descriptor
- after connection close OS/X always CLEAR_FEATURE ENDPOINT_HALT for each existing endpoint

USB analysis on OS/X:

    // get descriptor
    Time:-0.000059 | PID:SETUP | Address:1 | Endpoint:0 | CRC5:17
    Time:-0.000056 | PID:DATA0 | Data:80 06 00 01 00 00 12 00 | CRC16:072f
    Time:-0.000048 | PID:ACK
    Time:-0.000042 | PID:IN | Address:1 | Endpoint:0 | CRC5:17
    Time:-0.000039 | PID:NAK
    Time:-0.000034 | PID:IN | Address:1 | Endpoint:0 | CRC5:17
    Time:-0.000031 | PID:NAK
    // device descriptor
    Time:-0.000026 | PID:IN | Address:1 | Endpoint:0 | CRC5:17
    Time:-0.000023 | PID:DATA1 | Data:12 01 00 02 00 00 00 40 83 04 48 37 00 01 01 02 03 01 | CRC16:ee11
    Time:-0.000008 | PID:ACK
    Time:-0.000002 | PID:OUT | Address:1 | Endpoint:0 | CRC5:17
    Time:0.000001 | PID:DATA1 | CRC16:0000 | Errors:CRC16 Error
    Time:0.000004 | PID:ACK

    // OUT DATA
    Time:0.012197 | PID:OUT | Address:1 | Endpoint:2 | CRC5:03
    Time:0.012200 | PID:DATA0 | Data:f1 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | CRC16:22dd
    Time:0.012214 | PID:ACK
    // IN DATA
    Time:0.013276 | PID:IN | Address:1 | Endpoint:1 | CRC5:1a
    Time:0.013279 | PID:DATA0 | Data:26 00 83 04 48 37 | CRC16:e78e
    Time:0.013286 | PID:ACK

    // CLEAR_FEATURE:ENDPOINT:ENDPOINT_HALT:2
    Time:0.023797 | PID:SETUP | Address:1 | Endpoint:0 | CRC5:17
    Time:0.023800 | PID:DATA0 | Data:02 01 00 00 02 00 00 00 | CRC16:f4aa
    Time:0.023809 | PID:ACK
    Time:0.023814 | PID:IN | Address:1 | Endpoint:0 | CRC5:17
    Time:0.023817 | PID:NAK
    Time:0.023822 | PID:IN | Address:1 | Endpoint:0 | CRC5:17
    Time:0.023825 | PID:DATA1 | CRC16:0000 | Errors:CRC16 Error
    Time:0.023829 | PID:ACK
    // CLEAR_FEATURE:ENDPOINT:ENDPOINT_HALT:81
    Time:0.026758 | PID:SETUP | Address:1 | Endpoint:0 | CRC5:17
    Time:0.026761 | PID:DATA0 | Data:02 01 00 00 81 00 00 00 | CRC16:608b
    Time:0.026770 | PID:ACK
    Time:0.026775 | PID:IN | Address:1 | Endpoint:0 | CRC5:17
    Time:0.026778 | PID:NAK
    Time:0.026788 | PID:IN | Address:1 | Endpoint:0 | CRC5:17
    Time:0.026791 | PID:DATA1 | CRC16:0000 | Errors:CRC16 Error
    Time:0.026794 | PID:ACK
    # CLEAR_FEATURE:ENDPOINT:ENDPOINT_HALT:83
    Time:0.027865 | PID:SETUP | Address:1 | Endpoint:0 | CRC5:17
    Time:0.027868 | PID:DATA0 | Data:02 01 00 00 83 00 00 00 | CRC16:e096
    Time:0.027877 | PID:ACK
    Time:0.027882 | PID:IN | Address:1 | Endpoint:0 | CRC5:17
    Time:0.027886 | PID:NAK
    Time:0.027890 | PID:IN | Address:1 | Endpoint:0 | CRC5:17
    Time:0.027894 | PID:DATA1 | CRC16:0000 | Errors:CRC16 Error
    Time:0.027897 | PID:ACK

(SOF packet is removed from dump)

I don't see on these actions nothing wrong.
So problem seems is inside STLINK and his USB device implementation,
where thanks this extra communication something 'lock' and in next connection it will not work.

From my point of view it it is bug in STLINK device.

[... 142 lines cut ...]
    Time:0.004738 | PID:NAK
    .... (until timeout)
    Time:0.006129 | PID:OUT | Address:1 | Endpoint:2 | CRC5:03
    Time:0.006132 | PID:DATA1 | Data:f1 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | CRC16:22dd
    Time:0.006146 | PID:ACK
    Time:0.007306 | PID:IN | Address:1 | Endpoint:1 | CRC5:1a
    Time:0.007309 | PID:DATA1 | Data:26 00 83 04 48 37 | CRC16:e78e        <- answer
    Time:0.007316 | PID:ACK
    Time:0.007322 | PID:IN | Address:1 | Endpoint:1 | CRC5:1a              <- ignored !?!?
    Time:0.007325 | PID:NAK
    Time:0.007330 | PID:IN | Address:1 | Endpoint:1 | CRC5:1a
    Time:0.007333 | PID:NAK
    Time:0.007339 | PID:IN | Address:1 | Endpoint:1 | CRC5:1a
    Time:0.007342 | PID:NAK
    .... (until timeout)
    Time:0.008670 | PID:OUT | Address:1 | Endpoint:2 | CRC5:03
    Time:0.008673 | PID:DATA0 | Data:f1 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | CRC16:22dd
    Time:0.008687 | PID:ACK
    Time:0.009288 | PID:IN | Address:1 | Endpoint:1 | CRC5:1a
    Time:0.009291 | PID:DATA0 | Data:26 00 83 04 48 37 | CRC16:e78e
    Time:0.009298 | PID:ACK

Generally all problems happened if number of XFERs is ODD, then in next connection first two XFERs fail and any further XFERs are OK. This is one of workaround, but better is to send EVEN number of packets and timeout never happen.

But here also exists other problem which happen during transfer large data (bigger than 64bytes) it has similar behavior but this workaround not help! I will look on this next Monday, because today I have not so much time.

hjelmn, 2015-11-27:
Can you test with Linux as well? It would be interesting to see how that differs. You can also try 10.10 through a VM. Download the installer for Yosemite from the App Store and you can use that to make a VM running 10.10 in VMWare Fusion.

diabolo38, 2015-11-27:
the descriptor read on window is related to libusb finding all device on bus . 
The "clr halt"   on macos look quite not normal ,  at least i would not expect the os layer to do that and leave this to the device driver when needed (some device driver use that to start device internal streaming /data flow ) . 

The strange order and number with good/bad xfer remind me an issue i've seen macos 10.6 .

see http://libusb.6.n5.nabble.com/libusb-Short-Bulk-In-after-Naked-in-issue-under-MacOS-td5714873.html

I'll try stlink on my old mac on monday when back to work and let you know what happen on bus for this osx  version.

hjelmn, 2015-11-27:
The extra CLEAR_FEATURE ENDPOINT_HALT might be causing #96 and #123.

mcuee, 2015-11-28:
Yes that is one possibility.  I remember CLEAR_FEATURE ENDPOINT_HALT caused
problem with some USB PIC FW implementations.

pavelrevak, 2015-11-28:
Ok, clear halt is called by os/x or libusb?
And whats about the ignored answer (os/x or libusb)?

hjelmn, 2015-11-28:
The clear halt is not coming from libusb. I have a feeling it comes from within IOUSBHostFamilywhen an interface is closed. Either from IOUSBHostInterface or IOUSBInterfaceUserClient.

hjelmn, 2015-11-29:
Not sure why the clear halt answer is ignored. Hoping someone from Apple look at this when they get back from vacation next week.

diabolo38, 2015-11-29:
Clear halt may be ignored says stalled  if device is not yet configured or no interface been selected yet...  That is what stm32 lib and all Linux gadget does

diabolo38, 2015-11-30:
I do not see any clear halt by just connecting an stlink v2 to my macbook-pro running OSx 10.6.8

[image]

with my stm32 base device 
[image]
transaction 167 "Get device  descriptor" is related to libusb opening device
The transaction after are normal bulk in/out device operation 

So ther's is "clr halt" with OSX 10.6.8 in normal operation.

A clr halt could be the  result of a "ClearPipeStallBothEnds" call that would be normal after a first "stalled" transaction.

mcuee, 2015-12-09:
Just updated my Mac Mini 2011 (no USB 3.0) to 10.11.2 and the issue is still there.

MyMacMini:stlink xiaofanc$ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.11.2
BuildVersion:	15C50
MyMacMini:stlink xiaofanc$ python stlink_test_1.py 3
[Errno 60] Operation timed out
[Errno 60] Operation timed out
array('B', [38, 0, 131, 4, 72, 55])

diabolo38, 2015-12-09:
back on last log from pavelrevak
that is a successful in with 6 data long packet

Time:0.007309 | PID:DATA1 | Data:26 00 83 04 48 37 | CRC16:e78e   

That shall have end up the read  transfer successfully  instead host is doing  extra in ? 
we claim 6 data so  that is not normal from host .
the same is visible on some previous log

By reading quickly stmicro usb code i could see that clr feature seam well handled,  and shall fix a potential data toggle issue. What seam to be happening here ,  a capture with an analyzer would definitively help here.

I got some information on stlink protocol it is an cmd out response in type protocol 
So it is normal to have failing read without a previous write 
also doing two write without a read is not quite normal .

hjelmn, 2016-01-12:
Apple responded to my bug linked to #123. They claim the issue may be resolved in the latest 10.11.4 beta. Can you give that a try and let me know if the issue is resolved?

shaneburrell, 2016-01-14:
This does seem to be fixed (so far so good) in 10.11.4.

hjelmn, 2016-01-14:
Great! Sounds like #123 and this bug were caused by the same OS issue. I will close both bugs after the official release of 10.11.4. That will allow more time for testing.

Cheong2K, 2016-01-29:
Confirmed, fixed in OSX 10.11.4 beta.

hjelmn, 2016-01-29:
Alright. Since this is not a libusb bug I will close this. Anyone who encounters this bug should upgrade to 10.11.4 final when it is available.

closed 2016-01-29 by hjelmn