libusb/libusb #113: libusb_get_port_numbers() return same path for different USB ports on Windows XP (sp3)

diverger, 2015-10-16:
I have two USB devices with same VID/PID, and same serial number attached on my laptop. I need to separate them in my application. The only solution I can figure out is identify them by their physical location on my laptop. So I intend to use libusb_get_port_numbers() to implement it. But I finally find it always give me same path for them!!! But the same code worked well on another PC with Win7 installed. 

Is it a bug of libusb or a bug of XP? 

BTW, it seems the two port on my laptop are on the same root hub.

mcuee, 2015-10-17:
This is illegal as per USB specification -- the USB serial number should be either unique or no serial number. So you can not expect to get consistent results. In fact, it is said that older version of Windows may crash if you try this.

mcuee, 2015-10-17:
Ref: http://blogs.msdn.com/b/oldnewthing/archive/2004/11/10/255047.aspx
So it is a bug of your usb device and not Windows XP or libusb.

diverger, 2015-10-17:
Sorry, the two device indeed have no *iSerialNumber* filed, in their descriptor, the *iSerialNumber* index both are "0". So it meas they have *no* serial number.

diabolo38, 2015-10-17:
unlike the doc says it's not safe to use libusb_get_port_number  to get an  unique id for a device on a system (it is base on the bus address).  
That fucntion can return same value for two device when system has mutliple controler  for instance what is the case with almost all  since usb3 appeared.
that function doc shall be change  :( 

instead use libusb_get_port_numbers 
the return numbers list should well be a unique as it is the address route from hosts down to device 
you can concatenate the  byte into a unique 64 bit value for ease of uses.
as depth of that route is limited to 7 it shall be safe.

diverger, 2015-10-17:
@diabolo38: l don't quite understand you. I indeed use the function *libusb_get_port_numbers*. But it seems that function behaves different on XP and Win7, at least in my situation.

diabolo38, 2015-10-17:
Oops sorry then I did not seen you already switched to the new one.

I'm not surprised it does not behave the same way on different o's.  I will if the routes return is the same.

mcuee, 2015-10-18:
I think you should be able to get unique path by using both
libusb_get_bus_number and libusb_get_port_numbers. Please look at the
example xusb.
https://github.com/libusb/libusb/blob/master/examples/xusb.c#L821

diabolo38, 2015-10-18:
@mcuee  it seam from 1st post the route return is "1" for both device on xp  :(   what  will not be correct 

@diverger  can you try  the sample xusb to double check routes of your devices ?

I have an old laptop with XP-SP3 i'll  may try to replicate the soem time next week , 
I have only a single device i can reprogram right now with me.  
In any case  I'm scared i would not be able to help much finding  a solution in cas of real issue.  
That  part of the libusb code dealing with enumeration is tricky and cryptic to me.  I did look once for enumeration issue on usb3 ( http://comments.gmane.org/gmane.comp.lib.libusb.devel.general/20945 .)

Ii can already warn you about this enumeration issue on some  usb3 controller  (ti7330-40) under win 7.  From my  understanding ubs3.0 is managed by vendor driver prior to win8 and not all driver seam to implement correctly some ioctl used by libusb (certainly different  to what windows driver and winusb native code uses). Hopefully they are quite no xp machine with that usb3 h/w an only few win7  (HP Z820 is one). 

If you are looking for a replacement for a "window only" driver why not considering direct use of winusb?

diabolo38, 2015-10-18:
Even on win7 port numbers can look disappointing and several same "port route" can appears

listdevs  output on my win7 pc with a couple of usb device 

8086:3b34 (bus 1, device 1)
8086:3b3c (bus 2, device 1)
045e:0766 (bus 2, device 3) path: 1.5
046d:0a18 (bus 1, device 7) path: 1.1
0483:3748 (bus 1, device 5) path: 1.3
04b3:3107 (bus 2, device 4) path: 1.3
0533:0603 (bus 1, device 6) path: 1.4
058f:6361 (bus 1, device 3) path: 1.5
0a12:0001 (bus 1, device 4) path: 1.6
8087:0020 (bus 2, device 2) path: 1
8087:0020 (bus 1, device 2) path: 1

this may look wrong at first 
here is usbview in same config

[image]

that visual view make clear there's no duplicate , what is then wrong ?
nohting wrong to get a unique route we must not forget to add the bus number first. 
Combining bus + address also give a simpler unique  tuple 

here is my old XP-SP3 listdevs output   with all its 3 port fitted 

8086:2658 (bus 1, device 1)
8086:2659 (bus 2, device 1)
8086:265b (bus 4, device 1)
8086:265a (bus 3, device 1)
8086:265c (bus 5, device 1)
0483:3748 (bus 3, device 2) path: 2
04b3:3107 (bus 2, device 2) path: 2
0533:0603 (bus 4, device 2) path: 2

all on path 2 address 2  ,   but all on different bus , strange but why not.
bus_number . device_address will also give different ids 
3.2 
2.2 
4.2

diverger, 2015-10-20:
@diabolo38, you've got the point, below is my situation. You can see, my two ports also locate on different root hubs. So both ports report my '1'.
[image]

[image]

diverger, 2015-10-20:
@mcuee, @diabolo38: Thanks for your suggestions. By compibning *libusb_get_bus_number* and *libusb_get_port_numbers*, I can get the unique paths and identify them. So, it's a *BUG* of my code.

diabolo38, 2015-10-21:
it  wasn't clear on the , doc  libusb_get_port_numbers  seam enough .
I did the exact same use only port numebrs , and i'll have to change code not to fail in similar cases.

it may get worth reviewing the doc .

diverger, 2015-10-21:
@diabolo38: From the name  *libusb_get_port_numbers* only get the *port* number, that say, the port path from the root hub, only *hub* has *ports*. But PC may have several root hub. :(

closed 2015-10-23 by mcuee