libusb/libusb #115: Help:  How to read data from usb device?

jimying, 2015-10-22:
Recently,  I want to use libusb to read data from my Sandisk usb device.
I use async mode to read , but the transfer callback function never be called.
Could anyone tell me why?

My sample code as follow:
```
#include <stdio.h>
#include <libusb.h>

void trans_cb(struct libusb_transfer *transfer){
    fprintf(stderr, "callback!\n");
}

int main(){
    libusb_context *ctx = NULL;
    libusb_init(&ctx);

    libusb_device_handle *handle = libusb_open_device_with_vid_pid(ctx, 0x0781, 0x5406);
    libusb_detach_kernel_driver(handle, 0);
    libusb_claim_interface(handle, 0);
    struct libusb_transfer *transfer = libusb_alloc_transfer(0);
    char buf[18];
    libusb_fill_bulk_transfer(transfer, handle, 0x81, buf, 18,trans_cb, NULL, 0);
    libusb_submit_transfer(transfer);

    while(1)
    {
        libusb_handle_events(ctx);
    }

    libusb_exit(ctx);

    return 0;
}
```

jimying, 2015-10-22:
lsusb tool to get my Sandisk usb device Info:

```
Bus 003 Device 020: ID 0781:5406 SanDisk Corp. Cruzer Micro U3
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0781 SanDisk Corp.
  idProduct          0x5406 Cruzer Micro U3
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```

mcuee, 2015-10-22:
Please use libusb mailing list for technical support. This github issue
tracking is not for technical support.
https://github.com/libusb/libusb/wiki#Support

mcuee, 2015-10-22:
BTW, your code does not make sense for USB Mass Storage device.
https://github.com/libusb/libusb/wiki/FAQ#Can_I_use_libusb_to_open_a_file_on_an_USB_storage_device

closed 2015-10-23 by mcuee

gejamin, 2017-07-09:
Hi,bro,have you already solve the problem?

jimying, 2017-07-11:
USB has 4 type transfer( control transfer,  interrupt transfer, bulk
transfer and Isochronous transfer).    Three step to send request to usb
device.

1. you should know which transfer type you want to request

2.  use libusb_alloc_transfer()  to alloc a transfer request,  and use
 libusb_fill_xxxxxx ()   (Every transfer has diffrent function) to fill
data to transfer

3. use libusb_submit_transfer() to submit request to usb device,

when  usb response,   the  transfer callback function will get it.