libusb/libusb #116: libusb_kernel_driver_active() / libusb_detach_kernel_driver() problems on ubuntu vivid

bvernoux, 2015-10-25:
libusb_kernel_driver_active() res=0 => No airspy kernel driver detected why ?
libusb_detach_kernel_driver() res=-5 => LIBUSB_ERROR_NOT_FOUND = -5 (probably because No airspy kernel driver detected)

For more details see https://github.com/airspy/host/issues/27

Does anyone can help on this issue ?

dickens, 2015-10-26:
Look at the dmesg output from the link:
[ 2000.641372] airspy 1-2:1.0: usb_control_msg() failed -32 request 0a
[ 2000.641375] airspy 1-2:1.0: Could not detect board
[ 2000.641380] airspy: probe of 1-2:1.0 failed with error -32

This means the airspy driver probe failed, which means the driver is indeed not attached. libusb is operating correctly here, and the reason for the driver probe failure is outside the scope of libusb.

closed 2015-10-26 by dickens

bvernoux, 2015-10-26:
Do you have an idea why when we blacklist the kernel driver (airspy) all work fine ?

dickens, 2015-10-26:
Blacklisting the module and the module's driver probe failing both have the same outcome, which is that the driver will not be bound. It does not matter in this case whether the kernel driver is actually loaded or not, and there is a difference between loaded and bound. libusb_kernel_driver_active() only reports whether a kernel driver (other than usbfs) is currently bound to a device interface.