libusb/libusb #117: Short bulk transfer generate a flood of kernel warnings in xhci_hcd

xlz, 2015-10-25:
I'm working on the driver of Kinect v2, which uses USB 3.0. It provides a JPEG endpoint with 30 frames per second, each of which is about 800KB. 

I try to submit a bulk transfer with buffer size of 2MB to be safe. Each transfer completes correctly, but generates a kernel warning like this

```
[ 2709.860400] xhci_hcd 0000:00:10.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
```
It generates 30 such kernel warnings per second. This is clearly not sustainable. The alternative that avoids such flood of kernel warnings is to submit 16KB buffers for one JPEG frame, and then cancel the rest of transfers of this frame upon a short transfer, implementing outside libusb. I think this duplicates the functionality in libusb and defeats its purpose.

"Event TRB" warnings are probably fixed in 4.3-rc7
https://www.spinics.net/lists/linux-usb/msg129940.html
https://www.spinics.net/lists/linux-usb/msg129957.html

But the supposedly supported kernel from 2.6.32 to 4.2 would not have the correctly behavior for short USB 3.0 bulk transfers.

mcuee, 2015-10-26:
I do not understand what you are trying to ask libusb project to do here.
If this is a kernel anomaly and have a application workaround, you can
either upgraded the kernel or try to use the application workaround.

xlz, 2015-10-26:
The bug is what is described in the documentation as "Early transfer completion" is clearly broken for USB 3.0 bulk transfers on a wide range of linux kernel versions.

I could duplicate relevant libusb functionality (correct handling of early completion) outside of libusb, but that defeats the purpose of using libusb in the first place.

I develop a device driver which uses libusb. I can't require my users to upgrade kernel version.

jwrdegoede, 2015-10-26:
We are not going to fix this in libusb, automatic splitting of a transfer in multiple smaller ones and cancelling the rest on completion is not something which libusb should be doing, not at all.

This may even cause issues for some other type of applications.

To me it seems what you're hitting is mostly a kernel bug, there are indeed some cases like transferring a variable size jpeg issue over a bulk transfer, where a short completion of the transfer is normal and the kernel should not be generating 30 messages / second for this.

And the kernel devs agree that this is a bug, and have a patch available, so I do not see the issue. You will simply just have to ask your users to get a fixed kernel, and you can ask the kernel-devs to include the fix in future stable kernel releases. Users should upgrade their kernel to those anyways for security fixes.

Also if your doing a kinect v2 driver you really should be doing a kernel driver (using the gspca framework) and not a userspace driver, so that you can provide a standard v4l2 device, just like is being done with the kinect v1.

closed 2015-10-26 by jwrdegoede

diabolo38, 2015-10-26:
what  host controller are you using ? are you sure the root cause of that  message is an short or early completion ?

I do not see such error on a 3.6.11-fc16.i686 kernel with a ti xhci host and  my USB 3.0 device
using llibusb to do bulk request of 1MB up to 4MB and getting "short" packet  form the device at various offset.)
Some time back in 2013-2014 with older kernel and "cheese" streaming from "ubs3.0 uvc" gadget webcam and i never saw any error message like this one.

xlz, 2015-10-26:
`Intel Corporation 8 Series USB xHCI HC [8086:9c31] (rev 04)`.

Yes, I tested with 4.3-rc7 and no "WARN Event TRB" message was seen. It can be reproduced at least with 4.2.

diabolo38, 2015-10-26:
Ask Intel to fix the driver :)

At least that is not present on ti driver.

xlz, 2015-10-26:
I understand why it's not libusb's job to fix kernel bug. I just have to implement this splitting and cancelling outside of libusb to support users' kernels though.