libusb/libusb #118: MAX_ISO_BUFFER_LENGTH too large for kmalloc

xlz, 2015-10-26:
Right now if an isochronous transfer's length is larger than `MAX_ISO_BUFFER_LENGTH` (49152*128, 49152 is the arbitrary limit set according to USB 3.0 maximum wBytesPerInterval, 128 is an arbitrary packets per URB limit), it is split into multiple URBs with `MAX_ISO_BUFFER_LENGTH` or less bytes. `proc_do_submiturb()` will use `kmalloc()` to allocate a contiguous memory for each isochronous URB. But `MAX_ISO_BUFFER_LENGTH` is too large (`>KMALLOC_MAX_SIZE`) for `kmalloc()`, and `-ENOMEM` would be returned later. 

Among SLAB, SLOB, SLUB allocators, and all archtectures included in mainline kernel, the minimum of `KMALLOC_MAX_SIZE` seems to be 4MB. (Or maybe less? Will do some testing soon)

If `MAX_ISO_BUFFER_LENGTH` is smaller than `KMALLOC_MAX_SIZE` then libusb can correctly split isochronous tranfer of large size. Also it seems kmalloc'ing large buffers is discouraged in general. So smaller iso buffer limit may be desirable.

Kernel usb core also performs a per packet length check on URBs, requiring length to be <= max_iso_packet_size (`-EMSGSIZE` or else). There is also a 64KB limit of TRB size. There may be some error reporting needed for `ENOMEM` and `EMSGSIZE`.

@dickens

closed 2017-12-27 by dickens

xlz, 2017-12-27:
Thank you for the work. But it appears the issue reported is not exactly resolved. Let me try again:

Steps to reproduce: user creates a large libusb iso "transfer" of 256 packets of 49152 bytes, submit the transfer.

Expected: libusb splits the transfer into appropriately small URBs and the transfer completes.

Actual: libusb splits the transfer into 2 URBs of 128*49152 (6MB). The kernel reports ENOMEM because 6MB is too large for kmalloc().

dickens, 2017-12-27:
Yes, I understand that the failure point here is the kmalloc() for the total buffer length. In my opinion, this isn't the responsibility of libusb to handle, it is a kernel policy dictated by the maximum allocation size of the particular allocator in use. If 128*49152 is too large for kmalloc(), perhaps the kernel shouldn't be allowing 128 packets with 49152 as the maximum length.

My change was intended to align libusb with the checking that the kernel does to provide an early screening for failures that would likely occur during URB submission.

I'd be interested to see what the folks on the kernel's linux-usb mailing list have to say about this. If you decide to inquire about this, please CC me.

xlz, 2017-12-27:
For the record I posted to linux-usb here https://marc.info/?l=linux-usb&m=151440529029954.

mcuee, 2021-07-21:
https://marc.info/?l=linux-usb&m=151460083102390&w=2

List:       linux-usb
Subject:    Re: Isochronous URB limits don't work with kmalloc limits
From:       Alan Stern <stern () rowland ! harvard ! edu>
Date:       2017-12-30 2:27:06
Message-ID: Pine.LNX.4.44L0.1712292103210.5056-100000 () netrider ! rowland ! org
[Download RAW message or body]

On Wed, 27 Dec 2017, Lingzhu Xiang wrote:

> Hi,
> 
> At libfreenect2 we are interested in using large buffers for
> isochronous transferring of large depth camera frames.
> 
> proc_do_submiturb() hardcodes two limits for iso transfers: 128
> packets per URB, and 49152 per packet. When I submit a large URB of
> 128 packets of 49152 bytes I will have ENOMEM, because this URB
> allocates a buffer of 128*49152 (6MiB) and it is too large for
> kmalloc.
> 
> The problem is userspace has no good way of querying the maximum
> acceptable size by kmalloc and consequently doesn't know how large a
> transfer is known to work with the kernel.

As far as I know, there is no maximum acceptable size for kmalloc.  The 
kernel tries to allocate the amount of memory you ask for, and if it 
fails then there's not much you can do.

> Actually the 6MiB URB is created via libusb, which tries to split my
> even larger iso transfer into smaller URBs. The best libusb can do is
> use the 128/49152 limits from the kernel and splits my transfer into
> 6MiB URBs (and does not work).
> 
> How to approach this problem? Is there a recommended max transfer size
> for kmalloc somewhere in the USB core? Should the packet number limit
> be lowered for iso transfer? Can the kmalloc limit be queried?

There is no recommended max size for kmalloc in the USB core (or 
anywhere else).

Yes, you can decrease the number of packets you submit at one time.  
This will decrease the size of the memory area that has to be 
allocated, making it more likely that the allocation will succeed.

There is no kmalloc limit; hence you cannot query it.

Slightly related to your questions...  You can use mmap to pre-allocate
a single large buffer (or multiple smaller buffers) and then use that
for all of your transfers.  This is still subject to the problem of not
being able to allocate a large-enough memory region, but:

	It's a one-time allocation; it doesn't need to be repeated
	each time you submit a transfer request.

	It allows for zero-copy operation, since the data doesn't
	need to be copied between a user buffer and a kernel buffer.

Alan Stern

--
To unsubscribe from this list: send the line "unsubscribe linux-usb" in
the body of a message to majordomo@vger.kernel.org
More majordomo info at  http://vger.kernel.org/majordomo-info.html

commit 8aad7fd1 (2017-12-27): linux_usbfs: Improve isochronous transfer submission and error reporting

The Linux kernel has changed the maximum allowed packet length per
isochronous packet numerous times, which can create difficulties in
trying to report appropriate errors back to the user when submitting too
large of a packet on older kernels.

In an attempt to improve this situation, this commit adds logic that
will use different per-packet limits based on the detected kernel
version. Additionally, the logic has been improved to split URBs based
on the number of isochronous packets per URB, which is currently (and
has b [... cut]