libusb/libusb #119: SuperSpeed Isochronous RX Throughput on Linux vs OS X Yosemite ("Easy Peasy")

MichaelABarger, 2015-10-26:
The following code achieves true SuperSpeed isochronous throughput from a custom peripheral device to Mac OS X Yosemite with no issues. The same code fails in Linux upon libusb_submit_transfer:

    [ 4.335298] [0000081a] libusb: debug [libusb_claim_interface] interface 0
    [ 4.335328] [0000081a] libusb: debug [libusb_set_interface_alt_setting] interface 0 altsetting 1
    [ 4.345612] [0000081c] libusb: debug [submit_iso_transfer] need 2 32k URBs for transfer
    [ 4.345651] [0000081c] libusb: error [submit_iso_transfer] submiturb failed error -1 errno=22
    [ 4.345663] [0000081c] libusb: debug [submit_iso_transfer] first URB failed, easy peasy
    [ 4.345675] [0000081c] libusb: debug [disarm_timerfd] 

The device uses 1024B packets, 15 packets per burst, and 3 bursts per microframe. The microframe interval is 1. The only way I was able to exceed a paltry 8~24 MBps was to increase ISO_PACKET_LEN from (1024) to (1024 * 15 * 3). It's exactly this change that causes libusb_submit_transfer to fail on Linux, while it results in good SS throughput on Mac OS X Yosemite.

What am I doing wrong? I am assuming that this is my fault rather than libusb's, but the different behavior between *NIX OSes confounds me. I would appreciate it very much if someone more knowledgeable than me about Linux USB could point me in the right direction. Thank you.

(The below is obviously a partial and edited block of the code for clarity's sake, but it should show everything that is relevant to creating, initializing, and submitting the isochronous transfers in question.)

    static const int ISO_PACKET_CT = 128;
    static const int ISO_PACKET_LEN = 1024 * 15 * 3;
    const unsigned int ISO_BUF_LEN = ISO_PACKET_CT * ISO_PACKET_LEN;
    static const uint8_t ISO_ENDPOINT = 0x83;

    /* START transfers */
    for( i = 0; i < OCTA_ISO_XFER_CT; i++ )
    {
        unsigned char *buf = this->iso_buffer[i];
        struct libusb_transfer *xfer = this->iso_xfer[i];
        struct XferPkg *pkg = &this->iso_pkg[i];

        buf = malloc( ISO_BUF_LEN );
        if( !buf )
        {
            _debug( "Failed to allocate memory" );
            return NULL;
        }
        xfer = libusb_alloc_transfer( ISO_PACKET_CT );
        if( !xfer )
        {
            _debug( "Failed to allocate transfer" );
            return NULL;
        }

        libusb_fill_iso_transfer( xfer, this->h, ISO_ENDPOINT, buf, ISO_BUF_LEN, ISO_PACKET_CT, _iso_cb, pkg, 0 );
        libusb_set_iso_packet_lengths( xfer, ISO_PACKET_LEN );
        retval = libusb_submit_transfer( xfer );
        if( retval != 0 )
        {
            _debug( "Failed to submit transfer" );
            return NULL;
        }
    }

xlz, 2015-10-28:
First, if bMaxBurst is 15, then you should use 16 (bMaxBurst + 1).

1024*15*3*128 is 5.625MB. The buffer is too large for linux kernel to allocate. See #118. For now you can use ISO_PACKET_CT = 64.

MichaelABarger, 2015-10-28:
Thank you very much for your response!

The burst size is indeed 15; bMaxBurst is 14. I should have been clearer about that.

Unfortunately, using 64, 32, or even 1 for ISO_PACKET_CT results in identical behavior (failure with same message). The only way to not get this error message is to use a reduced number for ISO_PACKET_LEN: the largest value that will work is 1024 * 10 * 3 (instead of 1024 * 15 * 3); even then, the throughput is very poor, only achieving 40.5 MBps throughput instead of the 350+ MBps that the above code achieves on OS X.

The PC I am using for this test is a recently-built higher-end model, running a fresh install of Ubuntu 15.04. This same code on Mac OS X Yosemite runs at the peak isochronous SuperSpeed throughput, but fails entirely on Linux unless ISO_PACKET_LEN is reduced, in which case throughput is poor; Does libusb-1.0 require any sort of OS configuration or environment variables set to take advantage of SuperSpeed functionality?

xlz, 2015-10-28:
> need 2 32k URBs for transfer

I don't think this is coming from the latest code.

The latest code has
```
usbi_dbg("need %d %dk URBs for transfer", num_urbs, MAX_ISO_BUFFER_LENGTH / 1024);
//and
#define MAX_ISO_BUFFER_LENGTH		49152 * 128
```

MichaelABarger, 2015-10-28:
Correct, this was one version previous to current, from apt-get. At your suggestion, I went ahead and built the latest version directly from the GitHub source and relinked against it.

It did not immediately work (same error), but I changed ISO_PACKET_CT from 32 to 16, and once I did--it worked, and I was getting good throughput!

I'm not sure if the inability to run with ISO_PACKET_CT with 32 is a bug or not, but even so I am happy since I am getting the throughput that I needed.

Thank you!

closed 2015-10-28 by MichaelABarger

xlz, 2015-10-28:
It should work with 32. May have to do with kernel version too.
In any case 32 is better than 16 because 16 means more syscalls and more prone to delay and packet loss.

MichaelABarger, 2015-10-28:
My Linux kernel version is 3.19.0-28-generic . Definitely didn't work with 32, nor even 24--had to go all the way down to 16 or else the error that this ticket was created for would occur.

I'd much rather be able to run with 32 for the reasons you posed, but with 16 I get maximum throughput and at 24 or 32 it's just a critical error and nothing.

xlz, 2015-10-28:
does your dmesg have anything relevant?

MichaelABarger, 2015-10-28:
Only thing that it has after running with 32 is:

    xhci_hcd 0000:00:14.0: WARN Event TRB for slot 4 ep 6 with no TDs queued?

Not sure if that helps us in any way, but it's all we've got.

xlz, 2015-10-30:
What is the value of your `OCTA_ISO_XFER_CT`? The default usbfs memory limit is 16MB. Your `OCTA_ISO_XFER_CT*ISO_BUF_LEN + lots of overhead` needs to be less than 16MB by default.

Maybe `echo 64 >/sys/module/usbcore/parameters/usbfs_memory_mb`?

MichaelABarger, 2015-10-30:
OCTA_ISO_XFER_CT was 16. I assumed it wasn't the issue, because it appeared that this libusb error was being generated on the first initiated transaction, but after playing with the numbers, I found that any setting with a value of 11 or less would allow it to work with ISO_PACKET_CT at 32.

Thanks for your help, Xiang!