libusb/libusb #121: Parallel build is broken

mispix, 2015-11-07:
When building latest libusb with "make -j 4", I'm getting:

...
Making all in libusb
make[2]: Entering directory `/home/mp/work/sigrok/libusb/libusb'
  CC       libusb_1_0_la-core.lo
  CC       libusb_1_0_la-descriptor.lo
  CC       libusb_1_0_la-io.lo
make[3]: Entering directory `/home/mp/work/sigrok/libusb/libusb'
  CC       libusb_1_0_la-core.lo
  CC       libusb_1_0_la-strerror.lo
  CC       libusb_1_0_la-sync.lo
  CC       libusb_1_0_la-descriptor.lo
mv: cannot stat ‘.deps/libusb_1_0_la-core.Tpo’: No such file or directory
make[2]: *** [libusb_1_0_la-core.lo] Error 1
make[2]: *** Waiting for unfinished jobs....

LudovicRousseau, 2015-11-07:
It is a known problem. But nobody proposed a solution yet.
see http://sourceforge.net/p/libusb/mailman/message/34456286/

mcuee, 2015-12-09:
Chris posted the fix here. We also need Haiku users to test it though.
http://libusb.6.n5.nabble.com/libusb-PATCH-Misc-Simplify-Haiku-build-and-fix-broken-parallel-build-td5715358.html

dickens, 2015-12-09:
I tested on Haiku and was able to build it and run listdevs in the examples.

>

dickens, 2015-12-10:
Maybe @akshay1994 can confirm?

LudovicRousseau, 2015-12-10:
@dickens I guess you also tested your patch on GNU/Linux?

I tried to extract you patch from the email to the list but it is not easy.
Do you have a git branch with your patch somewhere? or a Pull Request?

dickens, 2015-12-10:
@LudovicRousseau Yes, I tested on GNU/Linux. I've attached the patch here, so hopefully it is easy to download.

[patch.txt](https://github.com/libusb/libusb/files/58809/patch.txt)

LudovicRousseau, 2015-12-10:
The patch works for me. I propose to apply it.

commit 3dc781ce (2015-12-17): Misc: Simplify Haiku build and fix broken parallel build

The Haiku build was previously being done as a nested package
due to its C++ requirement, but for some reason setting SUBDIR
in an automake conditional breaks parallel builds. To fix this
and simplify the Haiku build process, this commit adds an
unconditional check for a C++ compiler using AC_PROG_CXX and
builds the Haiku sources as part of the main libusb project.

Note that AC_PROG_CXX itself does not cause the configure script
to fail if a C++ compiler is not found. Therefore on non-Haiku
platforms there is no requirement to have a C [... cut]
closed 2015-12-17 by dickens