libusb/libusb #123: libusb_bulk_transfer never returns on OS X El Capitan

samw3, 2015-11-15:
This project https://github.com/vdudouyt/minipro doesn't work for me.  I did some basic debugging and traced it down to a libusb_bulk_transfer call that just never returns.  Any ideas?

mcuee, 2015-11-15:
You need to provide more details including the debug log to show how you
trace it down.

mcuee, 2015-11-15:
It will be good to provide the log before OS X 10.11 as well if the
program works there.

samw3, 2015-11-15:
I don't really have a good setup for debugging C programs.  I just put a bunch of printf's in before all the calls and then recompiled.  How can I best get logs for you?

samw3, 2015-11-15:
Here's the log with LIBUSB_DEBUG=4

Let me know if you need something else.

```

samw3 minipro $ ./minipro -p 'W27C512 @DIP28' -r "test.hex"
a
b
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000010] [0000050f] libusb: debug [libusb_init] created default context
[ 0.000038] [0000050f] libusb: debug [libusb_init] libusb v1.0.20.11014
[ 0.000167] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.000226] [0000050f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x0
[ 0.001237] [0000050f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.001249] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.001254] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0100
[ 0.001258] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.001262] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0xff
[ 0.001266] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x00
[ 0.001270] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x08
[ 0.001273] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.001277] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8005
[ 0.001281] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0198
[ 0.001285] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.001288] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.001292] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.001298] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.001303] [0000050f] libusb: debug [darwin_check_configuration] ignoring configuration on root hub simulation
[ 0.001308] [0000050f] libusb: debug [process_new_device] allocating new device in context 0x7ff65b600020 for with session 0x0
[ 0.001314] [0000050f] libusb: debug [process_new_device] found device with address 0 port = 0 parent = 0x0 at 0x7ff65b6010ba
[ 0.001389] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1a4ccdfb11b8f
[ 0.001433] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x1a4ccdfb11b8f against cached device with sessionID 0x0
[ 0.001440] [0000050f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1a4ccdfb11b8f
[ 0.034672] [0000050f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.034701] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.034705] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.034709] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.034713] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.034716] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x00
[ 0.034720] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x08
[ 0.034723] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0a5c
[ 0.034729] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x4500
[ 0.034732] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0100
[ 0.034736] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.034740] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.034744] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.034747] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.034815] [0000050f] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.034826] [0000050f] libusb: debug [process_new_device] allocating new device in context 0x7ff65b600020 for with session 0x1a4ccdfb11b8f
[ 0.034835] [0000050f] libusb: debug [process_new_device] found device with address 2 port = 0 parent = 0x0 at 0x7ff65b5059ca
[ 0.034995] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1a4ccee9756e6
[ 0.035066] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x1a4ccee9756e6 against cached device with sessionID 0x1a4ccdfb11b8f
[ 0.035075] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x1a4ccee9756e6 against cached device with sessionID 0x0
[ 0.035080] [0000050f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1a4ccee9756e6
[ 0.069302] [0000050f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.069330] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.069337] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.069417] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0xff
[ 0.069432] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x01
[... cut ...]
[60.080456] [0000050f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[60.080466] [0000050f] libusb: debug [handle_events] poll() 1 fds with timeout in 60000ms
[120.085196] [0000050f] libusb: debug [handle_events] poll() returned 0
[120.085282] [0000050f] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[120.085304] [0000050f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[120.085317] [0000050f] libusb: debug [handle_events] poll() 1 fds with timeout in 60000ms

```

samw3, 2015-11-15:
I'm not sure if this helps, but here is what OS X reports about the device in the system report:

MiniPro TL-866 Programmer  :

  Product ID:	0xe11c
  Vendor ID:	0x04d8  (Microchip Technology Inc.)
  Version:	0.00
  Speed:	Up to 12 Mb/sec
  Manufacturer:	www.autoelectric.com     
  Location ID:	0x14100000 / 15
  Current Available (mA):	1000
  Current Required (mA):	100
  Extra Operating Current (mA):	0

diabolo38, 2015-11-16:
on the log i can see a first successful xfer 
[ 0.075253] [0000050f] libusb: debug [usbi_handle_transfer_completion] transfer 0x7ff65b505658 has callback 0x10ca5aa60
[ 0.075260] [0000050f] libusb: debug [sync_transfer_cb] actual_length=5

then next transfer never complete and keep timing-out 
I guess you have issued a synchronous i/o  with infinite or long timeout to do some read and it is never ending.  
Are you sure the device is working and is ready to send/receive data ?

samw3, 2015-11-16:
I can use the device fine from within my windows VM, so I know the device works just fine.  

As far as it being ready, I just rebooted the computer, plugged it in and waited a couple seconds.  It appears fine in the system report.  

But when I run the software it times out.  

If I then start up the VM and run it's software,  the device works ok in "Windows".

So it appears the device is working fine...

mcuee, 2015-11-18:
Thanks for the updates. Does it work with previous of Mac OS X? If yes,
please provide the debug logs as well. If not, then that is a different
story.

samw3, 2015-11-18:
Ok, I tried it on my Mac Mini which runs Yosemite (OS X 10.10.3), and the minipro program works fine with the libusb 1.0.20 installed using http://brew.sh.  This is what I initially tried on my El Capitan machine.

Here is the first part of the LIBUSB_DEBUG=4 log, up to where has read a few packets as it's reading the chip.

```

Mini:minipro samw3$ export LIBUSB_DEBUG=4
Mini:minipro samw3$ ./minipro -p "W27C512 @DIP28" -r test.bin
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000010] [00000513] libusb: debug [libusb_init] created default context
[ 0.000068] [00000513] libusb: debug [libusb_init] libusb v1.0.20.11004
[ 0.000308] [00000513] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x60cfacaf
[ 0.000353] [00000513] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x60cfacaf
[ 0.001744] [00000513] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.001766] [00000513] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.001773] [00000513] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.001778] [00000513] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.001784] [00000513] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.001791] [00000513] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.001797] [00000513] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.001802] [00000513] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.001807] [00000513] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8006
[ 0.001812] [00000513] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0200
[ 0.001818] [00000513] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x02
[ 0.001848] [00000513] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x01
[ 0.001859] [00000513] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.001865] [00000513] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.001904] [00000513] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.001911] [00000513] libusb: debug [process_new_device] allocating new device in context 0x7f8c5b409030 for with session 0x60cfacaf
[ 0.001959] [00000513] libusb: debug [process_new_device] found device with address 1 port = 0 parent = 0x0 at 0x7f8c5b40959a
[ 0.002020] [00000513] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x613d4191
[ 0.002054] [00000513] libusb: debug [darwin_get_cached_device] matching sessionID 0x613d4191 against cached device with sessionID 0x60cfacaf
[ 0.002062] [00000513] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x613d4191
[ 0.002513] [00000513] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002526] [00000513] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002532] [00000513] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.002538] [00000513] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002543] [00000513] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002548] [00000513] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.002553] [00000513] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002559] [00000513] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.002565] [00000513] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8006
[ 0.002570] [00000513] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0200
[ 0.002575] [00000513] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x02
[ 0.002609] [00000513] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x01
[ 0.002619] [00000513] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002625] [00000513] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002665] [00000513] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002673] [00000513] libusb: debug [process_new_device] allocating new device in context 0x7f8c5b409030 for with session 0x613d4191
[ 0.002681] [00000513] libusb: debug [process_new_device] found device with address 1 port = 0 parent = 0x0 at 0x7f8c5b50007a
[ 0.002734] [00000513] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x64d31265
[ 0.002766] [00000513] libusb: debug [darwin_get_cached_device] matching sessionID 0x64d31265 against cached device with sessionID 0x613d4191
[ 0.002774] [00000513] libusb: debug [darwin_get_cached_device] matching sessionID 0x64d31265 against cached device with sessionID 0x60cfacaf
[ 0.002866] [00000513] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x64d31265
[ 0.003357] [00000513] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003369] [00000513] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[... cut ...]
[ 0.410785] [00000513] libusb: debug [darwin_handle_transfer_completion] handling bulk completion with kernel status 0
[ 0.410794] [00000513] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f8c5b40c1d8 has callback 0x10ea78068
[ 0.410802] [00000513] libusb: debug [sync_transfer_cb] actual_length=256
[ 0.410813] [00000513] libusb: debug [libusb_free_transfer] transfer 0x7f8c5b40c1d8
[ 0.410827] [00000513] libusb: debug [libusb_release_interface] interface 0
Reading Code...  0%[ 0.415721] [00000513] libusb: debug [libusb_claim_interface] interface 0

```

hjelmn, 2015-11-18:
I am starting to think there are serious bugs in the new version of IOUSBFamily. Unfortunately the source is not available. I havent been able to find a debug version either. I will check again today.

hjelmn, 2015-11-18:
No luck. Looks like Apple no longer provides either the source or the log versions.

hjelmn, 2015-11-18:
I don't even see any async completion events (not even timeouts) in the 10.11 log. Thats a little odd. Let me see if I can determine what Apple could have broken.

hjelmn, 2015-11-18:
Can you try running on 10.11 with the following line removed from darwin_usb.c:1267

```
    CFRelease (cInterface->cfSource);
```

That line shouldn't be a problem but I can't find any documentation on whether we should be releasing the async event source.

samw3, 2015-11-18:
I commented it out, but I found it on line 1275.  Recompiled and rebuilt the mini pro against it, but unfortunately the results are the same.

hjelmn, 2015-11-18:
Ok, since a bulk transfer completed before the interface is closed I want to start checking anything that could possibly get screwed up.

samw3, 2015-11-18:
Ok cool. I'll be happy to test. Just let me know how/where to pull any changes.

samw3, 2015-11-21:
@hjelmn I just tried your latest commit, and unfortunately it doesn't fix things yet.  Here is a new log in case there is some more telling info in it.

```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000007] [0000050f] libusb: debug [libusb_init] created default context
[ 0.000042] [0000050f] libusb: debug [libusb_init] libusb v1.0.20.11015
[ 0.000285] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1a4ccdfb11b8f
[ 0.000313] [0000050f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1a4ccdfb11b8f
[ 0.034565] [0000050f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.034600] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.034606] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.034609] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.034613] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.034616] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x00
[ 0.034620] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x08
[ 0.034623] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0a5c
[ 0.034627] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x4500
[ 0.034632] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0100
[ 0.034636] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.034639] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.034642] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.034646] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.034714] [0000050f] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.034726] [0000050f] libusb: debug [process_new_device] allocating new device in context 0x7fea50406c00 for with session 0x1a4ccdfb11b8f
[ 0.034753] [0000050f] libusb: debug [process_new_device] found device with address 2 port = 3 parent = 0x0 at 0x7fea5050014a
[ 0.034962] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1a4ccee9756e6
[ 0.034991] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x1a4ccee9756e6 against cached device with sessionID 0x1a4ccdfb11b8f
[ 0.034997] [0000050f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1a4ccee9756e6
[ 0.069617] [0000050f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.069648] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.069655] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.069660] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0xff
[ 0.069665] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x01
[ 0.069670] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.069676] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.069681] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.069686] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x828f
[ 0.069691] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0112
[ 0.069697] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.069706] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.069712] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.069717] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.069781] [0000050f] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.069789] [0000050f] libusb: debug [process_new_device] allocating new device in context 0x7fea50406c00 for with session 0x1a4ccee9756e6
[ 0.069800] [0000050f] libusb: debug [process_new_device] found device with address 3 port = 3 parent = 0x0 at 0x7fea50600daa
[ 0.069983] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x87df9203a7c3d
[ 0.070015] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x87df9203a7c3d against cached device with sessionID 0x1a4ccee9756e6
[ 0.070023] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x87df9203a7c3d against cached device with sessionID 0x1a4ccdfb11b8f
[ 0.070030] [0000050f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x87df9203a7c3d
[ 0.071307] [0000050f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.071337] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.071345] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.071351] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x00
[... cut ...]
[ 0.077299] [0000050f] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[ 0.077303] [0000050f] libusb: debug [ep_to_pipeRef] pipe 2 on interface 0 matches
[ 0.077362] [0000050f] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.077371] [0000050f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.077376] [0000050f] libusb: debug [handle_events] poll() 1 fds with timeout in 60000ms
```

hjelmn, 2015-11-21:
damn. was hoping it would make a difference. will keep looking

hjelmn, 2015-11-22:
MiniPro claims the interface before each transfer then closes it again. Can you try moving the claim (https://github.com/vdudouyt/minipro/blob/master/minipro.c#L63) to after the libusb_open_device_with_vid_pid (https://github.com/vdudouyt/minipro/blob/master/minipro.c#L26) and the release (https://github.com/vdudouyt/minipro/blob/master/minipro.c#L67) to before the call to libusb_close (https://github.com/vdudouyt/minipro/blob/master/minipro.c#L26)? It should work as is but I want to see if that makes a difference. Might help locate where the bug is in either libusb or IOUSBFamily.

samw3, 2015-11-22:
Ok, finally something different!  After making those changes to minipro.c, if I run the EPROM read command, it works *once*.  It reads the ROM and then exits as it should to the terminal prompt.  However, if I run the same command again, then I get the original timeout error

Here's the successful log and below is the timeout log.
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000006] [0000050f] libusb: debug [libusb_init] created default context
[ 0.000042] [0000050f] libusb: debug [libusb_init] libusb v1.0.20.11015
[ 0.000273] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1a4ccdfb11b8f
[ 0.000301] [0000050f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1a4ccdfb11b8f
[ 0.035326] [0000050f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.035369] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.035376] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.035381] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.035386] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.035391] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x00
[ 0.035396] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x08
[ 0.035401] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0a5c
[ 0.035406] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x4500
[ 0.035411] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0100
[ 0.035416] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.035422] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.035427] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.035432] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.035502] [0000050f] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.035512] [0000050f] libusb: debug [process_new_device] allocating new device in context 0x7fac18c06c00 for with session 0x1a4ccdfb11b8f
[ 0.035521] [0000050f] libusb: debug [process_new_device] found device with address 2 port = 3 parent = 0x0 at 0x7fac18e0130a
[ 0.035681] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1a4ccee9756e6
[ 0.035704] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x1a4ccee9756e6 against cached device with sessionID 0x1a4ccdfb11b8f
[ 0.035709] [0000050f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1a4ccee9756e6
[ 0.070152] [0000050f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.070184] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.070195] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.070201] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0xff
[ 0.070206] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x01
[ 0.070211] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.070217] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.070222] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.070227] [0000050f] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x828f
[ 0.070233] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0112
[ 0.070238] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.070266] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.070275] [0000050f] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.070279] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.070330] [0000050f] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.070335] [0000050f] libusb: debug [process_new_device] allocating new device in context 0x7fac18c06c00 for with session 0x1a4ccee9756e6
[ 0.070344] [0000050f] libusb: debug [process_new_device] found device with address 3 port = 3 parent = 0x0 at 0x7fac18e0212a
[ 0.070457] [0000050f] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x8cc17e5bf019a
[ 0.070479] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x8cc17e5bf019a against cached device with sessionID 0x1a4ccee9756e6
[ 0.070484] [0000050f] libusb: debug [darwin_get_cached_device] matching sessionID 0x8cc17e5bf019a against cached device with sessionID 0x1a4ccdfb11b8f
[ 0.070488] [0000050f] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x8cc17e5bf019a
[ 0.071565] [0000050f] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.071585] [0000050f] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[... cut ...]
[ 0.079168] [0000050f] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[ 0.079171] [0000050f] libusb: debug [ep_to_pipeRef] pipe 2 on interface 0 matches
[ 0.079205] [0000050f] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 0.079213] [0000050f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.079217] [0000050f] libusb: debug [handle_events] poll() 1 fds with timeout in 60000ms
```

samw3, 2015-11-22:
@hjelmn, I should also mention that if I unplug the device and then plug it back in, I get another successful run, and then it's back to the timeout until I unplug again.

hjelmn, 2015-11-22:
ok. that sounds like a bug in IOUSBFamily. The device stops working after the interface is closed. I will double check libusb then look at opening a bug on Apple radar.

hjelmn, 2015-11-22:
btw. can you look st the system log and see if anything is printed when the device stops working.

samw3, 2015-11-22:
I don't see anything in system.log, but I did notice something when I run ioreg:

Here it is when I first plug in the device:
```
samw3 log $ ioreg -b | grep -i minipro
    | | | |   | +-o MiniPro TL-866 Programmer  @14100000  <class IOUSBHostDevice, id 0x10001d908, registered, matched, active, busy 0 (6 ms), retain 20>
      |     +-o MiniPro TL-866 Programmer  @14100000  <class AppleUSBDevice, id 0x10001d90a, registered, matched, active, busy 0 (20 ms), retain 11>
```

This is it after the first run:
```
samw3 log $ ioreg -b | grep -i minipro
    | | | |   | +-o MiniPro TL-866 Programmer  @14100000  <class IOUSBHostDevice, id 0x10001d908, registered, matched, active, busy 0 (6 ms), retain 22>
      |     +-o MiniPro TL-866 Programmer  @14100000  <class AppleUSBDevice, id 0x10001d90a, registered, matched, active, busy 0 (20 ms), retain 13>
```

This is it after I ctrl-C the second run:
```
samw3 log $ ioreg -b | grep -i minipro
    | | | |   | +-o MiniPro TL-866 Programmer  @14100000  <class IOUSBHostDevice, id 0x10001d908, registered, matched, active, busy 0 (6 ms), retain 24>
      |     +-o MiniPro TL-866 Programmer  @14100000  <class AppleUSBDevice, id 0x10001d90a, registered, matched, active, busy 0 (20 ms), retain 15>
```

I noticed the retain counts were going up.  Could that be a problem?

hjelmn, 2015-11-23:
That is a little odd but I think it is harmless. I am testing libusb with an old Diamond Rio S10 and the reference count is > 200 after a single run.

mcuee, 2015-11-23:
If your application works after unplug and plug it back in, it could be
related to device FW bug as well. A possible work-around is to issue a
 libusb_reset_device() at the end of the program. It is a bit heavy-handed
but it deals with such situation well. One of the potential FW bug (could
be linked to HW USB silicon design) can be related to data toggle handling
problem.

samw3, 2015-11-23:
@mcuee I just tried it and it still hangs the second time.  I put it after the libusb_release_interface and before the libusb_close.  Is that right?

```
[ 1.604991] [0000050f] libusb: debug [libusb_release_interface] interface 0
[ 1.606539] [0000050f] libusb: debug [libusb_reset_device] 
[ 1.606616] [0000050f] libusb: debug [darwin_reset_device] darwin/reset_device: checking if device descriptor changed
[ 1.607024] [0000050f] libusb: debug [darwin_reset_device] darwin/reset_device: checking if configuration descriptor 0 changed
[ 1.607408] [0000050f] libusb: debug [darwin_reset_device] darwin/reset_device: device reset complete
[ 1.607430] [0000050f] libusb: debug [libusb_close] 
```

mcuee, 2015-11-23:
No. call  libusb_reset_device() at the end of the program.

mcuee, 2015-11-23:
I mean you need to make sure you call it  before the last libusb_close().

samw3, 2015-11-23:
There is only one libusb_close in this program, and the reset is right before that.  So, yeah, that doesn't solve anything.

mcuee, 2015-11-23:
I see. In that case, then there could be other problems.

mcuee, 2015-11-23:
Ref: https://github.com/libusb/libusb/issues/111

You can see from the above that there could be potential FW problem
which OS X 10.11 seems to be less tolerant. But it could be libusb or
Mac OS X problem as well.

hjelmn, 2015-11-23:
This is very likely an IOUSBFamily problem. I do not see even a timeout event in the posted log. Still trying to figure out how we can debug this. Apple (for whatever reason) has decided to close source their stack and has dropped the log releases.

mcuee, 2015-11-24:
Not so sure if the following brute force method can help to OP.

http://lists.apple.com/archives/usb/2015/Jan/msg00004.html

mcuee, 2015-11-24:
Just tried it and usbtrace mentioned in the above link does not seem
to work for me under 10.11.

It seems to me we may need to re-open Issue 111.

mcuee, 2015-11-24:
Just re-opened Issue 111.

diabolo38, 2015-11-25:
For info On similar issue with an Stm32 I tried patching darwin by using the "reset both end " method thar normaly solve data toggle but it did not work for me

mcuee, 2015-11-27:
Now the issue is very similar to Issue 111 (first run okay, 2nd run not okay). A HW USB analyzer may help.

hjelmn, 2015-11-27:
@mcuee I agree. We need to open a bug on radar.apple.com and provide hardware traces comparing 10.10 and 10.11. Its the only way we are going to get Apple to fix it.

hjelmn, 2015-11-27:
Apple Radar bug 23677332. I linked that bug this this and the other 2 we are tracking. Please continue to update as you get additional information.

samw3, 2015-11-29:
I've found out the chip handling the USB connection in this device is a Microchip PIC18f87j50.  Not sure if that adds anything.  Also, USB analyzers are expensive!  Is there anything else I can do to help, or are we just waiting on Apple?

mcuee, 2015-11-29:
No problem. I have a PIC18F87J50 USB PIM. I will see if I can reproduce
similar problems with my HW. All in all, I think this is more like an Apple
Mac OS X bug and not a libusb bug.

ubuntuslave, 2015-12-02:
I was having the following error when trying to use a USB 3.0 camera in El Capitan with `libusb 1.0.20`:

    libdc1394 error: usb: Bulk transfer 0 failed with code 6

As a quick hack (for now), I got rid of this problem when tested the same camera on another Macbook with USB 2.0 ports, and it worked. Thus, instead of using the USB 3.0 port, I'm going through a USB 2.0 hub, and it works fine like that (with the caveat of lower transfer rates).

hjelmn, 2015-12-08:
Can you test and see if the issue still exists in 10.10.2?

ubuntuslave, 2015-12-09:
Great! My camera now works directly via the USB 3.0 port after upgrading to OS X 10.11.2

Thanks!

samw3, 2015-12-09:
The minipro still does not work properly.  It works only once as it did previously, then I need to unplug/replug it to get it to work one more time.

mcuee, 2015-12-09:
This issue is similar to Issue 111 and upgrading to OS X 10.11.2 does not help there.

ubuntuslave, 2015-12-17:
It worked for a week after the upgrade, and now it's  **back to the same issue**. I'm still bypassing the USB 3 by using a USB 2 hub.

hjelmn, 2016-01-05:
Still waiting on Apple's response. I originally created a single bug pointing at all three libusb bugs (since they are likely all related) but that was closed by Apple with a request that there be 1 Apple bug for each libusb bug. I opened one pointing to this bug and have not seen any movement yet. Hoping for something in the next week or so.

cstawarz, 2016-01-08:
> It works only once as it did previously, then I need to unplug/replug it to get it to work one more time.

I've been having a very similar problem under El Capitan with a [LabJack U6](https://labjack.com/products/u6).  My Mac app communicates with the device via @labjack's [exodriver](https://github.com/labjack/exodriver), which in turn uses libusb.  After unplugging and replugging the device, I can use my application once, but subsequent runs fail.

However, I have found a strange workaround:  If my first connection attempt fails, and I then immediately try two more times (*without* calling libusb_close or making any other changes), the *third* attempt reliably succeeds.  At the USB level, each attempt involves a single bulk write followed by a bulk read.  It doesn't seem to be a question of timing, as simply putting a delay (even a multi-second one) between the first and second attempts has no effect.

I have no idea why this works, and it may just be a quirk of this particular device, but I thought it was worth sharing.

hjelmn, 2016-01-12:
Update. Looks like Apple may have this fixed in the latest 10.11.4 beta. Can you access the beta to test?

samw3, 2016-01-12:
Great to hear.  However, I don't have a machine I can put an OS X beta on to test.  But I'll be sure to try it once 10.11.4 is released.

jlwebster, 2016-01-14:
Hey folks, I'm not in a position to properly test this fully, but I am using texane/stlink and it is now working under OS X 10.11.4 beta! Yay! Been waiting for this for a long time now.

hjelmn, 2016-01-14:
Public beta for 10.11.4 is now out.

Cheong2K, 2016-01-29:
Confirmed, fixed in OSX 10.11.4 beta.

hjelmn, 2016-02-19:
Going to close this as it is an OS issue. 10.11.4 should be out shortly.

closed 2016-02-19 by hjelmn