libusb/libusb #125: larger xfer size than 4096 bytes on windows platform

iddq, 2015-11-19:
Hello,

Can you please test it on windows7/cygwin?

if I set the xfer size to larger than 4096 bytes, I got an error message libusb_control_msg returned -2

Thanks

diabolo38, 2015-11-19:
This is  not yet clearly documented  but is normal 
the max control size is 4K on window as on linux (mac?).
that is linked to some h/w and s/w limitation.

iddq, 2015-11-19:
I tried with more than 4K with an native win application and that worked fine.
If I try this from cygwin or from vmware or virtualbox with linux guests it doesn't work.
I'm not sure the problem is not in the libusb windows port.
If you can please test it sometime.

hiyuh, 2015-11-20:
maybe, i'd have to say SAMEHERE(TM).
i'll recheck this 4KiB issue at some time.

iddq, 2015-11-20:
I experience problem with control transfer

diabolo38, 2015-11-20:
That is the known limit

issue for ken control in virtual bow is likely virtual bow related even if I can confirm several MB bulk can is ok on virtual box windows and Mac host

iddq, 2015-11-20:
sorry I don't understand this limitation.
It works well under native linux. (I did not try it personally but I got this information from a trusted sorce)
but It does not work under windows/cygwin, virtualbox and vmware with linux guest
and I tried this scenario with a native windows application and it works well also.

diabolo38, 2015-11-20:
don't try to understand the limit it is here and exist  even at os level I already doubt about more than 4K  control on window and linux with most USB controler (non usb3)

see Alan answer on  https://sourceforge.net/p/libusb/mailman/message/33339533/ and what Microsoft say 
https://msdn.microsoft.com/en-us/library/windows/hardware/ff538112%28v=vs.85%29.aspx?f=255&MSPPError=-2147217396

iddq, 2015-11-22:
Thank you guys. I think he forgot to tell me he has modified the source. It works well with transfer size 0xffff after the modification.

iddq, 2015-11-23:
I just get the info he works under mac so maybe this limitation is already removed on that platform.

mcuee, 2015-11-26:
That should be correct. I do not see that limit in the Mac OS X backend.

mcuee, 2015-11-26:
But the limit on Linux and Windows are real. Probably a documentation fix needs to be provided.

iddq, 2015-11-26:
I think you can remove this limitation, it works well with 65535 bytes.

diabolo38, 2015-11-26:
i suspect it work on most of today h/w  but not on some older usb controler.

anyway large  control make little sense.

mcuee, 2015-11-27:
To iddq: there is no issue with libusb except that this should be
documented. The 4096 is a limit to be compatible with older HW under Linux.
It is also imposed by Windows for full speed and low speed for most of the
Windows versions. We can not remove this limit.

http://sourceforge.net/p/libusb/mailman/message/33339533/
https://msdn.microsoft.com/en-us/library/windows/hardware/ff538112(v=vs.85).aspx

The application has to deal with this limit.

The only thing libusb project can do is to document this limit.

mcuee, 2015-11-27:
Closing this issue as there is another issue raised to update the documentation.
https://github.com/libusb/libusb/issues/110

closed 2015-11-27 by mcuee