libusb/libusb #126: Packet misorder on windows

kangjianbin, 2015-11-20:
In long run test, we found that usb packets were mis-order.
I guess the problem is in function 'windows_handle_events'.
It iterates on each FD, and then search the FD in flying_transfers.
The problem is libusb can't guarantee that FDs are in the order of submitted transfer.
So if two overlapped events are signaled almost at the same time, we will find mis-order transfer.

mcuee, 2015-11-27:
You should provide more detail of the issues you encounter, like debug log or things like that.

kangjianbin, 2015-11-29:
Hi mcuee,

We use 16 transfer for async packet input. For a transfer, a variable 'is_finish'
is used to indicate if this transfer is finished (that is, its callback function is called).
When a trasfer is complete, a global variable 'cb_count' will be increase by 1.
After 'libusb_handle_events_timeout' return,
we will search all transfers whose index are smaller than 'cb_count' by order and calculate
finished transfer 'finish_count'.
Sometime we found that finish_count' is less than 'cb_count' and data was misorder.
Below is the sample code. We did see some logs like:

    usb_process_event: Found misorder: 3, 4

```c
/* Transfer context */
struct xf_context {
	int		index;
    int		is_finished;        /* Indicate this transfer was finished */
    ....
	struct libusb_transfer *in;
	struct usb_be_context *ctrl;
};
```
```c
/* Global control */
struct usb_be_context {
	cyusb_handle *handle;
	int		is_exit;

	struct xf_context xf_ctx[XF_NUMBER * 2];
	int		swap_index;
	int		cb_count;               /* How many transfered was finished (callback function was called) */
	int		finish_count;           /* How many transfer was processed (we copied data from it) */
	int		error_count;
};
```
```c
/* transfer callback function */
static void LIBUSB_CALL usb_cb_in(struct libusb_transfer *in)
{
	struct xf_context *ctx = in->user_data;
	struct usb_be_context *ctrl = ctx->ctrl;

	ctx->is_finished = 1;            /* Indicate that this transfer is finished */
        ctrl->cb_count += 1;             /* global callback count increase by 1 */
	if (in->status != LIBUSB_TRANSFER_COMPLETED) {
		if (in->status != LIBUSB_TRANSFER_CANCELLED) {
			xlog(XL_WARN, "XF: failed: %u, index: %u\n",
			     in->status, ctx->index);
		}
		ctx->error = in->status;
	}
}
```
```c
/* After @libusb_handle_events_timeout return, call this function to verify */
static void usb_process_event(struct usb_be_context *ctrl)
{
	int i;
	int start = ctrl->finish_count;
	int start_index = usb_get_start_index(ctrl->swap_index);

	for (i = start; i < ctrl->cb_count; i += 1) {
		struct xf_context *ctx = &ctrl->xf_ctx[start_index + i];
		struct libusb_transfer *in;

		if (!ctx->is_finished) {
			/* Make sure our packets are in-order */
			break;
		}
                .....
                /* process the transfer */
		ctrl->finish_count += 1;
        ....
	}
	if (ctrl->finish_count != ctrl->cb_count) {
        /* We found this happended and data was misorder */
		xlog(XL_WARN, "%s: Found misorder: %u, %u \n",
		     __FUNCTION__, ctrl->finish_count, ctrl->cb_count);
    }
    ...
}
```

kangjianbin, 2015-11-29:
We found a method to avoid the bug. See the above function 'usb_process_event'. When a transfer was completed, we didn't process it immediately. Instead we wait unit all previous transfers to complete.

```c
        if (!ctx->is_finished) {
           	/* Make sure our packets are in-order:
                 If previous transfer isn't completed, we can't process this transfer 
                immediately.
                */
            break;
        }
```
With this modification, we run the data test for about 48 hours without error.

dickens, 2015-12-02:
I understand exactly why this issue can occur. An upcoming merge of event
abstraction code will address this issue, but I have no plans of addressing
this with the current code as it is not a straightforward fix.

mcuee, 2015-12-09:
Do we need to have a FAQ entry for this as this can be a serious bug for Windows libusb user.

mcuee, 2020-07-08:
Is this still a concern?

dickens, 2020-08-13:
Closing as this should be addressed by ba6b8bcb7ea204e65a3deec3be81aacc9f4b6d5a

closed 2020-08-13 by dickens