libusb/libusb #127: El Capitan: Problem with port number

dumych, 2015-11-20:
I've found strange behaviour on new Mac Os (El Capitan). I just little bit customized 
output format of  libusb example (https://github.com/libusb/libusb/blob/master/examples/listdevs.c), used libusb_get_port_number() instead of libusb_get_port_numbers()  and noticed that all my devices have port number 0. Is this OK?

I'm using Mac mini(Late 2012). I've checked 19 and 20 versions of libusb. 

My debug output:
 bus number 20 
 port number 0 
 device address 15 

 bus number 20 
 port number 0 
 device address 12 

 bus number 29 
 port number 0 
 device address 5 

 bus number 26 
 port number 0 
 device address 3 

 bus number 29 
 port number 0 
 device address 4 

 bus number 29 
 port number 0 
 device address 3 

 bus number 26 
 port number 0 
 device address 2 

 bus number 29 
 port number 0 
 device address 2 

 bus number 20 
 port number 0 
 device address 2 

 bus number 26 
 port number 0 
 device address 1 

 bus number 29 
 port number 0 
 device address 1 

 bus number 29 
 port number 0 
 device address 0 

 bus number 29 
 port number 0 
 device address 0 

 bus number 29 
 port number 0 
 device address 0

hjelmn, 2015-11-20:
ill add that to the todo list. Apple appears to have rewritten IOUSBFamily

mvp, 2015-11-20:
I also noticed that `libusb_get_port_numbers()` does not work on El Capitan for my utility [uhubctl](https://github.com/mvp/uhubctl/blob/master/uhubctl.c#L334), `libusb_get_port_numbers()` always returns 0.

Koromix [has a code](https://github.com/Koromix/ty/blob/master/libty/device_darwin.c#L301-L305) to support new Apple USB stack.
I think this file: [darwin_usb.c](https://github.com/libusb/libusb/blob/master/libusb/os/darwin_usb.c#L783) needs to be changed to handle new USB stack as "ty" does.

diabolo38, 2015-11-20:
the port alway 0 is not necessary wrong on other os (win xp) and base on the bus/hub structure that can happen . 
if you combine bus + device address and it always the same for different plug then that will be a real problem

mvp, 2015-11-20:
@diabolo38: in this particular case, it is definitely a bug. uhubctl always worked on any OSX version before, and on any Linux. But, on El Capitan, `libusb_get_port_numbers()` simply does not work: it DOES NOT fill array with port numbers, and returns 0 for filled array length.

hjelmn, 2015-11-20:
In this case it is wrong. Will patch later today.

mvp, 2015-11-20:
For reference: commit which adds new USB stack support to "ty" by Koromix:
https://github.com/Koromix/ty/commit/eee4cb952583ab213df7c1eb0871c36aa82aea0b

closed 2015-11-21 by hjelmn

mvp, 2016-01-19:
It seems like this fix is not really necessary - everything works now without this patch with libusb 1.0.20 under OSX 10.11.2 (15C50). I guess Apple added back exporting PortNum.

hjelmn, 2016-01-19:
Huh, guess someone else complained about it. Was a bug in the legacy USB stack support in 10.11. We will still have support for directly using the new stack in the next libusb version.

commit b7526c19 (2015-11-21): darwin: updates to support el capitan

With the new USB stack Apple changed the underlying device class from
IOUSBDevice to IOUSBHostDevice. Additionally, they removed the PortNum
property from the IO registry entry. This commit add code to detect the
currently running xnu version. If the version is at least 15 (10.11.0)
the new device class is used to find usb devices and the port number
is read from the IOService parent of the USB device instead of the
PortNum property.

Closes #127

May have some effect on either #123 or #96.