libusb/libusb #130: Make doc/Makefile.am better

hiyuh, 2015-12-09:
current `doc/Makefile.am` is simply not cool.
```
EXTRA_DIST = doxygen.cfg.in

docs: doxygen.cfg
        doxygen $^

docs-upload: docs
        ln -s html api-1.0
        scp -r api-1.0 pbatard@web.sourceforge.net:/home/project-web/libusb/htdocs
        rm -f api-1.0
```
* `docs` is used, why not `doc`...
*  `scp` command has hardcoded dev username and sf.net URI.
* users can not run `doxygen` and `install` via top Makefile.

a case of users optionally install doc via top Makefile is worth to consider.
* uBlock Origin recently blocks sf.net b/c some their scam.
* a few dev scene needs libusb doc w/o online connection.

NOTE: this is recreated from issue #105 to split.

diabolo38, 2015-12-09:
actually make doc doesn't require upload to sf 
it is a separate target "docs-upload:" likely use buy pete to  upload the online  doc to sf ;) 

Generated docs appear locally under html so usable offline  ?

What is not fully working is to build/install doc out of original source tree cos of path in the doxyfile , 
PROJECT_LOGO           = libusb.png
or 
IMAGE_PATH             =
It would need to be updated to point the source , as such we get 

error: Project logo 'libusb.png' specified by PROJECT_LOGO does not exist!

doc is still generated but the logo image link is broken, I tried to fix that but did not succeed.

anilgit90, 2016-07-26:
I believe the docs are generated properly, I did not encounter Project logo issue as suggested by @diabolo38.

dickens, 2020-08-13:
Closing, should be addressed in 4a5540a925e47923ed20b4ea68aabe550dac976d

closed 2020-08-13 by dickens