libusb/libusb #132: Use sensible function declarations for doc

hiyuh, 2015-12-09:
current libusb doc inherits following no sensible function declarations,
```
$ grep -e 'libusb_device_handle \*dev' libusb/libusb.h
        libusb_device_handle *dev_handle;
int LIBUSB_CALL libusb_get_configuration(libusb_device_handle *dev,
void LIBUSB_CALL libusb_close(libusb_device_handle *dev_handle);
libusb_device * LIBUSB_CALL libusb_get_device(libusb_device_handle *dev_handle);
int LIBUSB_CALL libusb_set_configuration(libusb_device_handle *dev,
int LIBUSB_CALL libusb_claim_interface(libusb_device_handle *dev,
int LIBUSB_CALL libusb_release_interface(libusb_device_handle *dev,
int LIBUSB_CALL libusb_set_interface_alt_setting(libusb_device_handle *dev,
int LIBUSB_CALL libusb_clear_halt(libusb_device_handle *dev,
int LIBUSB_CALL libusb_reset_device(libusb_device_handle *dev);
int LIBUSB_CALL libusb_alloc_streams(libusb_device_handle *dev,
int LIBUSB_CALL libusb_free_streams(libusb_device_handle *dev,
int LIBUSB_CALL libusb_kernel_driver_active(libusb_device_handle *dev,
int LIBUSB_CALL libusb_detach_kernel_driver(libusb_device_handle *dev,
int LIBUSB_CALL libusb_attach_kernel_driver(libusb_device_handle *dev,
        libusb_device_handle *dev, int enable);
        struct libusb_transfer *transfer, libusb_device_handle *dev_handle,
        libusb_device_handle *dev_handle, unsigned char endpoint,
        struct libusb_transfer *transfer, libusb_device_handle *dev_handle,
        struct libusb_transfer *transfer, libusb_device_handle *dev_handle,
        libusb_device_handle *dev_handle, unsigned char endpoint,
int LIBUSB_CALL libusb_control_transfer(libusb_device_handle *dev_handle,
int LIBUSB_CALL libusb_bulk_transfer(libusb_device_handle *dev_handle,
int LIBUSB_CALL libusb_interrupt_transfer(libusb_device_handle *dev_handle,
static inline int libusb_get_descriptor(libusb_device_handle *dev,
static inline int libusb_get_string_descriptor(libusb_device_handle *dev,
int LIBUSB_CALL libusb_get_string_descriptor_ascii(libusb_device_handle *dev,
```
there is no reason to use insane named arg, `dev` or `dev_handle` for `libusb_device_handle`.
b/c it looks simply eyesore due to many `libusb_device *dev` are w/ them.
just use `h`, `hnd` or `handle` looks simply better.

NOTE: this is recreated from issue #105 to split.

LudovicRousseau, 2016-01-06:
I agree with this request.

I just got caught by the API of ```int libusb_reset_device(libusb_device_handle * dev)```
You do not pass a ```libusb_device``` value, despite the ```dev``` name, but an handle.

Any objections to rename **dev** into **dev_handle** as in other libusb functions?

dickens, 2016-01-06:
No objection to consistency, but this proposal is to ditch any `dev_` prefix for `libusb_device_handle` arguments and simply use `h`, `hnd`, or `handle`. I vote for `hnd` for the handle since we are truncating to `dev` for devices.

LudovicRousseau, 2016-01-06:
The API *already* uses ```dev_handle``` for ```libusb_device_handle *```. See the examples above.

We do not have to limit the size of the parameter names. Explicit names should be preferred. ```hnd``` is not explicit for me.

hiyuh, 2016-01-07:
note, i also consider to rename `libusb_dev_handle` to `libusb_handle`,
b/c the mid word `dev` looks redundant to me.

LudovicRousseau, 2016-01-07:
@hiyuh, ``libusb_device_handle`` is the name of a type used by the API. We can't change it without breaking the API.
That is an idea that should be discussed in the API 2.0 (or 1.1) project. Can you open a different issue?

hiyuh, 2016-01-07:
@LudovicRousseau, yes i can.
but if we have following code before API updating, it won't break existing libusb apps?
```
typedef libusb_device_handle libusb_handle;
```

commit 8a0c1437 (2016-02-24): Misc: Make API parameter names consistent and sensible

Prior to this commit, API functions taking a libusb_device_handle
argument had the parameter named dev, handle, or dev_handle. This
commit changes the name of all libusb_device_handle parameters to
dev_handle in both the documentation and actual code.

Closes #132
closed 2016-02-24 by dickens