libusb/libusb #142: Getting errors from LIBUSB

TimurIsk, 2016-01-19:
I use libusb-1.0 for communication with the device.
```javascript
int rc=libusb_init(NULL); 
Q_ASSERT_X(rc == 0,"initialization","---libusb is not initialized----" );
libusb_set_debug(NULL, 3); 
m_handle = libusb_open_device_with_vid_pid(NULL, VID, PID); 
if (m_handle == NULL)
         return false; 
if (libusb_kernel_driver_active(m_handle,DEV_INTF)) 
        libusb_detach_kernel_driver(m_handle, DEV_INTF); 
if (libusb_claim_interface(m_handle,  DEV_INTF) < 0) 
   return false; 
```
I have two indepent software products, which both handle the device. If i use device in the first software, i will not be able to open_device_with_vid_pid in the second. How can i get status what device is busy? Libusb debug shows me error

diabolo38, 2016-01-20:
what os ? 
note that under window winusb back end on does not let you claim mutliple device 
lbusbk and maybe libusb win32 back end will do 
Still even if you can open and claim the device multiple time in multiple process those are responsible to interlock themselves and share handling the device properly. 

under linux mac the same is not possible , you might be able to open device multiple time in multiple process but only claim access to it in one process at a time. You can claim and release the interfaces in a shared collaborative  way but it is much slower (measured myself to be 10 to 100x slower for small bulk access)

TimurIsk, 2016-01-22:
I use windows

diabolo38, 2016-01-22:
sorry if  you do not catch all  maybe you have to read and learn a bit more libusb details ;) 

you should also consider using winusb api directly if  you only target window os,  likely you will face similar issue but maybe doc and support is better ?.

with libusb if you want to use device in a share way you have to managed it in your app. 
on window They are several  solution or option 
-libusbk back end driver, device kept open and usable in all process , you use collaborative access control using mutex/semaphore object  in all process to get control to device.. 
-winusb back end driver, you must open/close device (claim interface) on every access and retry until this fail (over process in used ) maybe add some time out for safety.

Up to me the first option is the fastet and also safest one ,
I'm not sure  you can rely on claim interface  status to detect the "busy" state or other device error.
The  status may not tell you and let you take recovery action in case a process crash during an access and leaving device in a bad state you could recover (halt flush ... etc). 
In option 1 Window mutex object will return WAIT_ABANDONED in case the lock is obtained cos the previous owner died with lock held see https://msdn.microsoft.com/fr-fr/library/windows/desktop/ms687032%28v=vs.85%29.aspx.

if you want keep things simple and not manage sharing and keep exclusive control on a single process  then use winusb backend and say to user "device busy".

mcuee, 2016-02-11:
I think this is not a real issue.

closed 2016-02-11 by mcuee