libusb/libusb #144: error C2011: 'timespec': 'struct' type redefinition (Windows 10 issue)

j796160836, 2016-01-24:
Hi, I got some error during install `bleno` and I found that might be some issue about installing libusb module.
https://github.com/sandeepmistry/bleno

```
lenovo-PC:bleno$ npm install usb

> usb@1.1.1 install C:\Users\lenovo\Documents\node\bleno\node_modules\usb
> node-pre-gyp install --fallback-to-build

Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  core.c
  descriptor.c
  hotplug.c
  io.c
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\descriptor.c) [C:\Users\lenovo\Documents\node\bleno\
node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\descriptor.c)
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\io.c) [C:\Users\lenovo\Documents\node\bleno\node_mod
ules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\io.c)
..\libusb\libusb\io.c(1227): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data [C:\Users\le
novo\Documents\node\bleno\node_modules\usb\build\libusb.vcxproj]
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\hotplug.c)strerror.c [C:\Users\lenovo\Documents\node
\bleno\node_modules\usb\build\libusb.vcxproj]

  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\hotplug.c)..\libusb\libusb\io.c(1902): warning C4244: '=': conversion from 'time_
  t' to 'long', possible loss of data

..\libusb\libusb\io.c(2435): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data [C:\Users\le
novo\Documents\node\bleno\node_modules\usb\build\libusb.vcxproj]
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\core.c) [C:\Users\lenovo\Documents\node\bleno\node_m
odules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\core.c)sync.c

  poll_windows.c
  threads_windows.c
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\strerror.c) [C:\Users\lenovo\Documents\node\bleno\no
de_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\strerror.c)
  windows_usb.c
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\sync.c) [C:\Users\lenovo\Documents\node\bleno\node_m
odules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\sync.c)
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\os\threads_windows.c)win_delay_load_hook.c [C:\Users
\lenovo\Documents\node\bleno\node_modules\usb\build\libusb.vcxproj]

  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\os\threads_windows.c)
..\libusb\libusb\os\threads_windows.c(194): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of da
ta [C:\Users\lenovo\Documents\node\bleno\node_modules\usb\build\libusb.vcxproj]
..\libusb\libusb\os\threads_windows.c(196): warning C4244: '=': conversion from 'const time_t' to 'long', possible loss
 of data [C:\Users\lenovo\Documents\node\bleno\node_modules\usb\build\libusb.vcxproj]
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\os\poll_windows.c) [C:\Users\lenovo\Documents\node\b
leno\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\os\poll_windows.c)
[... 52 lines cut ...]
npm ERR!
npm ERR! Failed at the usb@1.1.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the usb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls usb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\lenovo\Documents\node\bleno\npm-debug.log
lenovo-PC:bleno$
```

Some info I had install
1. Python 2.7.11
2. Visual Studio Community 2015 Edition
3. node-gyp@3.2.1
4. node@5.3.0
5. Windows 10

How can I fix is problem?
Any help or suggestion is appreciate.

j796160836, 2016-01-24:
I closed the thread in `node-usb` because of duplicate. The problem not solved.

mcuee, 2016-01-25:
This should have been fixed in latest git. Please try the git version of
libusb.

HollaDieWaldFee0815, 2016-04-13:
I simply removed the typestruc lines in threads_windows.h file and voila its working ;)

younesherlock, 2016-11-08:
@HollaDieWaldFee0815 : I totally agree ! I did the same in it works after five days of struggle !!! :@

SkyLined, 2016-12-30:
Alternatively, if you are installing another package through npm that has a dependency on usb and installation is failing, you may want to temporarily remove this struct from the SDK, so you can install usb and your package, before reverting the SDK.

You can comment out the struct in the Windows SDK file by looking for the file that contains it in the error message: you should see a message that refers to this file with a path similar to this:
```
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39)
```
Run notepad (or another editor) as administrator, as this file cannot be edited without admin rights. open the file, look for the struct, it should look like this:
```C++
    struct timespec
    {
        time_t tv_sec;  // Seconds - >= 0
        long   tv_nsec; // Nanoseconds - [0, 999999999]
    };
```

Just put "//" in front of every line to comment them out and save the file. Then try installing again; it should now work. After installation, remove the "//"-s again and save the file to undo your temporary modifications.

mcuee, 2016-12-31:
I tend to think this has been fixed in later version of the Windows SDK. In version 10.0.14393., the above was guarded by the following.
`#ifndef _CRT_NO_TIME_T
       struct timespec
       {
            time_t tv_sec;  // Seconds - >= 0
            long   tv_nsec; // Nanoseconds - [0, 999999999]
       };
#endif
`

mcuee, 2016-12-31:
Reopen the issue.

Good4m, 2017-01-25:
I am going to do some more testing later today. I will post my findings asap.

mcuee, 2017-02-13:
@Good4m Any new updates?

Good4m, 2017-02-15:
Negative. Sorry I just don't have the time I thought I had. You know how it is =/

mcuee, 2017-02-15:
Take your time. Will leave this ticket as open.
-- 
Xiaofan

dickens, 2017-08-25:
@Good4m, @mcuee, any update on this?

mcuee, 2017-09-10:
I think we should close this item since there is no further updates and I
believe it is not a libusb issue and this has been fixed in later version
of the Windows SDK.

mcuee, 2017-09-10:
Close this for now. Will reopen if someone provides more updates and if later version of Windows SDK shows problems.

closed 2017-09-10 by mcuee