libusb/libusb #147: Control transfer with IN && wLength==0 fails on Win*

fenugrec, 2016-02-01:
This is a duplicate of an email I sent to the mailing list last week :
I have a pretty simple test program (compiled against libusb 1.0.20, tested on Win 7/XP and linux) : it tries a control transfer with a vendor request:
bmreqtype = LIBUSB_ENDPOINT_IN | LIBUSB_REQUEST_TYPE_VENDOR | LIBUSB_RECIPIENT_DEVICE;
wLength = 0;
then it tries an interrupt transfer on EP 1.

I know, this is unusual (wLength=0 with direction=IN), but it's perfectly OK according to USB 2.0

Symptoms :
- on windows, I can run the program once; on the second attempt it fails at libusb_claim_interface() with LIBUSB_ERROR_NOT_FOUND. The only fix is to electrically reset the device.
- on linux, I can run the program multiple times without problems.

This is strange, so I did a bus capture. I have the sigrok capture files (a few 10-20kB files) if anyone's interested. Here is a distilled summary :

*** Windows capture
PID: SETUP ADDR 2 EP 0
PID: DATA0 [ C0 77 00 00 00 00 00 00 ] (request_type=0x77, vendor-defined)
PID: ACK
PID: OUT ADDR 2 EP 0
PID: DATA1 [ ] (empty)
PID: ACK

*** Linux capture
PID: SETUP ADDR 4 EP 0
PID: DATA0 [ C0 77 00 00 00 00 00 00 ]
PID: ACK
PID: IN ADDR 4 EP 0
PID: NAK
PID: IN ADDR 4 EP 0
PID: NAK
PID: IN ADDR 4 EP 0
PID: NAK
PID: IN ADDR 4 EP 0 (this is ~ 30 us after the DATA0 packet, probably the time it took for the device to react to the request)
PID: DATA1 [ ] (empty)
PID: ACK

As can be seen, on windows there's a PID: OUT that doesn't seem to belong in there ? This probably confuses the USB stack on my device and causes my "only works once unless I reset" problem.

mcuee, 2016-02-24:
This seems to your device FW bug and I think it has nothing to do with
libusb (it should not hang anyway).

mcuee, 2016-02-24:
BTW, I think  LIBUSB_RECIPIENT_DEVICE may be a mistake, you may meant
o use LIBUSB_RECIPIENT_INTERFACE. Have you tried that?

On Wed, Feb 24, 2016 at 4:19 PM, Xiaofan Chen <xiaofanc@gmail.com> wrote:
> This seems to your device FW bug and I think it has nothing to do with
> libusb (it should not hang anyway).

-- 
Xiaofan

fenugrec, 2016-02-24:
Hi,
Regardless of the FW bug (you may be right, it's probably not expecting a PID:OUT at that point ) :
- why does libusb on windows ever send a PID: OUT ?
- LIBUSB_RECIPIENT_DEVICE is allowed by USB, and is indeed what I wanted. I haven't tried _RECIPIENT_INTERFACE, do you think it would make a difference...

mcuee, 2016-02-25:
For the PID:OUT, how do you know it comes from libusb? It may come from
Windows and your FW needs to deal with it.

fenugrec, 2016-02-25:
I don't know what causes the PID:OUT... That's why I'm inquiring here, I don't know how the libusb backend works on windows . The "interesting" thing is that I see the same thing on win XP and win 7 !

mcuee, 2016-02-25:
BTW, I can not find your email in the mailing list. It is easier to discuss
in the mailing list and maybe others will have better ideas to help you.

fenugrec, 2016-02-25:
ah you're right, it never made it to the list, I have this in my archives: "Your mail to 'libusb-devel' with the subject
1.0.20 unusual control transfer fails only on windows

Is being held until the list moderator can review it for approval."

it probably never not get reviewed...  This was on Jan 22, is the message still "held for approval" ?

mcuee, 2016-02-25:
What is the reason for the mail to be held? Take note you need to subscribe
to the mailing list and the size can not be too big. Probably it is easier
that you send the email to the list again.

diabolo38, 2016-02-25:
I tested you case 
I also used a target recipient "device"  request type " vendor specific"  to form a valid request 
0xC0 "request type"  request  0x55 
That request arrived to my device (stm32) and it's default  f/w behavior is un-handled request "stall"  
Hence ibusb final status is  -9  "pipe error" 
On usb analyzer it's all going as expected 

I then started to see if with some handling on the device and non stall answer from f/w it was ok 
And it does not work ;( 
After setup stage host keep on repeating "out"  ? likely what yous see
I wan't say if device is wrong  i do not have much time for it sorry 

The host is not " caching"  but is not timing-out  , 
control never com back to s/w (until i reset the target device)
Analyzer trace below
[image]

So  theirs a possibility window is wrong here. 
You'r doing quite abnormal and stupid things i'm not much surprised your getting abnormal stupid ...    result ;)

mcuee, 2016-02-26:
Please refer to the discussions in the mailing list as well.
http://marc.info/?l=libusb-devel&m=145641394816327&w=2

mcuee, 2016-03-06:
In any case, this has nothing to do with libusb. So closing this issue.

closed 2016-03-06 by mcuee

alexanderplanar, 2018-12-17:
> I tested you case
> I also used a target recipient "device" request type " vendor specific" to form a valid request
> 0xC0 "request type" request 0x55
> That request arrived to my device (stm32) and it's default f/w behavior is un-handled request "stall"
> Hence ibusb final status is -9 "pipe error"
> On usb analyzer it's all going as expected
> 
> I then started to see if with some handling on the device and non stall answer from f/w it was ok
> And it does not work ;(
> After setup stage host keep on repeating "out" ? likely what yous see
> I wan't say if device is wrong i do not have much time for it sorry
> 
> The host is not " craching" but is not timing-out ,
> control never com back to s/w (until i reset the target device)
> Analyzer trace below
> [image]
> 
> So theirs a possibility window is wrong here.
> You'r doing quite abnormal and stupid things i'm not much surprised your getting abnormal stupid ... result ;)

Sorry for offtop, what the software did you use? (on the picture)