libusb/libusb #148: usbdk: device has no driver, libusb will not be able to access it

xlz, 2016-02-10:
I'm testing the usbdk backend with Kinect v2 `045e:02C4`. Most of the following is enumerating devices from libfreenect2

When it tries to libusb_open the Kinect, the error is LIBUSB_ERROR_NOT_FOUND.

```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [000009ac] libusb: debug [libusb_init] created default context
[ 0.000000] [000009ac] libusb: debug [libusb_init] libusb v1.0.20.11049
[ 0.000000] [000009ac] libusb: debug [windows_init] Windows 8.1 64-bit
[ 0.000000] [000009ac] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.000000] [000009ac] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.000000] [000009ac] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.000000] [000009ac] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.000000] [000009ac] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.000000] [000009ac] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.000000] [000009ac] libusb: debug [windows_init_clock] hires timer available (Frequency: 2630638 Hz)
[ 0.000000] [000009ac] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.000000] [000009ac] libusb: debug [htab_create] using 1021 entries hash table
[ 0.000000] [000009ac] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.000000] [000009ac] libusb: debug [libusb_get_device_list] 
[ 0.000000] [00000534] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.000000] [00000534] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.000000] [00000534] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.000000] [00000534] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
[ 0.000000] [000009ac] libusb: debug [windows_get_device_list] allocating new device for session [8C]
[ 0.000000] [000009ac] libusb: debug [windows_get_device_list] allocating new device for session [3F]
[ 0.000000] [000009ac] libusb: debug [get_api_type] driver(s): usbhub
[ 0.000000] [000009ac] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [000009ac] libusb: debug [windows_get_device_list] allocating new device for session [190]
[ 0.000000] [000009ac] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.000000] [000009ac] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [000009ac] libusb: debug [windows_get_device_list] allocating new device for session [2FB]
[ 0.000000] [000009ac] libusb: debug [get_api_type] driver(s): usbhub
[ 0.000000] [000009ac] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [000009ac] libusb: debug [windows_get_device_list] allocating new device for session [B1]
[ 0.000000] [000009ac] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.000000] [000009ac] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [000009ac] libusb: debug [windows_get_device_list] allocating new device for session [299]
[ 0.000000] [000009ac] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.000000] [000009ac] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [000009ac] libusb: debug [windows_get_device_list] allocating new device for session [41]
[ 0.000000] [000009ac] libusb: debug [windows_get_device_list] allocating new device for session [393]
[ 0.000000] [000009ac] libusb: debug [init_device] got bus number from ancestor #2
[ 0.000000] [000009ac] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.000000] [000009ac] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 
1086 bytes)
[ 0.000000] [000009ac] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 12): 
'\\.\USB#VID_5986&PID_026A#5&17CB10DD&0&12'
[ 0.000000] [000009ac] libusb: debug [windows_get_device_list] allocating new device for session [3F7]
[ 0.000000] [000009ac] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.000000] [000009ac] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 
bytes)
[ 0.015614] [000009ac] libusb: debug [init_de
vice] (bus: 1, addr: 3, depth: 1, port: 13): 
'\\.\USB#VID_0765&PID_5010#5&17CB10DD&0&13'
[ 0.015614] [000009ac] libusb: debug [windows_get_device_list] allocating new device for session [7F]
[ 0.015614] [000009ac] libusb: debug [init_device] got bus number from ancestor #2
[ 0.015614] [000009ac] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.015614] [000009ac] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 214 
bytes)
[ 0.015614] [000009ac] libusb: debug [init_device] (bus: 1, addr: 8, depth: 1, port: 1): 
'\\.\USB#VID_045E&PID_02C4#002676751047'
[ 0.015614] [000009ac] libusb: debug [windows_get_device_list] allocating new device for session [201]
[ 0.015614] [000009ac] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.015614] [000009ac] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 177 
bytes)
[ 0.015614] [000009ac] libusb: debug [init_device] (bus: 1, addr: 5, depth: 1, port: 11): 
'\\.\USB#VID_8087&PID_07DC#5&17CB10DD&0&11'
[ 0.015614] [000009ac] libusb: debug [windows_get_device_list] found existing device for session [190] (0.0)
[... 52 lines cut ...]
Descriptor for configuration #0: size 31
1. FilterID: 1, Port: 13, ID: 0765:5010, Configs: 1, Speed: 2
	USB\VID_0765&PID_5010 13
Descriptor for configuration #0: size 34
2. FilterID: 1, Port: 12, ID: 5986:026a, Configs: 1, Speed: 3
	USB\VID_5986&PID_026A 12
Descriptor for configuration #0: size 1086
3. FilterID: 1, Port: 11, ID: 8087:07dc, Configs: 1, Speed: 2
	USB\VID_8087&PID_07DC 11
Descriptor for configuration #0: size 177
4. FilterID: 1, Port: 7, ID: 138a:0017, Configs: 1, Speed: 2
	USB\VID_138A&PID_0017 9ea8fa601f3c
Descriptor for configuration #0: size 46
5. FilterID: 1, Port: 2, ID: 045e:02d9, Configs: 1, Speed: 3
	USB\VID_045E&PID_02D9 2
Descriptor for configuration #0: size 25
6. FilterID: 2, Port: 1, ID: 045e:02c4, Configs: 1, Speed: 4
	USB\VID_045E&PID_02C4 002676751047
Descriptor for configuration #0: size 214
7. FilterID: 4, Port: 1, ID: 8087:8000, Configs: 1, Speed: 3
	USB\VID_8087&PID_8000 1
Descriptor for configuration #0: size 25
```

@dmitryfleytman

dickens, 2016-02-24:
Your log indicates it is using the WinUSB backend, not the UsbDk backend. Check your compilation options.

xlz, 2016-02-24:
True, libusb_usbdk_dll_2013 is a separate project. The above result did not active or link to the libusb_usdk_dll files.

Is libusb_usbdk_dll a drop-in replacement or should I link to it as a separate version of libusb?

closed 2016-02-24 by xlz

xlz, 2016-02-24:
I'm asking that because the built libusb-usbdk-1.0.lib still has "DLL name" of libusb-1.0.dll, so if I link to libusb-usbdk-1.0.lib, the executable will link to libusb-1.0.dll instead of libusb-usbdk-1.0.dll.

dickens, 2016-02-24:
It is currently a drop-in replacement.