libusb/libusb #156: Missing hotplug callbacks on Darwin

hjelmn, 2016-02-27:
As per discussion on email there seem to be cases where hotplug events are missed on Darwin. Opening this bug to keep track of the problem.

hjelmn, 2016-02-27:
@LudovicRousseau I haven't forgotten. I plan to look at the code and see if there are any obvious problems before trying to reproduce the issue on my mac.

hjelmn, 2016-02-27:
Bah. Hit the wrong button.

LudovicRousseau, 2016-02-27:
The email discussion is at http://libusb.6.n5.nabble.com/libusb-Hotplug-event-lost-on-Mac-OS-X-td5715427.html

hjelmn, 2016-03-05:
I think I know what is happening. The hotplug event is triggered when the first device is plugged in. If you grab the device list in the hotplug callback then the other devices that were attached have not yet been processed. You should get multiple hotplug events though. Do you see a hotplug callback for each device?

LudovicRousseau, 2016-03-05:
I do not use libusb hotplug in my code. The device insertion is detected by another mechanism (using IOKit directly).
My driver is called because a new device has been detected. But my driver can't find it in the list returned by ``libusb_get_device_list()``.

In my case the new USB device is composed of a USB Hub and 2 smart card reader. Only 1 on the 2 smart card reader is present in the ``libusb_get_device_list()`` list.

If I remember correctly I tried adding a ``sleep()`` before calling ``libusb_get_device_list()`` to let some time to the hotplug mechanism to do its jobs. But that did not solved the problem.

hjelmn, 2016-03-05:
Huh, ok. I thought you had moved to using libusb for the detection. Sounds like the hotplug poll code is not working correctly... Will take a look and see what might be wrong.

hjelmn, 2016-03-05:
Can you create a gist with the log from the time you were able to reproduce the issue with loggin?

LudovicRousseau, 2016-03-06:
The problem is/was hard to reproduce and even harder with logging enabled :-(
I will try.

hjelmn, 2016-03-14:
Can you try adding a debugging message after the call to IOKitWaitQuiet in darwin_usb.c that prints out the return code of IOKitWaitQuiet? Its unlikely but call could be timing out.

LudovicRousseau, 2016-03-15:
@hjelmn the screen back light of my old (10 years) MacBook Pro just died. I have to setup a new test computer.
I do not forget this bug report.

mcuee, 2016-12-02:
Just wondering if this issue still exist or not with your new test computer.

mcuee, 2020-07-29:
Is this still applicable?

mcuee, 2020-12-07:
@LudovicRousseau Is still still applicable?

mcuee, 2021-06-05:
Closing this for now. Please reopen when things happen again.

closed 2021-06-05 by mcuee