libusb/libusb #158: Darwin: Can't claim device

tresf, 2016-02-29:
I'm using a high-level api wrapper, usb4java to claim a USB device on Mac.
 * The logic works 100% of the time on Windows and Ubuntu (with `udev` workarounds)
 * The logic rarely works on Mac OS X 10.11 (the 10.11.4 beta doesn't fix it)

The errors:
 * The original symptom of this problem was a `NullPointerException` due to assumptions about the active configuration. (reported by Java of course)
 * When I choose the first configuration blindly, I get `Configuration not active`.
 * When I try to force the active configuration using the low-level part of the wrapper, it does nothing.

Since `libusb` is so popular for other languages, how best to provide debug information for wrapper libraries?

I've filed a downstream bug here:
https://github.com/usb4java/usb4java/issues/51

And as well as our own downstream-downstream bug:
https://github.com/qzind/qz-print/issues/180

I have a C/C++ build environment on the 10.11 machine, but I'm not familiar enough with C nor the API to start writing test cases for a proper log, so any help is appreciated.

tresf, 2016-02-29:
Edit The `LIBUSB_DEBUG=4` worked fine for gathering logs.  Appended to bug report.

mcuee, 2016-02-29:
Can you post the output of "lsusb -vvv" for this device (under Linux)?
Thanks.

tresf, 2016-02-29:
@mcuee:

```
Bus 001 Device 023: ID 0eb8:f000  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0eb8 
  idProduct          0xf000 
  bcdDevice            2.05
  iManufacturer           1 Mettler Toledo
  iProduct                2 PS60    
  iSerial                 3 ????????
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4 HID-POS
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     292
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             100
Device Status:     0x0001
  Self Powered

```

hjelmn, 2016-02-29:
Can you give the debug output with the call to libusb_set_configuration. That call should be fixing your issue but it must be made before you try to claim an interface. You can use libusb_get_configuration to detect if the device is configured as a configuration of 0 means the device is unconfigured.

hjelmn, 2016-02-29:
Also, it might be better for you to use libhid to access this device. Due to the way Microsoft handles USB a lot of devices are erroneously HID to avoid having to make a driver.

tresf, 2016-02-29:
> Can you post the output of `lsusb -vvv` for this device (under Linux)?

@mcuee I hope the information I've posted is adequate. :+1: 

> Can you give the debug output with the call to `libusb_set_configuration`. That call should be fixing your issue but it must be made before you try to claim an interface. You can use [...]

@hjelmn This is where debugging is difficult.  I **think** I had already done this via https://github.com/tresf/qz-print/commit/30d5c2229b3cffaeba854263a9cb96f21d5ff98f#diff-66e51979c4dd4f537b85c3203a0f1568R55.

> That call should be fixing your issue but it must be made before you try to claim an interface.

@hjelmn Well, that's the part I'm a bit unfamiliar with.  Unlike the C or Python API, the High-Level Java API isn't clear how to do this, so I did it using the technique linked in the comment segment just above.  I'd be happy to provide these logs.

> [...] use libhid to access this device [if the device] erroneously use HID to avoid having to make a driver

@hjelmn Can you elaborate a bit on this recommendation?  This recommendation seems a bit drastic given the amount of success we've had with `libusb`, although what you are describing seems to be consistent with other implementations I've seen of the USB-scale code (we're looking to expand our hardware support beyond scales in the near future, hence `libusb`).

hjelmn, 2016-02-29:
The logs would be helpful to see if the device is actually getting configured.

As for using hidapi. In OS X USB devices that use the HID device class are opened by IOUSBHID and can not be used by libusb in any meaningful way without writing a codeless kext. This means you will need to use hidapi to use the device. It sucks but this is a side-effect of the manufacturer abusing the HID device class to simplify their Windows support. Its hard to tell if this will happen in this case because the configuration specifies the device class.

You should be able use libusb and hidapi side-by-side without any problems.

tresf, 2016-02-29:
> can not be used by libusb in any meaningful way without writing a codeless kext.

I've done the `kext` (awaiting approval from Apple for signing rights) per:

> The logic rarely works on Mac OS X 10.11, but does work reliably on OS X 10.7 (via #123, the 10.11.4 beta doesn't fix it, even with kext dummy extension, csrutil disable, kext-dev-mode=1, etc..)

> You should be able use libusb and hidapi side-by-side without any problems.

I'll have the logs later, and we may eventually bundle HID support if needed, but I'd first like to isolate the cause since this happens on 10.11 but not in 10.7 and not in Windows or Linux.

hjelmn, 2016-02-29:
Good luck on the signing rights. Apple is really strict on when they will allow KEXT signing.

tresf, 2016-02-29:
> Good luck on the signing rights. Apple is really strict on when they will allow KEXT signing. I couldn't get rights to sign a kext I wrote because I was not the one who created the hardware. Really irritating.

Thanks for the heads up.  The hardware manufacturer has been great to work with if I need to escalate, but getting approved for signing can be a tricky thing.  I know some Mac sound drivers have to go through the kernel extension route to override device-claim settings too, so this shouldn't be off the mark in regards to what's allowed.  If indeed the `kext` is impossible, the only other option is to drop USB support for Mac completely and fallback on HID-only, which would be a shame.

We write a JavaScript shim to talk to these devices through all major web browsers and we'd like to keep Mac on that list.  Anyway... back on topic... for now, I'll proceed as if it will be approved and get those logs over. :+1:

tresf, 2016-03-01:
I believe this to be the relevant portion of the logs where I set the active configuration... 

Should I not be closing this?  usb4java offers both a high-level api and a low level api, so I'm closing this out after setting the active configuration, then calling claim with the high-level device.  Is that a bad idea?

```diff
  [21.986931] [0000d903] libusb: debug [usbi_add_pollfd] add fd 63 events 1
  [21.986940] [0000d903] libusb: debug [usbi_add_pollfd] add fd 65 events 1
  [21.986986] [0000d903] libusb: debug [libusb_open] open 0.5
  [21.987021] [0000d903] libusb: debug [usbi_add_pollfd] add fd 67 events 1
  [21.987026] [0000d903] libusb: debug [darwin_open] device open for access
+ [21.987087] [0000d903] libusb: debug [libusb_set_configuration] configuration 1
+ [22.012154] [0000d903] libusb: debug [libusb_close] 
  [22.012190] [0000d903] libusb: debug [usbi_remove_pollfd] remove fd 67
  [22.012235] [0000d903] libusb: debug [libusb_exit] 
  [22.012244] [0000d903] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
  [22.012248] [0000d903] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
  [22.012252] [0000d903] libusb: debug [handle_events] poll() 2 fds with timeout in 0ms
  [22.012269] [0000d903] libusb: debug [handle_events] poll() returned 0
  [22.012274] [0000d903] libusb: debug [libusb_unref_device] destroy device 0.5
  [22.012278] [0000d903] libusb: debug [libusb_unref_device] destroy device 0.4
  [22.012282] [0000d903] libusb: debug [libusb_unref_device] destroy device 0.3
  [22.012286] [0000d903] libusb: debug [libusb_unref_device] destroy device 0.2
  [22.012290] [0000d903] libusb: debug [libusb_unref_device] destroy device 0.0
  [22.012293] [0000d903] libusb: debug [libusb_unref_device] destroy device 0.1
  [22.012297] [0000d903] libusb: debug [usbi_remove_pollfd] remove fd 63
  [22.016617] [0000d903] libusb: debug [usbi_remove_pollfd] remove fd 65
  [ERROR] 2016-02-29 20:55:55,832 @ qz.ws.PrintSocketClient:?
	Problem processing message
  javax.usb.UsbNotActiveException: Configuration is not active
	at org.usb4java.javax.Interface.checkActive(Interface.java:74)
	at org.usb4java.javax.Interface.claim(Interface.java:99)
	at qz.communication.UsbIO.open(Unknown Source)
	at qz.ws.PrintSocketClient.processMessage(Unknown Source)
	at qz.ws.PrintSocketClient.onMessage(Unknown Source)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:70)
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:68)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver$2.run(JettyAnnotatedEventDriver.java:210)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:635)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:555)
	at java.lang.Thread.run(Thread.java:745)
```

hjelmn, 2016-03-01:
Its hard to tell ordering from this log since the java error is not printed near any messages from the matching open call. Can you use gist to post the entire log?

The call to ```close``` should be fine as the device should remain configured until it is disconnected.

tresf, 2016-03-01:
> Can you use gist to post the entire log?

I thought of that and appended that right away. :wink:

hjelmn, 2016-03-01:
Weird. I do not see usb4java calling libusb_get_configuration anywhere. I would expect to see a message from that function:
```
                usbi_dbg("active config %d", *config);
```

I wonder where checkActive is getting the configuration from? It might be caching it somewhere that isn't catching your change to set the configuration.

mcuee, 2016-03-01:
Ref:
https://github.com/tresf/qz-print/blob/2.0/src/qz/utils/UsbUtilities.java
Firstly I do not see the code to setUsbConfiguration, only
getUsbConfiguration.

I will also recommend to forget about libusb for HID device under Mac OS X.
It is much easier to use HIDAPI instead. In fact, HIDAPI should be the
recommended tool to use for HID device if it is cross-platform projects.
Ref: http://www.signal11.us/oss/hidapi/
Ref:
https://github.com/libusb/libusb/wiki/FAQ#Does_libusb_support_HID_devices

tresf, 2016-03-01:
>  I do not see the code to setUsbConfiguration, only
getUsbConfiguration.

@mcuee the API doesn't provide one in the High-Level, on the low-level.  Also, you're looking at a moving target, it was here at the time of testing... https://github.com/tresf/qz-print/commit/a7b86dc1fb9d96cba71023a2ac52ff76c1a90cc7. 

> I wonder where checkActive is getting the configuration from? It might be caching it somewhere that isn't catching your change to set the configuration.

@hjelmn agreed.  I suppose I should not expect them to be synchronized when combining two completely separate apis from the same library.

What if I were to take `usb4java` out of the equation and tackle this in something like pyusb?  Is someone comfortable enough to assist me to get a quick snippet setup to claim the device?  (`homebrew`, etc)

> I will also recommend to forget about libusb for HID device under Mac OS X. It is much easier to use HIDAPI instead. In fact, HIDAPI should be the recommended tool to use for HID device if it is cross-platform projects.

Duly noted.  Are we OK to see this through?   Surely the research on `set_active_configuration` remains valid, no?  Our library isn't written for weight scales specifically, it's a generic USB wrapper, so regardless of the solution for this specific hardware, the `usb4java` portions will remain.

hjelmn, 2016-03-01:
@mcuee I agree with you but this device seems to be usable by libusb because the HID device class is specified in the configuration descriptor. Since the device will be exclusively opened by libusb I don't think that IOUSBHID will be able to attach once the device is configured. I may be wrong though.

@tresf I think I see an issue in your code @ https://github.com/tresf/qz-print/blob/30d5c2229b3cffaeba854263a9cb96f21d5ff98f/src/qz/communication/UsbIO.java#L30

You are getting the interface before the setConfiguration call. Try moving this line to after the setConfiguration call and see if that helps.

tresf, 2016-03-01:
@hjelmn Thanks for the spot check.  Tried this via https://github.com/tresf/qz-print/commit/94cd92871f2abb5a8e4937dd2999a46a4436cb0e#diff-52ae2363b650928540f4715b3eae938fR32 to no avail.

Growing a bit concerned about the lack of verbosity with the `usb4java` library, I decided to do the claim directly with the `LibUsb` low-level API via https://github.com/tresf/qz-print/commit/ba3d797b013c10c878e5aa3b207dd328fbf26b6e#diff-66e51979c4dd4f537b85c3203a0f1568R76 and the error message is quite different from the high-level API.

```diff
  [167.690699] [0000dd03] libusb: debug [libusb_set_configuration] configuration 1
  [167.690712] [0000dd03] libusb: debug [libusb_claim_interface] interface 0
+ [167.713521] [0000dd03] libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
  [167.713570] [0000dd03] libusb: debug [libusb_release_interface] interface 0
  [167.713577] [0000dd03] libusb: debug [libusb_close] 
```

Full gist here:
https://gist.github.com/tresf/0e6bde6566ad5bd2975c#file-usb4java-L50

hjelmn, 2016-03-01:
Hmmm, well that error suggest the device was successfully configured. I wonder if IOUSBHID somehow managed to grab the interface after the device was configured even with a user-space process holding the device. Unfortunately I can't look at the IOUSBFamily source code of 10.11 to verify as Apple has decided to no longer release the code. If IOUSBHID is grabbing the interface it would suggest that hidapi is going to be the only way to get this working. It looks like there has been some work on java bindings for the hidapi library: https://github.com/gary-rowe/hid4java

tresf, 2016-03-01:
>  IOUSBHID somehow managed to grab the interface after the device was configured even with a user-space process holding the device.

Ok, I finally claimed it... 

I had to use a remote desktop utility because this command kills the mouse... 

```bash
 sudo kextunload -b com.apple.driver.usb.IOUSBHostHIDDevice
```

Once this was run, I could claim the device.  :imp: 

There must be something wrong with my `kext`.
 * :white_check_mark: `kextutil` shows warnings, but no errors.
 * :white_check_mark: `kextload` doesn't return any errors
 * :no_entry: `kextstat` doesn't list it.

It's a pretty vanilla `kext` file.  Here's it unconfigured: [`apple/apple-kext.plist.in`](https://github.com/tresf/qz-print/blob/202470b32f93a5b71d9ef86fca3b2b91f75f7ed1/ant/apple/apple-kext.plist.in) and here's it configured: https://gist.github.com/tresf/0c808628e6f1c461cc0f

hjelmn, 2016-03-01:
Ah, you need to prevent IOUSBHID from loading on the interfaces. Example: https://github.com/hjelmn/gcusbadapter_osx/blob/master/gcusbadapter/Info_raw.plist

tresf, 2016-03-01:
> You need to prevent IOUSBHID from loading on the interfaces not the device. Example: https://github.com/hjelmn/gcusbadapter_osx/blob/master/gcusbadapter/Info_raw.plist

I was following the documentation here: https://developer.apple.com/library/mac/qa/qa1076/_index.html

 * Note that you should never add your own IOProbeScore property to a USB driver's property list.
 * Don't add extra keys; only use the key combinations shown in the following tables.
 * Table 1  Properties for matching a USB device (IOProviderClass is IOUSBDevice)
  * idVendor + idProduct + bcdDevice
  * idVendor + idProduct
  * ... etc.
 * Properties for matching a USB interface (IOProviderClass is IOUSBInterface)
  * idVendor + idProduct + bInterfaceNumber + bConfigurationValue + bcdDevice
  * idVendor + idProduct + bInterfaceNumber + bConfigurationValue
  * ...etc

I wasn't sure how reliable the `bInterfaceNumber` and `bConfigurationValue` would be, so I just listed the devices by `idVendor` + `idProduct`, which I feel will match the hardware device as broad as possible.

hjelmn, 2016-03-01:
The main takeaway from my example (which is a HID device) is that you need to match the IOProviderClass IOUSBInterface (or IOUSBHostInterface-- 10.11+ only) not IOUSBDevice. Your kext may very well be matching the device but that won't prevent IOUSBHID from grabbing the interfaces.

tresf, 2016-03-01:
> you need to match the `IOProviderClass` `IOUSBInterface` (or `IOUSBHostInterface`-- 10.11+ only) not `IOUSBDevice`. Your kext may very well be matching the device but that won't prevent `IOUSBHID` from grabbing the interfaces.

Gotcha, I'll switch that over and take another crack at it tonight.

Can you help explain the volatility of these settings?  e.g. Should these stay relatively identical for all HID devices? How would I know what these should be?

```xml
<key>bInterfaceNumber</key>
   <integer>0</integer>

<key>bConfigurationValue</key>
   <integer>1</integer>
```

hjelmn, 2016-03-01:
Those lines limit the matching to a specific configuration/interface. It prevents the kext from matching interfaces that were not intended.

tresf, 2016-03-01:
@hjelmn Thanks. I guess my question is, how do I know which options are safe to use for a particular HID device?  Are they almost always `bInterfaceNumber=0`, `bConfigurationValue=1` for HID, or does this vary from device to device?  Or does `lsusb` (linux, etc) give me the answers to what I'm asking?  If we hard-code these settings, I just want to know the process for getting these values. :+1:

hjelmn, 2016-03-01:
It device specific. You can tell from the device/configuration/interface descriptors which options you want. In most cases it will be ```bInterfaceNumber=0```, ```bConfigurationValue=1```.

tresf, 2016-03-02:
> In most cases it will be `bInterfaceNumber=0`, `bConfigurationValue=1`

Confirmed this on a Windows machine.

Unfortunately, I still can't get the kext to load.  You had previously mentioned `IOUSBHostInterface` on 10.11.  Is that a mandatory replacement for `IOUSBInterface` on El Capitan?  If so, I can do system detection and swap it out at install time.

hjelmn, 2016-03-02:
It should work with either the legacy IOUSBInterface or IOUSBHostInterface. There is probably some subtle bug that isn't being detected by kextload.

tresf, 2016-03-02:
> There is probably some subtle bug that isn't being detected by kextload.

Yeah, I'm having a hard time putting my finger on it.  Debugging these codeless `kext`'s seems to be a very "blackbox" process... (`kextstat |grep -v apple` won't list it, but I think that's normal).

I thought perhaps the package name ending in `kext` was causing issues.  I've renamed the package to something less `kext`-ish.  :smile:   The `kextutil` log [seems promising](https://gist.github.com/tresf/20b926d4b32603ccbc00)... 

I've detached and reattached the device several times and it seem to stay available.  Here's the current log (I'll be squashing commits very shortly if this starts to work.)

I'm still a bit uncertain what changed... :confused: 

On a side note, after some more digging around, I found another comment about Apple denying the `kext` for non-manufacturer requests from the Dolphin Emulator mailing list (getting Wii-mote support).

> [Apple] will only let Nintendo sign a kernel extension that supports the WUP-028 [Wii controller]. Nintendo will never do it so the WUP-028 is unusable on OSX without putting the kernel in development mode (bypassing signing). Those willing to turn off kext verification can ... [truncated]

Although I can *somewhat* understand Apple's stance in regards to the request coming from an emulator community... 

Do you know if Apple does the signing themselves, or do they offer a new developer certificate which has `kext` capabilities enabled?  The documentation refers to `codesign`, so I assume a new cert is issued.

tresf, 2016-03-02:
Rebooted and same familiar message... 

```
Error: USB error 3: Unable to claim interface: Access denied (insufficient permissions)
```

Unplugged the USB device and plugged it back in and now it is letting me claim it... 

Are you aware if `<key>OSBundleRequired</key><string>Root</string>` is required for boot-time claiming of an interface?

tresf, 2016-03-02:
Ok... it's finally working first try after a reboot without manually firing `kextload`... The only thing I changed was I took the space out of the `kext` name. (i.e `Foo Bar.kext` --> `FooBar.kext`).  I'll continue testing.

tresf, 2016-03-02:
... and now it's not working again unless `kextload`, unplug, plug back in...  We'll, I'm closing this issue out as we simply cannot get reliable results with the `kext` hack and it truly appears to be something caused by Apple.

We'll start implementing the `libusb` driver, at least to get the USB scale support working.  Thanks a bunch for all the help.

closed 2016-03-02 by tresf

tresf, 2016-03-10:
> Good luck on the signing rights. Apple is really strict on when they will allow KEXT signing. I couldn't get rights to sign a kext I wrote because I was not the one who created the hardware. Really irritating.

@hjelmn, As a follow-up to this, Apple **did** grant `kext` signing authority to our company for this purpose.  Due to the aforementioned device claiming issues, we are unlikely to use it for the purposes of grabbing the HID devices, but I wanted to offer an update on that issue, as it seems to be a hurdle that many organizations struggle with. :+1:

hjelmn, 2016-03-10:
@tresf Nice. I keep hitting a brick wall when trying to get signing right but I am an open-source developer. Here is their latest response:

```
KEXT signing is intended for commercial products that maintain an ownership link between the product and the KEXT.  In the case of a third party retroactively creating a KEXT a signing certificate cannot be generated.

Thanks,

Developer Technical Support
Apple Worldwide Developer Relations
```

tresf, 2016-03-10:
@hjelmn We're open source as well, but operate a commercial business model, which may be the deciding factor here.

hjelmn, 2016-03-11:
@tresf That probably is the difference. Find it a little odd that Apple is willing to shut out third party support when there is no chance of first party support.

ViralAgency, 2020-05-13:
The problem seems to be due to a conflict between different drivers, which use the same libraries, and in my case they were related to previous Samsung device installations. i had solved this way:

    kextstat | grep -v apple

To have a return like this:

```
  70    0 0x57574000 0x3000     0x2000     com.devguru.driver.SamsungComposite (1.2.4) <33 4 3>
  72    0 0x57831000 0x7000     0x6000     com.devguru.driver.SamsungACMData (1.2.4) <71 33 5 4 3>
  94    0 0x57674000 0x3000     0x2000     com.devguru.driver.SamsungACMControl (1.2.4) <33 4 3>
```
Then:

```
$ sudo kextunload -b com.devguru.driver.SamsungComposite
$ sudo kextunload -b com.devguru.driver.SamsungACMData
$ sudo kextunload -b com.devguru.driver.SamsungACMControl
```
Done. Enjoy