libusb/libusb #159: Wild timestamps in usbi_log_v()

gvanem, 2016-03-01:
In an application using libusb (MSVC-2015) on Win-10, I see log-lines with some *bogus* timestamps. E.g.:
```
[ 1.028547] [00002570] libusb: debug [windows_get_device_list] found existing device for session [96] (4.4)
[ 1.028547] [00002570] libusb: debug [windows_get_device_list] found existing device for session [2D3] (4.1)
[ 3.349353] [00002570] libusb: debug [windows_get_device_list] found existing device for session [14B] (4.14)
[ 3.349853] [00002570] libusb: debug [windows_get_device_list] found existing device for session [299] (3.2)
```

This is in a program executing in under 1 sec!

I think the delta-time calculation in `usbi_log_v()` is a bit wrong. I'm getting seemingly correct timestamps with this patch:
```
--- a/core.c 2016-03-01 06:58:03
+++ b/core.c 2016-03-01 10:54:07
@@ -2322,7 +2322,7 @@
                usbi_log_str(ctx, LIBUSB_LOG_LEVEL_DEBUG, "[timestamp] [threadID] facility level [function call] <message>" USBI_LOG_LINE_END);
                usbi_log_str(ctx, LIBUSB_LOG_LEVEL_DEBUG, "--------------------------------------------------------------------------------" USBI_LOG_LINE_END);
        }
-       if (now.tv_usec < timestamp_origin.tv_usec) {
+       while (now.tv_usec < timestamp_origin.tv_usec) {
                now.tv_sec--;
                now.tv_usec += 1000000;
        }
```

I.e. no huge jumps in delta-time with the above patch.

gvanem, 2016-05-19:
Seems this is still a problem; reopened.

dickens, 2016-05-30:
This issue doesn't really make sense. The range of the usec field is
0-999,999 inclusive and is ensured by the fact that the result is MOD
1,000,000 [1]. This means that when comparing two timestamps populated by
the same function, if one timestamp's usec field is less than the other,
adding 1,000,000 will certainly make its value higher than the other in
only one iteration.

[1] https://github.com/libusb/libusb/blob/master/libusb/core.c#L2294

closed 2016-06-12 by mcuee