libusb/libusb #160: autogen.sh

sammoes, 2016-03-01:
Hello, I am unaware how it works exactly, but there are already other programs that require version 1.0.0.20 but this is not in apt-get (at least my apt-get) so people may find that out after wondering why it doesn't work.
Then when building from source, the instructions in INSTALL may or may not be accurate but autogen.sh seems to work best but it is not mentioned in INSTALL.

hjelmn, 2016-03-01:
autogen.sh is not needed when building from a release tarball. That is why it is not mentioned in INSTALL.

hjelmn, 2016-03-01:
I would argue we should remove bootstap.sh and autogen.sh from the tarball.

sammoes, 2016-03-01:
The simplest way to compile this package is:
  1. `cd' to the directory containing the package's source code and type
     `./configure' to configure the package for your system.
     Running `configure' might take a while.  While running, it prints
     some messages telling which features it is checking for.
  2. Type `make' to compile the package.
  3. Optionally, type `make check' to run any self-tests that come with
     the package.
  4. Type `make install' to install the programs and any data files and
     documentation.
didn't work on a XU4 with ARM today although I was on Ubuntu 15.04. Also using autoconf caused many errors and it is not a path you may want to go down resolving them.
Then someone alerted me to autogen.sh on the mailing list and that worked like a charm from a freshly downloaded repo.zip

hjelmn, 2016-03-01:
configure should have worked without using the autogen.sh. The whole point of autotools is to generate a configure script will run on any POSIX compliant platform (within reason). It possible the autotools versions used to make the tarball were a little old. @mcuee what autoconf/automake versions did you use to generate the tarball?

sammoes, 2016-03-02:
You can check the mailing list, I put the complete output there, I can copy it here if you want. This is part of it (I did change the files sometimes in the process, mainly commenting out lines that showed up in stderr)

root@odroid:/home/odroid/libusb-master# autoconf
root@odroid:/home/odroid/libusb-master# ./configure 
./configure: line 2277: AM_INIT_AUTOMAKE: command not found
./configure: line 2278: AM_MAINTAINER_MODE: command not found
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for inline... inline
./configure: line 3378: AM_PROG_CC_C_O: command not found
checking operating system... configure: error: unsupported operating system

mcuee, 2016-03-02:
@Chris What autoconf/automake versions did you use to generate the release
tarball?

mcuee, 2016-03-02:
BTW, I think we should close this ticket. This is rather a support
request and it is now being addressed in the mailing list.

hjelmn, 2016-03-05:
Agreed. I think we need to just make sure we use as new a version of autotools as possible. Might be worth adding version checks to bootstrap.sh or configure.ac.

closed 2016-03-06 by mcuee