libusb/libusb #167: Misc: To release 1.0.22

mcuee, 2016-03-06:


mvp, 2017-12-18:
What is an ETA for 1.0.22 release? It is currently past due once again.
Current blockers https://github.com/libusb/libusb/issues/252 and https://github.com/libusb/libusb/pull/236 have not seen any updates for more than a year, and there is no evidence that they will be fixed or implemented anytime soon.

Is it possible to simply cut 1.0.22 from trunk? It would fix OSX issue https://github.com/libusb/libusb/issues/303, which I filed 7 months ago, and a lot of other problems.

cc: @dickens, @hjelmn.

LudovicRousseau, 2017-12-19:
I remember we forked libusb because the previous maintainer was not reactive enough to make new releases.
It looks like we are not making a better job :-(

dickens, 2017-12-22:
I'm sorry everyone, this has been a very busy season of life for me :(

I'm officially on vacation and will be giving this attention immediately. There are significant improvements for the Windows backend coming!

Let's get an RC out before the end of the year!

mvp, 2018-02-18:
Are there any expectations of making final 1.0.22 release soon?
This milestone is past due by 3 months...

mcuee, 2018-02-24:
I understand. It seems the rc is pretty good. The only problem seems to be some rough edges on the Windows side. I am actually think we can release 1.0.22 now.

mvp, 2018-02-26:
I hate to be that guy, but 2 more months later, we still don't have 1.0.22 release...
Soon we'll be celebrating 2 years since this milestone was created...

mcuee, 2018-02-26:
Rc3 has just been released so we should expect a release soon.
-- 
Xiaofan

closed 2018-03-25 by dickens