libusb/libusb #168: Hotplug events arrive too soon

spaceotter, 2016-03-08:
I'm getting hotplugging events for arrived devices before the permissions on the devnode have been set, so trying to open the device fails. If I spawn a thread that checks the permissions later, they are eventually set and the device can be opened. I wrote a program that uses libudev directly and when it gets hotplugging events, the permissions on the devnode are correct. I tried the hotplugtest and it does not print anything to the console.

hjelmn, 2016-03-08:
Are you building libusb with udev support?

spaceotter, 2016-03-09:
I'm using the one from my package manager. Using `nm` to peek at the symbols inside show some udev* symbols so I'm assuming that means it was compiled with udev support.

spaceotter, 2016-03-09:
I have to use a different build system than libusb, and there is something set up to build libusb from a more recent source. Using this has the same problem.

hjelmn, 2016-03-09:
We shouldn't be getting udev notification until the permission rules have run. Usually when the permissions are wrong its because the netlink hotplug backend is in use. In that case notification comes at the same time for libusb as it does for udevd. If you nm libusb you will have either ```linux_netlink_event_thread_main``` or ```linux_udev_event_thread_main```.

dickens, 2016-03-09:
In the environment where my main libusb application runs (RHEL 6.7), the
app gets hotplug notifications from libusb (using udev) and I run into this
same issue. I specifically check for LIBUSB_ERROR_ACCESS and do a small
loop with delays to try and open the device. Perhaps it is older versions
of libudev?

hjelmn, 2016-03-09:
Strange. I was under the impression the udevd netlink message only goes out after udevd matching is complete.

spaceotter, 2016-03-12:
I altered the build scripts, which were not building with udev support correctly. I am not sure why I continued to see the problem after the alteration the first few times, but it seems to be resolved now.

closed 2016-03-12 by spaceotter