libusb/libusb #169: Add Solaris backend

leilchen, 2016-03-10:
Many applications turn to libusb 1.0 API. Solaris needs to support that set of APIs.

mcuee, 2016-03-10:
Not so sure how many people are using Solaris, but that will be good.
Probably the existing OpenUSB Solaris Backend can be the starting point.

Ref: https://sourceforge.net/projects/openusb/

BTW, OpenUSB seems to be inactive now since the last commit is on
24-June-2013.

leilchen, 2016-03-10:
> Not so sure how many people are using Solaris, but that will be good.
There's requirement. I'll work on it.

> Probably the existing OpenUSB Solaris Backend can be the starting point.
>
> Ref: https://sourceforge.net/projects/openusb/
>
> BTW, OpenUSB seems to be inactive now since the last commit is on
> 24-June-2013.
As the maintainer of that project, I feel depressed. But as you said, that'll be a starting point for libusb 1.0 implementation.

Thanks,
Lei Chen

hiyuh, 2016-10-26:
pull request #177 is closed, so close this issue too?

leilchen, 2016-11-02:
Yes, please go ahead with proper procedures.

Thanks,

Lei Chen

closed 2016-11-03 by mcuee