libusb/libusb #172: Can not cancel transfer when running through valgrind

dennishamester, 2016-03-16:
I just ran into some bug which seems to happen only if I use valgrind. When I cancel a transfer I set a some variable to 1 in the transfer handler and wait for that to happen by using `libusb_handle_events_completed`. Everything works fine unless I run my program through valgrind. The cancel never reaches the transfer handler. Instead it continues with `LIBUSB_TRANSFER_COMPLETED` events (submitted from the handler). What's even weirder, is that `libusb_handle_events_completed` returns as if my variable has been set to 1, which it hasn't. It also doesn't wait for the 60s timeout. It returns immediately. Calling `libusb_handle_events_completed` multiple times doesn't help, the cancel never arrives at the handler. So in the end, I free the transfer while it hasn't completed.

So far, I have never seen this happening outside of valgrind. It also happens only when there is high traffic on that particular transfer at the time of the cancel. If it matters: it is an interrupt transfer.

Version: libusb 1.0.20 on Arch Linux amd64

Dimonsky, 2016-11-08:
I have the same problem under Windows.
The example coming with the library dpfp_threaded.c  is also a subject of this issue....

mcuee, 2016-12-02:
Hmm, dpfp_threaded.c is only available under Cygwin if using Windows. What do you mean by it is also a subject of this issue?

Dimonsky, 2016-12-02:
I used this module as an example for some Windows (and actually, cross-platform) code in my program.
In my code I call libusb_handle_events only once after libusb_cancel_transfer, because due to issue described above, this never leads to an expected result. 
(in original example it is called in while-loop, unitl _transfer is NULL.)

mcuee, 2016-12-02:
I see. Thanks for the clarification. Do you only see the problem under Valgrind or even without Valgrind?

Dimonsky, 2016-12-02:
without, just VS2013 without any spec. tools.

mcuee, 2016-12-07:
I see. This seems to be a bit different from the original poster who says that it seems to happen only under Valgrind. Just wondering if you have tried your program under non-Windows OS like Linux or Mac OS X and if it happens there as well? Or if your program is Windows only. It may be good that you provide some code snippets to demonstrate the problem. Thanks.

mcuee, 2020-08-31:
Close this for now. Please reopen if new info are provided.

closed 2020-08-31 by mcuee