libusb/libusb #175: Merge usb zerocopy patch

xlz, 2016-03-18:
The patch is here http://thread.gmane.org/gmane.comp.lib.libusb.devel.general/22679/focus=22798 (with one issue remained).

The kernel support has been merged to mainline https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=f7d34b445abc00e979b7cf36b9580ac3d1a47cd8.

mcuee, 2016-03-18:
IMHO it will be better that you reply to that email than creating a new
ticket.

mcuee, 2016-03-18:
You may need to subscribe to the libusb mailing list in order to post
to the mailing list.
URL:   https://lists.sourceforge.net/lists/listinfo/libusb-devel

xlz, 2016-05-16:
Linux 4.6 is released with usbfs zerocopy support. This should be merged.

dickens, 2016-05-30:
Closed via a283c3b5

closed 2016-05-30 by dickens