libusb/libusb #180: OS X 10.10 libusb_claim_interface return -3

mryraghi, 2016-04-06:
Hi,
I know that the question has already been raised in the past ([here](https://github.com/libusb/libusb/issues/97)). 
The purpose of my question is to get more information, if possible, about how to resolve this problem. I do not know what a "codeless kext" is.
Btw, I'm implementing this in a daemon process.

Thank you in advance,
Romeo

hjelmn, 2016-04-06:
Information on codeless kexts can be found on google. With 10.11 Apple now requires all kernel extensions to be signed by individuals/organizations with special signing extensions (difficult to get approved). Without this signature SIP needs to be disabled.

If the device is an HID device your best bet is to use hidapi and not libusb.

mryraghi, 2016-04-06:
Thank you for the fast reply!

closed 2016-06-12 by mcuee